gnss-sdr/gnss-sdr #398: Duration halved

esqt, 2020-07-06:
**Setup:**
- File signal source with complex samples
- Number of samples indicated manually with the `SignalSource.samples` parameter (not the default value 0)

---

**Behavior:**
The duration is the half of what was expected (_number of samples / sample rate / 2_ instead of _number of samples / sample rate_).

---

I think the issue comes from there:

https://github.com/gnss-sdr/gnss-sdr/blob/66a908fc42debcb7157acd773bccc056233bdc9a/src/algorithms/signal_source/adapters/file_signal_source.cc#L204-L209

By definition, the duration should always be the number of samples divided by the sample rate.

It would make more sense for the variable `samples_` to be the number of samples, which is not the case for complex samples with the current version:

https://github.com/gnss-sdr/gnss-sdr/blob/66a908fc42debcb7157acd773bccc056233bdc9a/src/algorithms/signal_source/adapters/file_signal_source.cc#L198

---

The following patch seems to fix the issue: [diff.txt](https://github.com/gnss-sdr/gnss-sdr/files/4880255/diff.txt)

carlesfernandez, 2020-07-14:
What item size are you using? Maybe there is some confusion between `ibyte` and `byte`, or `ishort` and `cshort` (probably they could be defined in a better way). Please check https://gnss-sdr.org/docs/tutorials/understanding-data-types/ on how are they defined in this context, and please come back if you find any inconsistency (or a better way to define type names).

esqt, 2020-07-15:
Well, I guess it all comes down to the definition of a sample. A complex "sample" can also be seen as two real valued "samples" (its real and imaginary parts). It's only a matter of definition.

Nevertheless, I feel like the `SignalSource.sampling_frequency` and `SignalSource.samples` should use the same definition. At the moment, the sampling frequency is not equal to the number of samples divided by the duration, which is confusing.