gnss-sdr/gnss-sdr #400: QUESTION about update AD9361_MANAGER.cc

noelpedro, 2020-07-10:
I'm confused about the following two snippets of code. The first snippet is from ad9361_manager.cc from version 10 the other is from the master master. My question is does the code work the same? Meaning rx0_i enables the real channel and rx0_q enables the imaginary channel for rx1or or both?

Does rx1 refer to just the real part of the adc or does it refer to the rx1 sma port?
Same question for rx2.

```
struct iio_channel *rx0_i;
struct iio_channel *rx0_q;

```

```
 std::cout << "* Enabling IIO streaming channels\n";
    if (rx1_enable_)
        {
            iio_channel_enable(rx_chan1);
        }
    if (rx2_enable_)
        {
            iio_channel_enable(rx_chan2);
        }
```

carlesfernandez, 2020-07-14:
Well, maybe you left some important code out of the snippets, but they are clearly not doing the same thing. The first one defines two pointers of type  `struct iio_channel` to the I&Q components of the receiving channel 0, and the second one enables receiving channels 1 and/or 2 (you can call them 0 and 1) upon configuration. The second snippet allows using the two receiving channels of the AD9361 chipset.

noelpedro, 2020-07-14:
Thanks,

And sorry for missing code

on v12 its:
`struct iio_channel *rx_channel1;`
`struct iio_channel *rx_channel2;`

So
`iio_channel_enable(rx_chan1);`

would enable I and Q for channel 1?

carlesfernandez, 2020-07-15:
Yes, you need to enable at least one channel. That final line will enable channel 1 (I & Q). Check the libiio interface here: https://analogdevicesinc.github.io/libiio/v0.4/group__Channel.html 

Hope this helps,
Carles

closed 2020-07-15 by noelpedro