gnss-sdr/gnss-sdr #402: SBAS in GNSS-SDR

AgileEngineeringLLC, 2020-07-11:
Is SBAS enabled in GNSS-SDR? I'd like to use both GPS and SBAS L1 for PVT, but not seeing if it's been added to GNSS-SDR.

Thanks much! Don

carlesfernandez, 2020-07-14:
Hi @AgileEngineeringLLC ,

there is some code for SBAS (WAAS and EGNOS) but, unfortunately, it needs some love. It's quite unmaintained, and some work would be needed to make it live again. It's something I would like to do (most of the required code is already there), but I never find the time for doing so. Any contribution would be appreciated :-)

Best regards,
Carles

dareis08, 2022-06-21:
was this ever picked up?  I'm looking at trying set up an SBAS receiver.

ValZapod, 2022-07-13:
WAAS will require ITRF2008 reference frame, while GPS moved to G2139, i.e. ITRF2014.

ValZapod, 2023-05-13:
>GPS moved to G2139, i.e. ITRF2014

GPS already moved to IGS20 (unofficially G2238), i.e. ITF2020, requires PSD.

Also wireshark now parses some of it: https://gitlab.com/wireshark/wireshark/-/merge_requests/10495/diffs