gnss-sdr/gnss-sdr #403: Output messages

matteocutugno, 2020-07-13:
Dear community,

is there a way to let GNSS-SDR output NMEA messages or other protocol on UART or SPI ?
In particular, I want a bladeRF x40 running GNSS-SDR 0.0.12 to output GGA, VTG, and RMC NMEA messages via UART or SPI to develop a loosely coupled sensor integration.
In addition, would be interesting to understand if GNSS-SDR can deliver raw measurements like Pseudoranges and Doppler shift in order to develop a tightly coupled sensor integration.

Thank you all

Matteo

AgileEngineeringLLC, 2020-07-13:
Following... great questions!
Don

carlesfernandez, 2020-07-14:
Dear @matteocutugno,

you can redirect the NMEA messages to whatever you want (see https://gnss-sdr.org/docs/sp-blocks/pvt/#nmea-0183)

```
PVT.flag_nmea_tty_port=true
PVT.nmea_dump_devname=/your/port/file/address
```

The default value for `nmea_dump_devname` is `/dev/tty1`, but you can redirect that to any device accepting serial data. Maybe @Arribas can add more to that.

Regarding tightly coupling integration with INS, you should develop specific tracking blocks for that, since you would need to interact at the tracking parameters level. GNSS-SDR architecture is ready for that, but the required additions are a huge amount of work. Maybe it would be simpler to start with loosely coupled approaches, for which GNSS-SDR provides everything you need for implementing GNSS+INS solutions.

Best regards,
Carles

carlesfernandez, 2020-07-17:
Hi @matteocutugno ,

I found a bug in GNSS-SDR v0.0.12. The value of the `PVT.nmea_dump_devname` parameter was ignored, so it ALWAYS forwarded NMEA messages to `/dev/tty1` when `PVT.flag_nmea_tty_port=true`, no matter the serial port identifier you wrote in your configuration file.

This is now solved in the `next` branch.

matteocutugno, 2020-07-26:
Thank you @carlesfernandez.
I will let the community know any step forward.

Matteo

matteocutugno, 2020-09-11:
Dear @carlesfernandez ,

is there a way to select wich NMEA sentence gnss-sdr can send over serial port?
For my application of sensor integration I need GGA, VTG, RMC while, by default, I saw that the software sends GGA, GSA, GSV, RMC.
Thanks

Matteo

matteocutugno, 2020-09-24:
Dear @carlesfernandez,
 
The bug in 0.0.12 that you found is fixed also in 0.0.13 next?
Because I can't get NMEA messages on serial port.
These are the lines that I added in my configuration file.

PVT.flag_nmea_tty_port=true
PVT.nmea_dump_devname=/dev/ttyUSB2

I'm using gnss-sdr version 0.0.13.git-next-47c23b2 in Ubuntu 16.04.
GNSS-SDR computes PVT and writes correctly NMEA on file but nothing on serial ports.
Am I missing something?

Thanks

Matteo

matteocutugno, 2020-09-25:
Dear @carlesfernandez 

Finally I'm able to use GNSS-SDR with IMU. 
GNSS-SDR sends NMEA messages over serial port correctly.
I added some lines of code in GNSS-SDR 0.0.13 next  in order to output also VTG messages and I disabled unwanted messages (GSA and GSV).
So, the version I modified of GNSS-SDR outputs NMEA GGA, RMC and VTG. This is all I need for loosely coupled integration.
In the next days I will carry out some tests to assess the quality of the navigation results.
If could be of insterest to the community I can share the code.
Have a good day 

Matteo

AgileEngineeringLLC, 2020-09-25:
congrats matteo!! would be great to have that added in a future release. 

Don

Don Kelly
Technologist | Engineer
Agile Engineering, LLC

LinkedIn: www.linkedin.com/in/kellydak
Cell: 281-221-2853
4403 Orange Leaf Court
Houston, TX 77059

AgileEngineeringLLC, 2020-09-28:
GNSS-SDR Community,

I installed a GeForce GTX (GPU) board, and am not trying to build GNSS-SDR with the -DENABLE_CUDA=ON flag. CMake gets 83% done and then gives me the following error (see screen capture below for details):

nvcc fatal: Unknown option ‘Wno-psabi’

I’m using Ubuntu 20.04, and I have the CUDA driver and Toolkit 11.1 installed. Any suggestions on how I might get past this error?

Thanks!

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059