gnss-sdr/gnss-sdr #410: How to read two file in two different band?

Tanggeshi, 2020-07-22:
I want to run gnss-sdr in dual-band mode by reading two different file in two different band, would you please give an example configure file ?  Thank you!

carlesfernandez, 2020-07-22:
Hi @Tanggeshi ,

this is the configuration example of a GPS L1 + Galileo E1 + GPS L5 + Galileo E5a receiver we use in our CI system, working with two files in different bands. They need to be captured synchronously and at the same sample frequency.

```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=12500000
GNSS-SDR.use_acquisition_resampler=true
GNSS-SDR.assist_dual_frequency_acq=true

;######### SIGNAL_SOURCE 0 CONFIG ############
SignalSource.implementation=Multichannel_File_Signal_Source
SignalSource.RF_channels=2
SignalSource.total_channels=2
SignalSource.filename0=/captures/gitlab_test_signals/static/180417100529.A_L1E1_12500000Hz_ichar.dat
SignalSource.filename1=/captures/gitlab_test_signals/static/180417100529.A_L5E5_12500000Hz_ichar.dat
SignalSource.item_type=ibyte
SignalSource.sampling_frequency=12500000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner

SignalConditioner1.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter0.implementation=Ibyte_To_Complex

DataTypeAdapter1.implementation=Ibyte_To_Complex

;######### INPUT_FILTER CONFIG ############
InputFilter0.implementation=Pass_Through
InputFilter0.input_item_type=gr_complex
InputFilter0.output_item_type=gr_complex
InputFilter0.dump=false
InputFilter0.dump_filename=../data/input_filter0.dat

InputFilter1.implementation=Pass_Through
InputFilter1.input_item_type=gr_complex
InputFilter1.output_item_type=gr_complex
InputFilter1.dump=false
InputFilter1.dump_filename=../data/input_filter1.dat

;######### RESAMPLER CONFIG ############
Resampler0.implementation=Pass_Through
Resampler0.item_type=gr_complex

Resampler1.implementation=Pass_Through
Resampler1.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=10
Channels_1B.count=10
Channels_L5.count=10
Channels_5X.count=10

Channels.in_acquisition=5

; Connect channels to sources
;GPS L1 Channels
Channel0.RF_channel_ID=0
Channel1.RF_channel_ID=0
Channel2.RF_channel_ID=0
Channel3.RF_channel_ID=0
Channel4.RF_channel_ID=0
Channel5.RF_channel_ID=0
[... 158 lines cut ...]
TelemetryDecoder_1B.dump=false

TelemetryDecoder_5X.implementation=Galileo_E5a_Telemetry_Decoder
TelemetryDecoder_5X.dump=false

;######### OBSERVABLES CONFIG ############
;#implementation:
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=OFF ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=OFF ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.elevation_mask=10
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=true
PVT.dump_filename=./PVT
```

Tanggeshi, 2020-07-23:
@carlesfernandez Thank you so much! It can run in two band mode now!  But I cannot get Rinex file after it stop normally. I can get Rinex file when I use one file and one band.  Is there any advice to check this problem?

jmfriedt, 2026-02-25:
If I may followup on this issue: when running ``gnss-sdr`` with the configuration file found at
https://github.com/jmfriedt/max2771_fx2lp/blob/main/260219_E1E5_X310_5MSps/File_Galileo_E1E5_ishort.conf
the processing performs well but at the end stop without closing the files, including the RINEX outputs. I see that I must hit 'q' to properly quit, and only then are the RINEX data flushed to disk, adding the remaining lines missing otherwise.
Am I missing an option for the processing to end without manually hitting 'q' which prevents running from a crontab/automated unsupervised processing?
Thanks.