gnss-sdr/gnss-sdr #414: build failure in Ubuntu 20.20

LocutusOfBorg, 2020-07-29:
Hello, this is the build log after importing the new 0.0.13

```
cd /build/gnss-sdr-0.0.12/obj-x86_64-linux-gnu/src/algorithms/acquisition/adapters && /usr/bin/c++  -DARMA_NO_BOUND_CHECKING=1 -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_CONCEPTS -DBOOST_DATE_TIME_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DGNSSSDR_INSTALL_DIR=\"/usr\" -DHAS_STD_SPAN=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGE_FILES -I/build/gnss-sdr-0.0.12/src/core/interfaces -I/build/gnss-sdr-0.0.12/src/algorithms/libs -I/build/gnss-sdr-0.0.12/src/algorithms/acquisition/gnuradio_blocks -I/build/gnss-sdr-0.0.12/src/algorithms/libs/opencl -I/build/gnss-sdr-0.0.12/src/algorithms/acquisition/libs -I/build/gnss-sdr-0.0.12/src/core/receiver -I/build/gnss-sdr-0.0.12/src/algorithms/channel/libs -I/build/gnss-sdr-0.0.12/src/core/system_parameters -isystem /build/gnss-sdr-0.0.12/obj-x86_64-linux-gnu/volk_gnsssdr_module/build/include -isystem /build/gnss-sdr-0.0.12/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/include -isystem /usr/include/glog  -g -O2 -fdebug-prefix-map=/build/gnss-sdr-0.0.12=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -std=c++2a -o CMakeFiles/acquisition_adapters.dir/gps_l1_ca_pcps_acquisition.cc.o -c /build/gnss-sdr-0.0.12/src/algorithms/acquisition/adapters/gps_l1_ca_pcps_acquisition.cc
In file included from /usr/include/boost/format/alt_sstream.hpp:174,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/gnuradio/logger.h:49,
                 from /usr/include/gnuradio/block.h:29,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /usr/include/gnuradio/blocks/file_sink.h:28,
                 from /build/gnss-sdr-0.0.12/src/algorithms/signal_source/adapters/uhd_signal_source.h:25,
                 from /build/gnss-sdr-0.0.12/src/algorithms/signal_source/adapters/uhd_signal_source.cc:20:
/usr/include/boost/format/alt_sstream_impl.hpp: In instantiation of 'boost::io::basic_altstringbuf<Ch, Tr, Alloc>::int_type boost::io::basic_altstringbuf<Ch, Tr, Alloc>::overflow(boost::io::basic_altstringbuf<Ch, Tr, Alloc>::int_type) [with Ch = char; Tr = std::char_traits<char>; Alloc = std::allocator<char>; boost::io::basic_altstringbuf<Ch, Tr, Alloc>::int_type = int]':
/usr/include/boost/format/alt_sstream_impl.hpp:227:9:   required from here
/usr/include/boost/format/alt_sstream_impl.hpp:261:45: error: no matching function for call to 'std::allocator<char>::allocate(std::size_t&, char*)'
  261 |                     newptr = alloc_.allocate(new_size, is_allocated_? oldptr : 0);
      |                              ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/10/string:41,
                 from /usr/include/c++/10/stdexcept:39,
                 from /usr/include/c++/10/system_error:41,
                 from /usr/include/c++/10/bits/std_mutex.h:39,
                 from /usr/include/c++/10/condition_variable:40,
                 from /build/gnss-sdr-0.0.12/src/core/receiver/concurrent_queue.h:24,
                 from /build/gnss-sdr-0.0.12/src/algorithms/signal_source/adapters/uhd_signal_source.h:23,
                 from /build/gnss-sdr-0.0.12/src/algorithms/signal_source/adapters/uhd_signal_source.cc:20:
/usr/include/c++/10/bits/allocator.h:167:7: note: candidate: 'constexpr _Tp* std::allocator< <template-parameter-1-1> >::allocate(std::size_t) [with _Tp = char; std::size_t = long unsigned int]'
  167 |       allocate(size_t __n)
      |       ^~~~~~~~
/usr/include/c++/10/bits/allocator.h:167:7: note:   candidate expects 1 argument, 2 provided
[ 97%] Building CXX object src/algorithms/acquisition/adapters/CMakeFiles/acquisition_adapters.dir/gps_l1_ca_pcps_acquisition_fine_doppler.cc.o

```

Looks like something is still missing?

LocutusOfBorg, 2020-07-29:
this might even be a bug in boost, fixed with this commit
https://github.com/boostorg/format/commit/5884c3d8140cdcd2d1bef374fcc216244d5faaa5

carlesfernandez, 2020-07-29:
Hi @LocutusOfBorg ,

This should now be fixed both in the `master` and the `next` branches. Thanks for raising this issue!

LocutusOfBorg, 2020-07-29:
So, thanks! I'll sponsor the package in mentors once the build is ok here
http://debomatic-amd64.debian.net/distribution#unstable/gnss-sdr/0.0.13-1/buildlog

closed 2020-07-29 by LocutusOfBorg

carlesfernandez, 2020-07-29:
Thanks @LocutusOfBorg !