gnss-sdr/gnss-sdr #417: FIFOs in FPGA for tracking

noelpedro, 2020-08-26:
Hi guys how big are the FIFOs that exert back pressure to ADC data relative to vector length of samples during tracking?

mmajoral, 2021-02-03:
Hi noelpedro,
sorry for my late reply.
Currently we use fixed buffer lengths. Each frequency band has a FIFO with a fixed length of 262144 samples. On top of that the input of each tracking multicorrelator has an independent FIFO with a length of 32768 samples. 
Because of the fact that we use fixed lengths, the length of the signal in seconds that is stored in the FIFOs depends on the sampling frequency. For instance, when using a sampling frequency of 12.5 MSps, the main FIFO can store 20 ms worth of data, and the tracking multicorrelator FIFOs can store 2.6 ms worth of data. If the main processor is temporarily busy, these FIFOs prevent the loss of data by temporarily storing the received signal samples. Using a fixed FIFO length, if you raise the sampling frequency then the SW has less time to react to the HW events, so the number of channels that you can simultaneously track depends not only on the computational power of the main processor but also on the sampling frequency.

noelpedro, 2021-02-03:
Hi Marc,

thanks for the reply.

The main 256KB channel fifo begins back pressure as soon as acquisition starts? Or sometime before?

Thanks

mmajoral, 2021-02-03:
The acquisition does not exert back pressure. It takes some time to perform a doppler sweep. If the acquisition exerted back pressure then the receiver would not be able to run on real time because the FIFOs would overflow. The acquisition has an internal buffer. When the SW orders the acquisition HW acceleration to acquire a certain satellite, the acquisition module captures the samples it needs and stores the samples in its internal buffers. Then it performs the acquisition algorithm on these samples, but it does not block the FIFOs. 
The tracking HW accelerators do exert back pressure after the integration period when they are waiting for the SW to update the tracking parameters.

closed 2021-02-03 by noelpedro