gnss-sdr/gnss-sdr #421: unable to get position for glonass data in gnss-sdr

marxwolf, 2020-09-02:
I use USRP X310, linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de, gnss-sdr version 0.0.13.git-next-df2f84dfb

The receiver configuration is like this:
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=192.168.10.2 ;

SignalSource.item_type=cshort
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.subdevice=A:0  ; <- Can be A:0 or B:0
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=cshort

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Fir_Filter
InputFilter.input_item_type=cshort
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2

InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0

InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0

InputFilter.band1_error=1.0
InputFilter.band2_error=1.0

InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=4000000
InputFilter.IF=0

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

The configuration file is just from the [configuration webpage](https://gnss-sdr.org/conf/). And I only modified the SignalSource.device_address to fit my device. 

The output result when I run is 

hu@hu:~/rfid/gps$ gnss-sdr --config_file=./my_gps_receiver.conf
Initializing GNSS-SDR v0.0.13.git-next-df2f84dfb ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929b
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1292 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1311 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Sampling Rate for the USRP device: 4e+06 [sps]...
UHD RF CHANNEL #0 SETTINGS
Actual USRP center freq.: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0.0101323 [Hz]...
Actual daughterboard gain set to: 37.5 dB...
Setting RF bandpass filter bandwidth to: 2e+06 [Hz]...
Check for front-end RXLO: locked is ... Locked
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 26 (Block IIF)
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 29 (Block IIR-M)
Current receiver time: 11 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 30 (Block IIF)
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
Current receiver time: 15 s
Loss of lock in channel 5!
Current receiver time: 16 s
Loss of lock in channel 6!
Current receiver time: 17 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 11 (Block IIR)
Current receiver time: 22 s
Loss of lock in channel 2!
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 23 (Block III)
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
Current receiver time: 31 s
Current receiver time: 32 s
Loss of lock in channel 2!
Current receiver time: 33 s
Loss of lock in channel 6!
Current receiver time: 34 s
Current receiver time: 35 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 31 (Block IIR-M)
Current receiver time: 36 s
Current receiver time: 37 s
Loss of lock in channel 1!
Current receiver time: 38 s
Current receiver time: 39 s
Current receiver time: 40 s
Loss of lock in channel 0!
Current receiver time: 41 s
Loss of lock in channel 3!
Current receiver time: 42 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 43 s
Current receiver time: 44 s
Current receiver time: 45 s
Current receiver time: 46 s
Current receiver time: 47 s
Loss of lock in channel 6!
Current receiver time: 48 s
Current receiver time: 49 s
Current receiver time: 50 s
Current receiver time: 51 s
Current receiver time: 52 s
Current receiver time: 53 s
Current receiver time: 54 s
Current receiver time: 55 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 24 (Block IIF)
Current receiver time: 56 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 57 s
Current receiver time: 58 s
Current receiver time: 59 s
Current receiver time: 1 min 0 s
Current receiver time: 1 min 1 s
Loss of lock in channel 4!
Current receiver time: 1 min 2 s
Current receiver time: 1 min 3 s
Current receiver time: 1 min 4 s
Current receiver time: 1 min 5 s
Current receiver time: 1 min 6 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 1 min 7 s
Current receiver time: 1 min 8 s
Current receiver time: 1 min 9 s
Current receiver time: 1 min 10 s
Current receiver time: 1 min 11 s
q
Quit keystroke order received, stopping GNSS-SDR !!
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 71.1594 [seconds]
GNSS-SDR program ended.

No position signal appears. And I donot know why? Who can help me?

marxwolf, 2020-09-02:
By the way, why the GPS indicator led in the front panel is always off. Since I find the in [NI websit](http://zone.ni.com/reference/en-XX/help/373380J-01/usrphelp/2944_front_panel/)

GPS | Indicates whether the GPSDO is locked: OFF—There is no GPSDO or the device or the GPS is not locked. GREEN—The GPSDO is locked.
-- | --

marxwolf, 2020-09-04:
@carlesfernandez I find for USRP x310, there is a GPS connection for GPS antenna, then which subdevice should I specify?

acebrianjuan, 2020-09-05:
Hi @marxwolf,

> I find for USRP x310, there is a GPS connection for GPS antenna, then which subdevice should I specify?

A similar question regarding the X310 USRP was asked not long ago. You can find it here https://github.com/gnss-sdr/gnss-sdr/issues/376

Regards,
Álvaro

marxwolf, 2020-09-05:
@acebrianjuan Thanks, but my problem is kind of different.
I do have a GPSDO and the 5V  Active GPS Antenna for USRP X300/X310. 
I plugged the active gps antenna in the back panel with no other external power supply.
[image]

I still set `SignalSource.subdevice=A:0`, also `SignalSource.clock_source=gpsdo` after read #376 , but it does not work.
Can you help me to figure it out?

Thank you.

acebrianjuan, 2020-09-06:
@marxwolf 

> I do have a GPSDO and the 5V Active GPS Antenna for USRP X300/X310.
> I plugged the active gps antenna in the back panel with no other external power supply.

I think I understand what the problem is. You need 2 GPS antennas: one for the internal GPSDO and the other one to get the actual position fix (with GNSS-SDR).

This is the setup you need:

- Connect first GPS antenna with a bas-T<sup>[[1]](#footnote1)</sup> to "RF0-RX2" connector in the front panel.
- Connect second GPS antenna directly (without a bias-T) to "GPS ANT" connector in the rear panel<sup>[[2]](#footnote2)</sup>.

Try it and let us know if it works for you.

Regards,
Álvaro

---
<a name="footnote1">1</a>: You need to connect a bias-T between the USRP and the active GPS antenna to inject power to the antenna's low noise amplifier (LNA), as explained in the documentation page https://gnss-sdr.org/conf/

<a name="footnote2">2</a>: See step 4 in https://files.ettus.com/manual/page_usrp_x3x0.html#x3x0_getting_started_assembling

marxwolf, 2020-09-06:
@acebrianjuan It is wired, why two GPS antenna needed? Does the rear panel with "GPS ANT" only provide clock information without position information? Also it is easier to use the rear panel, since it provides the power. 
I do not understand why.

marxwolf, 2020-09-06:
By the way, it is not easy to get bias-T and external power supply. If I can use the rear panel "GPS ANT" connector, it would be better.

acebrianjuan, 2020-09-06:
@marxwolf 

I have done a little bit more research about the X310 USRP and I realize that my last [comment](https://github.com/gnss-sdr/gnss-sdr/issues/421#issuecomment-687787981) was a little bit rushed and probably misleading. Apologies for that. I hope the following answers clear all your doubts.

Long story short:

- You only need to connect the GPS antenna with a bas-T to "RF0-RX2" (or "RF1-RX2") connector in the front panel.

Yes, you need the bias-T and you can't use the "GPS ANT" connector to circumvent this.

I try to explain why in my answers below.

> It is weird, why two GPS antennas needed?

You are right. It's not strictly necessary to use two GPS antennas. You can do fine with just one antenna. It all depends on your specific use case and hardware setup.

The X310 USRP internal oscillator has a stability of 2.5 ppm. As Carles explained [here](https://github.com/gnss-sdr/gnss-sdr/issues/376#issuecomment-628521168), a clock stability of 1 ppm or less is recommended for GNSS signal processing. This is why you need to either connect a high-stability external 10 MHz reference clock to the "REF IN" input or install a [GPS-disciplined OCXO](https://www.ettus.com/all-products/gpsdo-mini/) (which is an optional feature, not all X310 USRP's include it).

If you have the GPS-disciplined OCXO installed (like you do), you can use it without locking it to the satellite constellation (i.e. without connecting a GPS antenna to the "GPS ANT" port) and that will still provide a 10 MHz clock signal with frequency accuracies between 20 and 75 ppb, which is more than enough.

You can find more details here: https://kb.ettus.com/GPSDO_Selection_Guide

> Also it is easier to use the rear panel, since it provides the power. I do not understand why.
> It is not easy to get bias-T and external power supply. If I can use the rear panel "GPS ANT" connector, it would be better.

Unfortunately, the X310 USRP was not designed to work that way. As you can see in the schematic below, the "GPS ANT" port is not connected to the analog-to-digital converters (ADC). That's why you can't use the "GPS ANT" port as an input for the DSP.

You must use either "RF0-RX2" or "RF1-RX2".

[image]
You can find this schematic in the [X300/X310 Spec Sheet](https://www.ettus.com/wp-content/uploads/2019/01/X300_X310_Spec_Sheet.pdf).

Regards,
Álvaro

marxwolf, 2020-09-06:
@acebrianjuan Thanks for your generous help. I totally understand what do you mean. I will try to use a bias-T and to see the results. Then I will post the result here.

marxwolf, 2020-09-07:
I do more search on the internet. [Ettus official](https://www.ettus.com/all-products/gps-ant-5v/) says the  5V active GPS antenna intended for use with GPSDO. And this GPSDO costs more than $1, 000. I do not think this GPSDO just provides the clock reference.  The [module spersification](https://www.ettus.com/all-products/GPSDO-MINI/) says GPS Receiver | 50 Channels, Mobile, WAAS, EGNOS, MSAS capable. So is it possible to use rear panel "GPS ANT" connector, GPSDO, active GPS antenna, these three things to receive and decode the GPS signal and get the position information? Maybe not using the gnss-sdr code.

acebrianjuan, 2020-09-07:
> So is it possible to use rear panel "GPS ANT" connector, GPSDO, active GPS antenna, these three things to receive and decode the GPS signal and get the position information? Maybe not using the gnss-sdr code.

Sure! If you only wish to know your location (without running GNSS-SDR) you can read the GPS position using the sensor
API.

Read the following links:
- http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2015-December/045077.html
- https://kb.ettus.com/X300/X310#GPSDO 

Regards,
Álvaro