gnss-sdr/gnss-sdr #427: front-end-cal returns errors

Hermite123456, 2020-09-19:
Having an issue here:

~/gnss-sdr/conf$ sudo gnss-sdr --config_file=./front-end-cal.conf
Initializing GNSS-SDR v0.0.13.git-next-3daa9259e ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 40.2 dB...
Actual Bandwidth: 0 [Hz]...
E0918 23:30:47.402254  2561 gnss_flowgraph.cc:374] itemsize mismatch: sample_counter0:0 using 152, copy1:0 using 8
E0918 23:30:47.402407  2561 control_thread.cc:306] Unable to connect flowgraph
Total GNSS-SDR run time: 0.0530226 [seconds]
GNSS-SDR program ended.

Lokks like it cannot resolve the Actual RX rate & Freq. Any thoughts?

acebrianjuan, 2020-09-21:
Hi @Hermite123456 

I believe the cause of the problem is that you are passing the [front-end-cal.conf](https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/front-end-cal.conf) file to the **gnss-sdr** executable instead of the **front-end-cal** executable.

At the beginning of the front-end-cal.conf file you can find some useful comments on how to use the file. You will see that said configuration file is meant to be invoked by front-end-cal.

https://github.com/gnss-sdr/gnss-sdr/blob/3daa9259e835a170406507a36eb96dce3ff0cbec/conf/front-end-cal.conf#L6-L9

So, instead of doing:

> Having an issue here:
> 
> ~/gnss-sdr/conf$ sudo gnss-sdr --config_file=./front-end-cal.conf

do this:

```
$ front-end-cal --config_file=./front-end-cal.conf
```

NOTE: before running the calibration, make sure that you have a working internet connection as the program needs to download data from a [SUPL server](https://gnss-sdr.org/docs/sp-blocks/global-parameters/#assisted-gnss-via-supl-v10).

If everything is OK, you should see an output similar to the one below (this is my output using an RTL-SDR V.3 dongle):

```
alvaro@msi:~/git/gnss-sdr/install$ ./front-end-cal --config_file=../conf/front-end-cal.conf 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing... Please wait.
Logging will be done at /tmp
Use front-end-cal --log_dir=/path/to/log to change that.
Trying to read ephemeris from SUPL server...
SUPL: Trying to read GPS ephemeris from SUPL server...
SUPL: Received Ephemeris for GPS SV 1
SUPL: Received Ephemeris for GPS SV 3
SUPL: Received Ephemeris for GPS SV 4
SUPL: Received Ephemeris for GPS SV 8
SUPL: Received Ephemeris for GPS SV 11
SUPL: Received Ephemeris for GPS SV 17
SUPL: Received Ephemeris for GPS SV 19
SUPL: Received Ephemeris for GPS SV 21
SUPL: Received Ephemeris for GPS SV 22
SUPL: Received Ephemeris for GPS SV 28
SUPL: Received Ephemeris for GPS SV 31
SUPL: Received Ephemeris for GPS SV 32
SUPL: Received Almanac for GPS SV 1
SUPL: Received Almanac for GPS SV 2
SUPL: Received Almanac for GPS SV 3
SUPL: Received Almanac for GPS SV 4
SUPL: Received Almanac for GPS SV 5
SUPL: Received Almanac for GPS SV 6
SUPL: Received Almanac for GPS SV 7
SUPL: Received Almanac for GPS SV 8
SUPL: Received Almanac for GPS SV 9
SUPL: Received Almanac for GPS SV 10
SUPL: Received Almanac for GPS SV 11
SUPL: Received Almanac for GPS SV 12
SUPL: Received Almanac for GPS SV 13
SUPL: Received Almanac for GPS SV 15
SUPL: Received Almanac for GPS SV 16
SUPL: Received Almanac for GPS SV 17
SUPL: Received Almanac for GPS SV 18
SUPL: Received Almanac for GPS SV 19
SUPL: Received Almanac for GPS SV 20
SUPL: Received Almanac for GPS SV 21
SUPL: Received Almanac for GPS SV 22
SUPL: Received Almanac for GPS SV 23
SUPL: Received Almanac for GPS SV 24
SUPL: Received Almanac for GPS SV 25
SUPL: Received Almanac for GPS SV 26
[... 92 lines cut ...]
  15   -1789530710.47  (Eph not found)
  16   -1789578710.47  (Eph not found)
  17   2505297585.53   3024.26
  18   2505331085.53  (Eph not found)
  19   -1789563710.47   3776.63
  20   2505385085.53  (Eph not found)
  21   2505372085.53   -2210.94
  22   2505348585.53   -348.84
  23   -1789571610.47  (Eph not found)
  24   -1789532710.47  (Eph not found)
  25   2505308585.53  (Eph not found)
  26   -1789513210.47  (Eph not found)
  27   2505313085.53  (Eph not found)
  28   -1789533710.47   -1099.27
  29   -1789570410.47  (Eph not found)
  30   2505370085.53  (Eph not found)
  31   2505323585.53   2175.89
  32   -1789532210.47   -2582.30
GNSS-SDR Front-end calibration program ended.
```

I hope this is helpful to you.

Regards,
Álvaro

acebrianjuan, 2020-09-21:
As a side note, I suggest you avoid running gnss-sdr (or any other program) with sudo unless it's strictly necessary in your system/environment.

acebrianjuan, 2020-09-21:
> Actual Bandwidth: 0 [Hz]...

> Lokks like it cannot resolve the Bandwidth or something. Any thoughts?

Regarding your question about the 0 Hz bandwidth reported by the program:

If you take a look at the code in the file [`osmosdr_signal_source.cc`](https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/signal_source/adapters/osmosdr_signal_source.cc), you will see that the bandwidth value comes from a function called `get_bandwidth()`.
https://github.com/gnss-sdr/gnss-sdr/blob/3daa9259e835a170406507a36eb96dce3ff0cbec/src/algorithms/signal_source/adapters/osmosdr_signal_source.cc#L113-L114

Said function belongs to the [`gr-osmosdr`](https://osmocom.org/projects/gr-osmosdr/wiki/GrOsmoSDR) library and is defined in the [`osmosdr/source.h`](https://github.com/osmocom/gr-osmosdr/blob/master/include/osmosdr/source.h) and [`osmosdr/source_iface.h`](https://github.com/osmocom/gr-osmosdr/blob/master/lib/source_iface.h) headers.

As you can see below, the function is `virtual` and it is supposed to return the frontend's bandpass filter bandwidth setting. Notice that the default implementation for this function always returns 0 (as the function is expected to be overridden by each specific frontend driver).
```
  /*!
   * Get the actual bandpass filter setting on the radio frontend.
   * \param chan the channel index 0 to N-1
   * \return the actual filter bandwidth in Hz
   */
  virtual double get_bandwidth( size_t chan = 0 ) { return 0; }
```

I've browsed through the library code and it looks like there is no specialized implementation of the `get_bandwidth()` function for the RTL-SDR source type (see [`rtl_source_c.h`](https://github.com/osmocom/gr-osmosdr/blob/master/lib/rtl/rtl_source_c.h) and [`rtl_source_c.cc`](https://github.com/osmocom/gr-osmosdr/blob/master/lib/rtl/rtl_source_c.cc)). Therefore, the default implementation is used instead.

This probably explains why we always see `Actual Bandwidth: 0 [Hz]...` in the GNSS-SDR logs when using an RTL-SDR dongle.

Regards,
Álvaro

Hermite123456, 2020-09-23:
Thank you very much for this info. 

The link of this address seems offline. Is that normal?

https://gnss-sdr.org/docs/sp-blocks/global-parameters/#assisted-gnss-via-supl-v10