gnss-sdr/gnss-sdr #431: BladeRF 2.0 Micro Setup For GNSS-SDR

AgileEngineeringLLC, 2020-09-27:
Attached is a short memo on setting up the BladeRF 2.0 Micro RF front end for GNSS-SDR. I ran into a number of issues, so perhaps this would be helpful for others that are using the BladeRF 2.0 Micro.

[BladeRF paper, 092420.docx](https://github.com/gnss-sdr/gnss-sdr/files/5289010/BladeRF.paper.092420.docx)

AgileEngineeringLLC, 2020-10-01:
> Very informative memo Don! (@AgileEngineeringLLC <https://github.com/AgileEngineeringLLC>)
>
> Have you considered turning your document into a tutorial in the project's website (gnss-sdr.org <https://gnss-sdr.org/>), very much like the ones for the RTL-SDR <https://gnss-sdr.org/docs/tutorials/gnss-sdr-operation-realtek-rtl2832u-usb-dongle-dvb-t-receiver/> and SiGe GN3S <https://gnss-sdr.org/docs/tutorials/sige-gn3s-sampler-v2-usb-front-end/> front-ends? I think it could be easily done with the information you have gathered.
>
> Would you agree to this?
>
> I could dedicate some spare time to do the job.
>
> Regards,
> Álvaro

Álvaro,

Sure. I’ll take a look at the RTL-SDR and the SiGe tutorials and come up some ideas.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

AgileEngineeringLLC, 2020-10-03:
GNSS-SDR Community,

If you’ve ever wondered what a “good signal” means for operating GNSS-SDR, below is a memo on my experiences to date. Would welcome any feedback on what others have found!

P.S. Alvaro, I have ~10 instructional memos now on my linkedIn site on using GNSS-SDR. Would welcome migrating any of them over to the GNSS-SDR site, just let me know how or what is needed from me.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

acebrianjuan, 2020-10-05:
Don,

> If you’ve ever wondered what a “good signal” means for operating GNSS-SDR, below is a memo on my experiences to date. Would welcome any feedback on what others have found!

Did you mean to attach your memo in your last comment? I would like to take a look at it but it looks like it wasn't picked up. I believe GitHub ignores attachments when you send a comment reply via email.

> P.S. Alvaro, I have ~10 instructional memos now on my linkedIn site on using GNSS-SDR. Would welcome migrating any of them over to the GNSS-SDR site, just let me know how or what is needed from me.

That's great! My suggestion would be to start out by writing a tutorial about the operation of GNSS-SDR with the BladeRF 2.0 Micro front-end for GPS L1 and see how it works out. Writing a tutorial can be a bit of work, but it should be a lot easier using the information you gathered in you memo.

The source code of the [gnss-sdr.org](https://gnss-sdr.org/) website can be found at: https://github.com/gnss-sdr/geniuss-place.
Instructions on how to make a contribution (like adding a new tutorial) are detailed in the [CONTRIBUTING.md](https://github.com/gnss-sdr/geniuss-place/blob/master/CONTRIBUTING.md) file.

I can open a draft pull request in the [geniuss-place](https://github.com/gnss-sdr/geniuss-place) repository where we can both push changes (as time allows, no need to rush :wink:) until we have something that @carlesfernandez can review for merging. For this collaboration mechanism to work I can either grant you push access permissions on [my fork](https://github.com/acebrianjuan/geniuss-place) (by adding you as a contributor) or receive pull requests from you to [my fork](https://github.com/acebrianjuan/geniuss-place) as explained [here](https://stackoverflow.com/a/39807499).

This would make a very nice [Hackoberfest](https://hacktoberfest.digitalocean.com/) contribution if it works out well.
What do you say Don?

Regards,
Álvaro

AgileEngineeringLLC, 2020-10-05:
Alvaro,

Sure, I can give it a try. I’m not currently a heavy github user, and am not familiar with uploading files, but guessing it can’t be too hard once I get it working.

I’ll reload the memo to the Issues page, as github mush be grabbing it without the attachment.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

I can open a draft pull request in the geniuss-place <https://github.com/gnss-sdr/geniuss-place> repository where we can both push changes (as time allows, no need to rush 😉) until we have something that @carlesfernandez <https://github.com/carlesfernandez> can review for merging. For this collaboration mechanism to work I can either grant you push access permissions on my fork <https://github.com/acebrianjuan/geniuss-place> (by adding you as a contributor) or receive pull requests from you to my fork <https://github.com/acebrianjuan/geniuss-place> as explained here <https://stackoverflow.com/a/39807499>.

lenhart, 2021-02-24:
Hey @AgileEngineeringLLC, 
thanks for the writeup! (A tutorial + information on the respective sites would of course be very cool, as I found this issue just by chance in the debugging process)

I have a question and some remarks to it already before testing out the configs myself.

- Did you write the frontend for the gnss-sdr-monitor yourself or is this already available somewhere?
- sudo privs / udev-rules: it suffices to add the current user to the plugdev group (+ reboot. logout/login theoretically suffices, but i had instances when it didn't)
- Bias-tee issue switching on/off: my supervisor gave me a LNA, as it has a LED on it with which the status can be checked. It seems as if you have some too. I found out that adding biastee=rx (or =1) in the osmocom_args in gnss-sdr enables it too. This is not really documented and contrasts the commonly used bias=1 parameter, but stll works.

Cheers

AgileEngineeringLLC, 2021-02-24:
You’ll find more write-ups on my linkedIn site as well. Unfortunately I’m not able to make the time to create full-blown tutorials and post to the gnss-sdr site, but I’ll be glad to provide my memos for anyone that’s up for doing it!

No, I did not write the monitor program. It’s a great GUI tho!! Please search github gnss-sdr-monitor for the source code.

I use a variety of LNAs and biastee boards when needed. I show some of them in my write-ups.

Good luck!

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

acebrianjuan, 2021-02-27:
Hi @lenhart 

>     * Did you write the frontend for the gnss-sdr-monitor yourself or is this already available somewhere?

I wrote the it, and you can find it here: https://github.com/acebrianjuan/gnss-sdr-monitor

Álvaro

lenhart, 2021-02-27:
Hey @acebrianjuan 
thank you for the good work! The GUI looks really nice (any plans to mention it in the monitor tutorial on the gnss-sdr page? :-) )

I am still struggeling to set everything up, but I'll try some more before bugging you with it ;-) 
Cheers!

acebrianjuan, 2021-02-28:
Hi @lenhart,

The GUI is mentioned in the project's website in a few places :)

- https://gnss-sdr.org/docs/sp-blocks/monitor/#monitor-listeners
- https://gnss-sdr.org/conf/#hackrf-and-gnss-sdr-in-action (alongside Don's video).
- https://gnss-sdr.org/docs/sp-blocks/pvt/#implementation-rtklib_pvt (in the `enable_protobuf` description)

We probably need to write a new set of tutorials to reflect the newly introduced monitoring capabilities (Acquisition, Tracking and Ephemeris streams) so this will be a good opportunity to add any other future references to the GUI.

I hope you are making good progress.

Thank you.
Álvaro

lenhart, 2021-11-29:
Hi Don, (@AgileEngineeringLLC )

I am coming back to your writeup because we are currently unable to get gnss-sdr to work with the bladeRFs we have. 
Can you maybe post your entire gnss-sdr config file you used for the writeup? We experience a constant loss of lock and are wondering if some tracking parameters should be changed, or if the key is the clock disciplining is most important (which you did too). Any help or anecdotes what made a difference for you, or how you went about getting a proper lock is greatly appreciated.
The situation is quite frustrating because we know that it worked before. But if it was just pure luck then, or if the cold weather is messing with the equipment is up to guesses..

(at Álvaro: forgot to answer then, but yes, afterwards I found it in the documentation too. Probably didn't look close enough and missed the introduction. Thanks anyhow!)

Best regards
Malte

AgileEngineeringLLC, 2021-11-29:
Malte,

I haven’t run GNSS-SDR in quite a while now, but attached are a few conf files I found.

Also, it appears I may have been configuring the bladeRF first by running a shell script (see the sh file). So you might check it out as well.

The last couple of years I’ve been running my own GPS code in C, and I’m using my own modified bladeRF API code also written in C. 

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

Hi Don, ***@***.*** <https://github.com/AgileEngineeringLLC> )