gnss-sdr/gnss-sdr #432: Build Problem Using -DENABLE_CUDA=ON

AgileEngineeringLLC, 2020-09-29:
I installed a GeForce GTX RTX 2060 (GPU) board, and am trying to build GNSS-SDR with the -DENABLE_CUDA=ON flag. Make gets 83% done and then gives me the following error (see screen capture below for more details):

nvcc fatal: Unknown option ‘Wno-psabi’

I’m using Ubuntu 20.04, and I have the CUDA driver and Toolkit 11.0 installed (which appears to be the latest version that works on Ubuntu). I tested CUDA functionality using Python Numba, so appears that the CUDA board is working properly.

Any suggestions on how I might get past this error?

Thanks!

[image]

carlesfernandez, 2020-09-29:
Is this still happening in the `next` branch?

carlesfernandez, 2020-09-29:
A quick workaround for `master` is to remove this line:

https://github.com/gnss-sdr/gnss-sdr/blob/0b1befd1280b6df4c7b3fa282bafc6afb60cf5cd/CMakeLists.txt#L2974

AgileEngineeringLLC, 2020-09-29:
Yes, I’m using next.

Thx, I will try removing this line!

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

https://github.com/gnss-sdr/gnss-sdr/blob/0b1befd1280b6df4c7b3fa282bafc6afb60cf5cd/CMakeLists.txt#L2974 <https://github.com/gnss-sdr/gnss-sdr/blob/0b1befd1280b6df4c7b3fa282bafc6afb60cf5cd/CMakeLists.txt#L2974>

AgileEngineeringLLC, 2020-09-29:
Carles,

Using the next branch allowed it to make. But now when I use

Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking_GPU
It appears to be running with GPUs now, but can’t obtain lock (see attached below).

If I remove -GPU, everything works great. Otherwise same exact conf file.

Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking

Are there any other parameters that must be tweaked to get lock when enabling GPU tracking?

Thanks much! 

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

AgileEngineeringLLC, 2020-10-02:
Closing this as I was able to build using the "next" version.

closed 2020-10-02 by AgileEngineeringLLC