gnss-sdr/gnss-sdr #434: Tracking with GPU Not Working

AgileEngineeringLLC, 2020-10-02:
I've been able to get the CUDA option to build, but not able to get it to run. If I run my conf file without "_GPU" 

Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking

Everything runs great. If I turn on GPU tracking:

Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking_GPU

Are there any other parameters that must be set to get lock when enabling GPU tracking?

Here's what the screen looks like with GPU enabled for tracking.
Thx for any suggestions! Don
P.S. I'm running "next," on Ubuntu 20.04.

[image]

alexbvo, 2022-10-13:
Dear @AgileEngineeringLLC , I am having the same issue. Did you progress on the solution? Regards. Alexandre

Rafaykhan12, 2022-12-28:
did this issue get  resolved ? I am unable to run tracking code with gpu 
it says 
Configuration error in Tracking_1C block: implementation Gps_L1_Ca_Dll_Pll_Tracking_GPU_cc not available.

jwmelto, 2022-12-28:
> did this issue get resolved ? I am unable to run tracking code with gpu
> it says
> Configuration error in Tracking_1C block: implementation Gps_L1_Ca_Dll_Pll_Tracking_GPU_cc not available.

That would seem to be a different issue. Did you build with GPU enabled?

jmfriedt, 2023-10-14:
confirmed: replacing GPS_L1_CA_DLL_PLL_Tracking with GPS_L1_CA_DLL_PLL_Tracking_GPU will prevent detecting the signal on a file working perfectly with a GPCPU decoding otherwise.

MathieuFavreau, 2025-09-22:
Sorry for the long delay, I would probably do something basic like timing the execution of GNSS-SDR, I do not need to know in a lot more details than that, any progress on your investigation?