gnss-sdr/gnss-sdr #435: How to Tell If Your Signal Is Good For GNSS-SDR Operation

AgileEngineeringLLC, 2020-10-05:
Here's a brief article on some things to look for in a "good" GNSS signal. I give examples of what a clean, fairly strong signal looks like, contrasted to a signal that's too weak to work in GNSS-SDR.

[Is My Signal Good for GNSS-SDR, 100320.pdf](https://github.com/gnss-sdr/gnss-sdr/files/5329927/Is.My.Signal.Good.for.GNSS-SDR.100320.pdf)

nik13513513, 2021-12-03:
> Here's a brief article on some things to look for in a "good" GNSS signal. I give examples of what a clean, fairly strong signal looks like, contrasted to a signal that's too weak to work in [...]

Thanks for an article!
Where i can watch your scripts for this plots?

AgileEngineeringLLC, 2021-12-03:
Sorry, I’ve been working on my GPS own code the last year rather than using GNSS-SDR. You might check out some of the articles I posted on LinkedIn (www.linkedin.com/in/kellydak <http://www.linkedin.com/in/kellydak>), some of them may have more info.

Don

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059