gnss-sdr/gnss-sdr #438: [ERROR][STREAMER] occured with usrp b2100

air-elephant, 2020-10-11:
Gnss-sdr always threw some error after ran on USRP B210 for a while, like: 

...
Position at 2020-Oct-11 09:08:09.000000 UTC using 7 observations is Lat = 31.264074809 [deg], Long = 121.623262225 [deg], Height = 43.322 [m]
Clk_bias: 4804381.2 [m], Clk_drift: -34.8 [m/s]
PPS-time: 0.999999945 [s], Pos-std: 0.009392949
Velocity: East: 0.000 [m/s], North: 0.000 [m/s], Up = 0.000 [m/s]
Current receiver time: 34 min 0 s
**[ERROR] [STREAMER]** The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
gr::log :WARN: usrp_source0 - USRP Source Block caught rx error code: 15
**[ERROR] [B200]** Got a ctrl packet with unknown SID 3814458221
[ERROR] [B200] Got a ctrl packet with unknown SID 3984268007
[ERROR] [B200] Got a ctrl packet with unknown SID 3966569905
[ERROR] [B200] Got a ctrl packet with unknown SID 3926466003
[ERROR] [B200] Got a ctrl packet with unknown SID 3991014055
[ERROR] [B200] Got a ctrl packet with unknown SID 3914402819
[ERROR] [B200] Got a ctrl packet with unknown SID 850398496
[ERROR] [B200] Got a ctrl packet with unknown SID 312738823
...
[ERROR] [B200] Got a ctrl packet with unknown SID 4244585000
[ERROR] [B200] Got a ctrl packet with unknown SID 2716074008
[ERROR] [B200] Got a ctrl packet with unknown SID 2716139544
terminate called after throwing an instance of 'uhd::assertion_error'
  what():  AssertionError: accum_timeout < _timeout
  in uint64_t radio_ctrl_core_3000_impl::wait_for_ack(bool)
  at /home/share/z00563259/raw/uhd/host/lib/usrp/cores/radio_ctrl_core_3000.cpp:220

Would you give me some favor about this issue?

Here is initialized information output from gnss-sdr:

Initializing GNSS-SDR v0.0.12.git-HEAD-81560f3ed ... Please wait.
Logging will be written at ./log
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-0-unknown
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz...
[INFO] [B200] Actually got clock rate 32.000000 MHz.
Sampling Rate for the USRP device: 4000000.000000 [sps]...
UHD RF CHANNEL #0 SETTINGS
Actual USRP center freq.: 1575420000.000297 [Hz]...
PLL Frequency tune error 0.000297 [Hz]...
Actual daughterboard gain set to: 73.000000 dB...
Setting RF bandpass filter bandwidth to: 2000000.000000 [Hz]...
Check for front-end LO: locked ... is Locked
NMEA files will be stored at ./outputs/
RINEX files will be stored at ./outputs/
/root/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available

**note**: I made some slight modificaiton on the next branch.

acebrianjuan, 2020-10-11:
Hi @air-elephant,

This issue does not seem to concern GNSS-SDR specifically but the USRP Hardware Driver (UHD) and your OS and hardware configuration.

Judging by the error messages, something is corrupting your [radio transport protocol](https://files.ettus.com/manual/page_rtp.html) messages which leads to a timeout scenario (see [radio_ctrl_core_3000.cpp#L199-L221](https://github.com/EttusResearch/uhd/blob/aea0e2de34803d5ea8f25d7cf2fb08f4ab9d43f0/host/lib/usrp/cores/radio_ctrl_core_3000.cpp#L199-L221)).

Please ask for help at [EttusResearch/uhd](https://github.com/EttusResearch/uhd ) and/or refer to issues:

- https://github.com/EttusResearch/uhd/issues/373
- https://github.com/EttusResearch/uhd/issues/222

Regards,
Álvaro

keeplooper, 2021-04-12:
I also use b210 as my hardware. but  when I use GNSS SDR to receive GPS data, I lose the lock almost immediately. Hope to see your configuration

elmaghba, 2021-05-20:
I'm also getting the same error. Did anyone find a solution for the issue?

FayeWong98, 2022-04-28:
@keeplooper 
Hello,I encountered the same problem of losing locks,and I used bias tee,can you give some suggestions?

jiachen1119, 2022-11-01:
> @keeplooper Hello,I encountered the same problem of losing locks,and I used bias tee,can you give some suggestions?

I also encountered the same problem, do you solve it?

K3C3, 2023-10-11:
Did any one find a solution for this issue?