gnss-sdr/gnss-sdr #441: direct_resampler_conditioner_cc.h missing book reference in documentation

cbochs, 2020-10-20:
Hello, this is a minor issue (or I'm missing something obvious here). I'm not sure where put this so please let me know if this is the incorrect place. I've noticed that the documentation in [direct_resampler_conditioner_cc.h](https://github.com/gnss-sdr/gnss-sdr/blob/master/src/algorithms/resampler/gnuradio_blocks/direct_resampler_conditioner_cc.h) indicates that there should be a book reference:

```
/*!
 * \file direct_resampler_conditioner_cc.h
 *
 * \brief Nearest neighborhood resampler with
 *        gr_complex input and gr_complex output
 * \author Luis Esteve, 2011. luis(at)epsilon-formacion.com
 *
 * This block takes in a signal stream and performs direct
 * resampling.
 * The theory behind this block can be found in Chapter 7.5 of
 * the following book.
```

However, there doesn't look to be any further details about the book's name. I'm doing some research into GNSS signal processing and was interested in reading the book that has been referenced here.

Any help would be greatly appreciated!

Thanks,

carlesfernandez, 2020-10-20:
Oh, It seems that the reference has remained blank since the very first commit of that file, 8 years ago. Can you remember that, @Arribas ?

My bet goes for "Understanding Digital Signal Processing", by Richard Lyons.

Anyway, resampling theory is well described in several SIgnal Processing textbooks. A nice wrap-up can be found here:

https://ocw.mit.edu/courses/mechanical-engineering/2-161-signal-processing-continuous-and-discrete-fall-2008/study-materials/updownsampling.pdf 

Hope this helps,
Carles

carlesfernandez, 2020-10-20:
It seems that the original author was @luis-esteve , not @Arribas

cbochs, 2020-10-20:
@carlesfernandez those are both great references, thank you. Looking at the contents of _Understanding Digital Signal Processing_, Ch. 7.5 mentions resampling so I think you are correct.

ZiglioUK, 2020-11-24:
More references:
Fred Harris, Multirate Signal Processing
Michael Rice, Digital Communications, A Discrete-Time Approach
Oppenheim, Schafer, Discrete-Time Signal Processing
Vetterli, Kovacevic, Wavelets and Subband Coding

lots more...

commit e4c0e060 (2020-10-21): Add missing reference in header (Fixes: #441)
closed 2021-01-08 by carlesfernandez