gnss-sdr/gnss-sdr #442: RX offset accuracy ?

jmfriedt, 2020-10-24:
We are struggling to generate a relevant 1-PPS hardware output from the gnss-sdr RX Offset information and I am concerned
with its accuracy.
The setup is as follows:
* an Ettus Research B210 is used as SDR frontend, clocked by an external 10 MHz source provided by a Rohde & Schwarz
SMA100A synthesizer
* the SMA100A synthesizer is running on its internal OCXO referenced against a hydrogen maser (HM) and known to be off frequency by 90 mHz at 10 MHz
* a frequency counter (HP 53132A clocked by a hydrogen maser) is used to monitor the output between our 1-PPS output generated from gnss-sdr and the 1-PPS output of a U-Blox GPS receiver
* our 1-PPS output is generated as a counter in the B210 FPGA (same clock as the ADC clock controlling the AD9361 frontend) running from 1 to 40e6
* the software clock control loop is disabled as we wish to control the SMA100A frequency with the RX Offset information:
PVT.enable_rx_clock_correction=false
[http://jmfriedt.org/fixed-90mHz_bis.png](url)
On this figure, top-right is RX Offset (as defined in gnss-sdr/src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc
 std::cout << "RX clock offset avant : " << d_user_pvt_solver->get_time_offset_s() << "[s]" << '\n';
and top-left is the output of the HP 53132A counter clocked by the hydrogen maser comparing the 1-PPS output with the U-Blox receiver output (checked to be accurate with respect to the French national time reference).

Obviously the two time offsets are not drifting at the same rate with 1us offset between our FPGA-1 PPS over 2h wrt GPS 1-PPS and 30 us of gnss-sdr RX offset wrt GPS time. When running a feedback loop between RX Offset and the clock controlling the B210 (10 MHz input), we get continuous oscillations and unable to lock on a stable 1-PPS offset delay.

jmfriedt, 2020-10-26:
For the record and closing the issue: the problem is not with gnss-sdr and its use of RTKLib but with the selection of the sampling rate we had requested (solution found by Gwenhael Goavec-Merou). We had requested 1.123 MS/s (as a random value adding 10% bandwidth to the theoretical 1.023 MS/s). The sampling rate of the B210 is generated by a fractional PLL in the AD9361 from the 40 MHz input clock: the value of 1.123 MS/s cannot be reached with a fractional PLL. Replacing this random value with 1.25 MS/s solves the drift issue, as shown below.
[image]

carlesfernandez, 2020-10-26:
Dear @jmfriedt I'm happy (and relieved) to see that there was not a major bug in the receiver, I was starting to peruse the code related to this offset. Thank you very much for your detailed report.

jmfriedt, 2020-10-29:
We are meeting another closely related issue. Having controlled the B210 LO to track the GPS time delay information, we get a finely tuned frequency that prevents the generated PPS from drifting with respect to the U-Blox 1-PPS output (top left chart). However when running this setup over a whole night, we observe periodic jumps by 700 or 800 ns of the RX Offset which leads, due to the LO feedback loop control, to our PPS jumping by the same delay with respect to the hardware PPS. To make sure the issue is indeed with the RX Offset and not our feedback loop, we clock both the B210 and the HP53132A time interval counter with the 10 MHz output of a hydrogen maser. Thanks to this "perfect" reference, the frequency (bottom left) remains constant and our 1-PPS does not drift with respect to the hardware 1-PPS and yet RX Offset indeed jumps periodically (top-right). The raw RX offset (bottom right) exhibited huge jumps but since gnss-sdr was displaying seven "O" after the 1-night acquisition, it is probable that we had 7 GNU Radio overflows desynchonizing the stream (actually going from bottom right to top right required the removal of 9 offsets bigger than 20 us).

[image]

We tried tracking the origin of RX Offset and found in src/algorithms/PVTlibs/rtklib_solver.cc that it is the result of (ll.946-957)
`pvt_sol.dtr[2]+pvt_sol.dtr[0]`
The former term is always 0 and the latter exhibits the exact same jumps and looks very much like a feedback loop:
[image][image]
Now pvt_sol.dtr[0] is the solution of a least square solution involving some Kalman filtering and it is beyond our current capability to track the origin of this periodic jump, so any advice on this issue is welcome.
Thanks.

jmfriedt, 2020-11-06:
To close this issue following the excellent advice provided by the authors of gnss-sdr and set the record straight that the issue is NOT with this software:
1/ processing simulated data (gps-sdr-sim at https://github.com/osqzss/gps-sdr-sim) or data collected by the Analog Devices PlutoSDR or with an Ettus Research X310 does NOT show the jumps => the problem is not with gnss-sdr (1) nor with the AD936x frontend (2) nor with libuhd in general (3).
2/ changing the B210 sampling rate from 1.25 MS/s to 2 MS/s has dropper the jump amplitude to 500 ns, demonstrating that the jump value is one sampling period. Furthermore, a detailed analysis of these two datasets (1.25 MS/s and 2 MS/s) demonstrates that this unique sample is dropped every 2^32 samples.
A bug report is being filed on the usrp-users mailing list.

closed 2020-11-06 by jmfriedt

acebrianjuan, 2020-11-09:
> A bug report is being filed on the usrp-users mailing list.

Just posting a link to the bug report here for anyone interested:
http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2020-November/063071.html

Álvaro