gnss-sdr/gnss-sdr #443: HackRF Setup Suggestions for GNSS-SDR

AgileEngineeringLLC, 2020-10-30:
I've been asked a number of times about how to setup a HackRF One SDR for GNSS-SDR. The HackRF is a great choice as a relatively inexpensive SDR, but there's a few setup things that need to be done to get it to work reliably with GNSS-SDR. 

Here are my recommendations: 

1) Don't even try to use the HackRF for GNSS-SDR without an external oscillator. Purchase a more accurate TCXO (like the inexpensive NooElec Tiny TCXO I show in my GNSS-SDR youtube video) or use a more expensive GPSDO (Leo Bednar, for example). You must have a good external timing source or you’ll likely never get lock. A good timing source is an absolute necessity, the one on the HackRF board is pretty poor.

2) Do you have bias tee enabled on your HackRF? You must have that enabled via an “arg” statement in your GNSS-SDR signal source. Otherwise your HackRF will not get a strong enough signal from the GPS antenna and will never lock. 

3) Are you using a good quality GNSS antenna, like an inexpensive uBlox multi-band antenna (about $65 on Sparkfun)? 

4) Does it have a (VERY) clear view of the sky? No blockages from nearby trees or buildings, even down to relatively low elevation?

5) Are you running the GNSS-SDR-Monitor so that you can visually see in real-time if sats are acquiring, and then tracking?

6) Have you added the RF shield (see the NooElec site) to the HackRF board? I think this helped me significantly. See my video on the GNSS-SDR site for how it looks mounted.

If you do all the above, you should get lock with the default HackRF conf file on github. Then you can try tweaking the settings to get it to work even better with your CPU and environment.

Happy HackRF-ing !!

jt3232, 2020-11-08:
Thanks for the suggestions!

If I might consider adding some:

You should consider not using a virtual machine if possible. I used Ubuntu 20.04 on VMWare and it had reliability issues. These consisted of not being able to stop the HackRF One from receiving, resulting in locking it up after every use. Thus, I had to manually reset the HRF1 each time I used it (the problem disappeared when switching to no VM). Also, the software seemed to lose its position fix after a certain amount of time and then could never reobtain the fix, no matter the antenna, Switching from a VM fixed this issue, as it never happened again, using the exact same config files and setup.

Also, if you are a total goof like I am, it is helpful to ground the antenna you might have. If your antenna is magnetic, grounding it drastically improves its performance. I had a metal pizza pan and that seemed to work wonders. 

Also, if using GNU Radio Companion (GRC) to listen to FM signals, a GPS antenna is not designed to receive those signals. However, me unscrewing the antenna half-way out, allowed it to received the FM signals. This made me think it was a faulty antenna, but that was not the case. Make sure your antenna is screwed in all the way no matter what. 

Lastly, the FM signals sound perfectly clear over GRC when not using a VM. When using a VM, the sound was static/skipping about 99% of the time. It was very frustrating. 

Good luck all!

Jaxon

closed 2021-02-08 by AgileEngineeringLLC

cantudo, 2022-12-14:
Hi, I am trying to use gnss-sdr with a hackrf setup, I have a pretty good external clock, but only a passive B25-JA3 GPS antenna, is a passive antenna enough or is an active one really mandatory? thank you

AgileEngineeringLLC, 2022-12-14:
my guess is it wouldnt work, but give it a try and let us know!DonDon KellyTechnologist | EngineerAgile Engineering LLCLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-2853Email: ***@***.*** Orange Leaf CourtHouston, TX 77059

vladisslav2011, 2022-12-14:
Hello.
It may or may not work. It depends on interference conditions and antenna quality. Air gap patch should work without an LNA. Ceramic patch usually requires an LNA.
Just connect it and test.
You may test the antenna with some generic SDR software (Gqrx for example). You'll see many L-Band satellite signals (Inmarsat, Thuraya, BeiDou sarsat payloads) and maybe narrowband GPS signal components (at large enough FFT) if the antenna is good.