gnss-sdr/gnss-sdr #444: Streaming GPS Ephemeris Data Over UDP Capability

jt3232, 2020-11-08:
Hello,

First off, let me say thank you all who have contributed to this incredible open source project.

I want to stream the GPS Ephemeris data that gets saved into the XML file. That data is important for a certain project I am currently working on. I also would like to continue streaming the PVT and GNSS Synchro data.

I have taken a look at the source code, but unfortunately, I have never programmed in C++ or C. I have however programmed a bit in Java, so I can slightly follow along in the code. In my inexperienced opinion, it looks like it should be relatively simple to stream the GPS Ephemeris data, as it is already nicely formatted when it gets archived in the gps_ephemeris.cc file in the System Parameters folder. I think one might be able to copy the sort of code found in the monitor UDP section and throw it somewhere in there.

Before I start trying to mess with this stuff, does anyone have any suggestions? Would it be something you guys want to work on?

Thanks,

Jaxon

jonathanmuller, 2021-01-13:
Hi, 

You know better than I your particular usage but would it not be simpler to get them from an other source ?

- https://cddis.nasa.gov/Data_and_Derived_Products/GNSS/real-time_data.html
- https://cddis.nasa.gov/Data_and_Derived_Products/GNSS/broadcast_ephemeris_data.html

If you can/want to share the broader needs there may hopefully be something already working with the current code

jt3232, 2021-01-26:
Hi Jonathan. Thank you for responding.

Unfortunately, I will, ideally, have several receivers setup across a field which each need to have their own information gathered. I believe specifically they each need a pseudo range from the satellites as well as other ephemeris data. These are not readily capable of being streamed over UDP on the software as of right now. I can see the information that gets written to a file, I wish we could change a few lines of code to also give it the capability to send over UDP.

Thanks,

Jaxon

jonathanmuller, 2021-01-27:
How about the info in ./docs/protobuf/gnss_synchro.proto ? 
I see "pseudorange_m" for example

They seem to be streamed by UDP ( https://gnss-sdr.org/docs/tutorials/monitoring-software-receiver-internal-status/#testing-the-monitoring-client )

And the ephemeris is available on the internet

Would that make it ?

acebrianjuan, 2021-01-27:
Hi @jt3232,

First of all, apologies for my late reply... I did notice this issue when you first published it a few months ago, but it somehow ended up slipping through my fingers.

Did you know GNSS-SDR can output RTCM-104 v3.2 messages?
This standard, among other things, provides the message type 1019 which is dedicated specifically to the streaming of GPS ephemeris data. GNSS-SDR supports the generation of said message type (see [rtcm.h](https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/PVT/libs/rtcm.h) and [rtcm.cc](https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/PVT/libs/rtcm.cc)), so I would definitely consider exploring this option before trying anything else.

You should be aware that the streaming of RTCM messages is done over TCP, so I don't know if this suits you since you were specifically asking about streaming over UDP. It all depends on your particular use case and needs.

Let me know if this helps.
Regards,

Álvaro

---

Links to the documentation:
- https://gnss-sdr.org/docs/sp-blocks/pvt/#rtcm-104
- https://gnss-sdr.org/design-forces/interoperability/#support-of-output-formats

List of RTCM message types:
- https://www.use-snip.com/kb/knowledge-base/rtcm-3-message-list/

carlesfernandez, 2021-02-22:
Hi,

this feature has been added to the `next` branch, please check https://gnss-sdr.org/docs/sp-blocks/pvt/#custom-streaming

jt3232, 2021-02-23:
Hi guys (@acebrianjuan @carlesfernandez). I am sorry for the delayed response as well. I have been getting back into learning the GNSS-SDR program since I hadn't used it for several months. 

First of all, thank you so much for adding the ability to stream the ephemeris data. This is great! I have since learned I need 2 more specific parameters: raw pseudorange and raw phase for each satellite. Am I able to output both the gnss_synchro data as well as the gps_ephemeris data at the same time?

As for using TCP, that requires two-way communication with each receiver at the field, which seems a little less ideal. I am not opposed to the idea yet, but I would be changing how this system works. 

I plan on closing this issue when I get around to recompiling the software and getting it to work.

You guys are fantastic!

Jaxon

acebrianjuan, 2021-02-28:
Hi Jaxon (@jt3232),

> First of all, thank you so much for adding the ability to stream the ephemeris data. This is great! I have since learned I need 2 more specific parameters: raw pseudorange and raw phase for each [...]

Yes, each of the monitoring streams operate independently on different UDP ports, so you can use them simultaneously.
You will need to activate each monitor port in your configuration, as shown in the examples below:

```
;######### OBSERVABLES MONITOR CONFIG ##########
Monitor.enable_monitor=true
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1111
```

```
;######### EPHEMERIS MONITOR CONFIG ############
PVT.enable_monitor_ephemeris=true
PVT.monitor_ephemeris_client_addresses=127.0.0.1
PVT.monitor_ephemeris_udp_port=1112
```

Thank you.
Álvaro

jt3232, 2021-03-01:
Hi guys (@carlesfernandez  @acebrianjuan),

I successfully recompiled the software and am now streaming what appears to be the GPS ephemeris data. I still have to decode it in LabVIEW, but it looks good so far.
Thank you so much for the help and implementing this feature.

I am closing this issue.

Jaxon

closed 2021-03-01 by jt3232