gnss-sdr/gnss-sdr #451: make install duplicates protobuf build dir

erihom, 2020-12-02:
Hi,

I'm building GNSS-SDR on CentOS7 which comes with protobuf version 2.5.0.
Since version >= 3.0.0 is required protobuf gets downloaded, built and installed in the build directory during make.
When running `make DESTDIR=/home/carles install` the destination directory will not only contain the installation files
`/home/carles/user/local/` but also the complete directory tree to the protobuf installation.

Assuming the project is checked out under `/home/carles/projects/gnss-sdr` you get this:

`/home/carles/home/carles/projects/gnss-sdr/build/protobuf-3.12.3/`

`make install` only prints the files getting installed in `/home/carles/usr/local` so this might not have been noticed so far.

Is that the intended behaviour?
Thanks!

carlesfernandez, 2020-12-03:
Well, this is not intended behaviour. Doing `make install` should not install anything related to protobuf. Thanks for spotting this.

erihom, 2020-12-08:
What is it actually used for, is it a runtime or a build time dependency? Or both?

carlesfernandez, 2020-12-09:
Both; the protoc compiler is used in building time, and then the protobuf library is used in runtime. If the library is downloaded by the CMake script, as in your case, the library is linked statically, so you could safely remove that extra path that is created in the install step. Still, this is not nice and I'm still trying to find a solution.

carlesfernandez, 2020-12-10:
Hi @erihom ,

I've just pushed a commit to the `next` branch that should solve this issue. Please let me know if it works for you.

erihom, 2020-12-16:
Hi @carlesfernandez ,

yes it works fine now.
Thanks a lot!

commit 7c057e5c (2020-12-10): Fix duplication of local protobuf build tree when installing (Fixes: #451)
closed 2020-12-16 by erihom