gnss-sdr/gnss-sdr #458: [Question] About the Doppler one should expect

fgallardo, 2020-12-29:
Hi all,

This is more a question rather than an issue, I hope it is fine to place it here.

So, I have been playing with the file 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat available in https://sourceforge.net/projects/gnss-sdr/files/data/ 

My understanding is that the file was recorded back in 2013 with an stand-off Novatel Antenna placed on the roof of the CTTC building (located in 41.27473812858956,1.98748865619449,52.41846634792164).

Ok, as it is a static receiver I would expect it to see Doppler frequencies of +- 5KHz (as per "Fundamentals of Global Positioning System Receivers", James Bao-yen Tsui. Equation 3.10).

Now, I got Doppler frequs. greater than +-5KHz for all GAL sats expect for E11.  The same for GPS. I checked the file 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.txt and it seems like you also got Doppler frequencies higher than +-5KHz.

There is a remark on such book ("Fundamentals of Global Positioning System Receivers". Section 6.15 "EFFECT OF SAMPLING FREQUENCY ACCURACY") on sampling frequency inacuracies that may lead to some Doppler effects, but it points to something in the range of houndreds of Hz, not several KHz.

I am confussed about it. Any idea from somebody around?

Thanks!
Fran

ZiglioUK, 2020-12-29:
That's funny, I think that set is static so dopplers should be within -/+3.5khz.
It could be due to a local oscillator offset.
What's the range of those dopplers?
As long as they're contained within a 7khz range, there should be no problem.

fgallardo, 2020-12-29:
Hi @ZiglioUK ,

If I take for instance E19, from the run I did:
26640 gnss_flowgraph.c, c:1326] Channel 2 ACQ SUCCESS satellite Galileo PRN E19 (Block IOV-FM3)												
26656 dll_pll_veml_tracking.cc:1617] Number of samples between Acquisition and Tracking = 209549 ( 0.0523873 s)												
 26656 dll_pll_veml_tracking.cc:1618] **PULL-IN Doppler [Hz] = 8290**. PULL-IN Code Phase [samples] = 5627

Or for instance, by taking another example in this case from 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.txt:
Tracking CH 3: Satellite GPS PRN 20 (Block IIR), **Doppler=8335.03125** [Hz] CN0 = 46.49216079711914 [dB-Hz]

ZiglioUK, 2020-12-29:
Well, you would have to acquire more satellites to see the range.
Perhaps the doppler window search or step parameters are set wrong.
What are your acquisition settings?
I'll check this file later today.

fgallardo, 2020-12-29:
I tried several settings and to me, the results seem consistent with those high Doppler to be "real" (from the Rx point of view, I mean). (i.e. if the Doppler span in the Search Space is reduced then the satellites are not acquired (those with a high Doppler value that falls out of the span) and if the resolution is increased the Doppler is still high and close to the original values reported).

I uploaded the Galileo config file I used, should you want to use it:
https://drive.google.com/file/d/1UUypau_WN2qkUpSrMHcy0r9mGpxTX5Mk/view?usp=sharing

As per the config, in the GPS case and results (Galileo values are below):

--------------------   GPS ---------------------
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250

Result:
I1229 14:21:49.687394 35228 pcps_acquisition.cc:385] positive acquisition, satellite G 1, sample_stamp 9008, test statistics value 0.0212765, test statistics threshold 0.008, code phase 2978, doppler **7000** , magnitude 0, input signal power 29591.4, Assist doppler_center 0
I1229 14:21:49.691675 35223 pcps_acquisition.cc:385] positive acquisition, satellite G 11, sample_stamp 185783, test statistics value 0.0202418, test statistics threshold 0.008, code phase 2569, doppler **5500**, magnitude 0, input signal power 30059.2, Assist doppler_center 0
I1229 14:21:49.721410 35205 pcps_acquisition.cc:385] positive acquisition, satellite G 17, sample_stamp 1352120, test statistics value 0.0120266, test statistics threshold 0.008, code phase 273, doppler **9750**, magnitude 0, input signal power 29362.9, Assist doppler_center 0
I1229 14:21:49.736287 35220 pcps_acquisition.cc:385] positive acquisition, satellite G 20, sample_stamp 1872464, test statistics value 0.0140558, test statistics threshold 0.008, code phase 776, doppler **8250**, magnitude 0, input signal power 30125.3, Assist doppler_center 0
I1229 14:21:49.796888 35199 pcps_acquisition.cc:385] positive acquisition, satellite G 32, sample_stamp 3850408, test statistics value 0.017943, test statistics threshold 0.008, code phase 1266, doppler **6500**, magnitude 0, input signal power 30721.9, Assist doppler_center 0

--------------------  GALILEO --------------------  
Acquisition_1B.doppler_max=15000
Acquisition_1B.doppler_step=125

NOTE: By 2013, only four GAL satellites were available, hence all results that are not related with one of the following sats can be already disregarded: E11, E12, E19, E20.

I1229 14:11:50.714798 34735 pcps_quicksync_acquisition_cc.cc:495] positive acquisition
I1229 14:11:50.714805 34735 pcps_quicksync_acquisition_cc.cc:496] satellite E 12
I1229 14:11:50.714807 34735 pcps_quicksync_acquisition_cc.cc:497] sample_stamp 2848000
I1229 14:11:50.714808 34735 pcps_quicksync_acquisition_cc.cc:498] test statistics value 0.104739
I1229 14:11:50.714839 34735 pcps_quicksync_acquisition_cc.cc:499] test statistics threshold 0.05
I1229 14:11:50.714843 34735 pcps_quicksync_acquisition_cc.cc:500] folding factor 2
I1229 14:11:50.714844 34735 pcps_quicksync_acquisition_cc.cc:501] possible delay correlation output
I1229 14:11:50.714845 34735 pcps_quicksync_acquisition_cc.cc:504] 1307			6.66308e+10
I1229 14:11:50.714848 34735 pcps_quicksync_acquisition_cc.cc:504] 9307			2.54114e+09
I1229 14:11:50.714850 34735 pcps_quicksync_acquisition_cc.cc:506] code phase 1307
I1229 14:11:50.714852 34735 pcps_quicksync_acquisition_cc.cc:507] doppler **8500**
I1229 14:11:50.714854 34735 pcps_quicksync_acquisition_cc.cc:508] magnitude folded 3164.99
I1229 14:11:50.714857 34735 pcps_quicksync_acquisition_cc.cc:509] input signal power 30218
--
I1229 14:11:50.930491 34729 pcps_quicksync_acquisition_cc.cc:495] positive acquisition
I1229 14:11:50.930500 34729 pcps_quicksync_acquisition_cc.cc:496] satellite E 19
I1229 14:11:50.930500 34729 pcps_quicksync_acquisition_cc.cc:497] sample_stamp 5216000
I1229 14:11:50.930501 34729 pcps_quicksync_acquisition_cc.cc:498] test statistics value 0.0577231
I1229 14:11:50.930510 34729 pcps_quicksync_acquisition_cc.cc:499] test statistics threshold 0.05
I1229 14:11:50.930511 34729 pcps_quicksync_acquisition_cc.cc:500] folding factor 2
I1229 14:11:50.930513 34729 pcps_quicksync_acquisition_cc.cc:501] possible delay correlation output
I1229 14:11:50.930514 34729 pcps_quicksync_acquisition_cc.cc:504] 7175			6.35968e+10
I1229 14:11:50.930517 34729 pcps_quicksync_acquisition_cc.cc:504] 15175			1.61056e+08
I1229 14:11:50.930519 34729 pcps_quicksync_acquisition_cc.cc:506] code phase 7175
I1229 14:11:50.930522 34729 pcps_quicksync_acquisition_cc.cc:507] doppler **8250**
I1229 14:11:50.930524 34729 pcps_quicksync_acquisition_cc.cc:508] magnitude folded 1770.19
I1229 14:11:50.930526 34729 pcps_quicksync_acquisition_cc.cc:509] input signal power 30666.9
--
I1229 14:11:50.962064 34725 pcps_quicksync_acquisition_cc.cc:495] positive acquisition
I1229 14:11:50.962074 34725 pcps_quicksync_acquisition_cc.cc:496] satellite E 20
I1229 14:11:50.962076 34725 pcps_quicksync_acquisition_cc.cc:497] sample_stamp 5536000
I1229 14:11:50.962077 34725 pcps_quicksync_acquisition_cc.cc:498] test statistics value 0.0746346
I1229 14:11:50.962087 34725 pcps_quicksync_acquisition_cc.cc:499] test statistics threshold 0.05
I1229 14:11:50.962090 34725 pcps_quicksync_acquisition_cc.cc:500] folding factor 2
I1229 14:11:50.962090 34725 pcps_quicksync_acquisition_cc.cc:501] possible delay correlation output
I1229 14:11:50.962092 34725 pcps_quicksync_acquisition_cc.cc:504] 7541			3.88316e+08
[... cut ...]
I1229 14:12:01.196053 34721 pcps_quicksync_acquisition_cc.cc:504] 6261			9.38602e+09
I1229 14:12:01.196056 34721 pcps_quicksync_acquisition_cc.cc:504] 14261			5.78838e+09
I1229 14:12:01.196059 34721 pcps_quicksync_acquisition_cc.cc:506] code phase 6261
I1229 14:12:01.196061 34721 pcps_quicksync_acquisition_cc.cc:507] doppler 15000
I1229 14:12:01.196064 34721 pcps_quicksync_acquisition_cc.cc:508] magnitude folded 1668.37
I1229 14:12:01.196066 34721 pcps_quicksync_acquisition_cc.cc:509] input signal power 31821.8

odrisci, 2020-12-29:
This is just the clock (oscillator) drift - I'm not sure what front-end was used to collect the data, but it looks like your Doppler averages at about 7 kHz. This corresponds to a clock drift of 7000/1575.42e6, or about 4.5 ppm, which is not bad. If you collect your own data you must be sure to configure your acquisition to cover the frequency uncertainty range due to all the possible sources of drift. At L1 each part per million of clock drift corresponds to about 1.5 kHz (1575.42 MHz/1e6). Similarly, each 685 km/hr of line of sight velocity uncertainty corresponds to about 1 kHz at L1. So for typical scenarios where the satellite velocity dominates you want to allocate +/- 3.5 kHz for velocity uncertainty plus +/- 1.5 kHz for each ppm of clock uncertainty. Satellite clock drift and user velocity are typically much smaller.

fgallardo, 2020-12-29:
Dear @odrisci :

I don´t know you but owe you a beer.

Not because of the processing of the CTTC file, but because I was not able to process my own recorded signals with my old USRP1. I was thinking it may be broken.

It turns out the USRP1 has a shitty internal oscillator (25 ppm as per http://openbts.org/w/index.php?title=Ettus_Research_USRP).

I took my old recordings and expanded the search space as per your suggestion: BOOM, it worked!

So:

-> As per the question: you were right.
-> As per the side effect: my USRP works!!
Thanks!!!

odrisci, 2020-12-29:
@fgallardo don't mention it! Glad this worked for you.

closed 2020-12-30 by fgallardo