gnss-sdr/gnss-sdr #460: Error setting gains for bladeRF x40

fdgm0001, 2021-01-10:
Hi guys. I'm trying to do a GPS receiver with blade RF x40. To do it, I am employing this configuration file #[gnss-sdr_GPS_L1_bladeRF.conf](https://github.com/gnss-sdr/gnss-sdr/blob/master/conf/gnss-sdr_GPS_L1_bladeRF.conf).
However, when I execute the file I obtain a error with the RX gain values:

```
Initializing GNSS-SDR v0.0.14.git-next-87882d3d23 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: bladerf=0
gr-osmosdr 46204cea (0.1.5git) gnuradio 3.8.1.0
built-in source types: osmosdr rtl uhd hackrf bladerf rfspace airspy soapy redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF Serial # 7357...7fb6 FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
**[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1869] bladerf1_get_gain_stage: gain stage 'LNA' invalid
Actual LNA Gain: 0 dB...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1869] bladerf1_get_gain_stage: gain stage 'VGA1' invalid
Actual VGA1 Gain: 0 dB...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1869] bladerf1_get_gain_stage: gain stage 'VGA2' invalid
Actual VGA2 Gain: 0 dB...**
Actual Bandwidth: 2.8e+07 [Hz]...
```

jonathanmuller, 2021-01-12:
Hi, did you already take a look at this issue ?
#236

fdgm0001, 2021-01-12:
Yes. I have configured the gains with bladeRF-cli, then I have run the config file of GNSS-SDR with the gain parameters commented and the result is the same.

Anyways, thank you so much for your contribution.

jonathanmuller, 2021-01-13:
When you say
> ... the result is the same ...

Do you mean : 

- You still have the invalid gain stage error
-> This is more or less "expected", it still tries to get the gain values, but this doesn't affect further operations
- Setting the gains manually did not seem to have any effect on GNSS-SDR
-> How did you come to this conclusion ? I tested this and the (manually set) gain were definitely having an effect on the TTFF (for example)

fdgm0001, 2021-01-13:
The first option. After the invalid gain error, the program is still running and losses of lock occur every time.

jonathanmuller, 2021-01-15:
Then the problem probably does not come from the gains (that are now manually set and working) but from your setup.
I'm not the most qualified to answer your answer regarding the bias tee but you can read some user experience in the tickets
#443 and #435 for example
So I would suggest you to have the best setup possible for a start :
- Clear sky view
- Good active antenna (thus tee bias)
- etc