gnss-sdr/gnss-sdr #463: Space applications (rocketry, LEO satellites etc.)

kozuch, 2021-01-13:
Could GNSS-SDR be used in the field of spaceflight for things high-power rocketry, low earth orbit satellites (like the CubeSat/PocketQube formats) etc? I am just curios whether this is possible or if someone even tried it already.

Are there any known issues why GNSS-SDR may not work for these types of applications (I mean software issues in the library and suppose one uses proper hardware setup)?

jafrado, 2021-01-13:
Yes, and it does work with the right HW/SW solution (tested already). But there are other commercially available solutions much lower cost for HPR. Saw your post on the RTK list too about limits, as mentioned 80km is an easy target with commercial receivers today, outside the realm of currently any HPR application.

alexbvo, 2022-08-03:
Dear @jafrado can you please, share examples of commercial solutions using the GNSS SDR for spacraft? In addition, also the commercially available soultion for HPR?