gnss-sdr/gnss-sdr #464: Error with -DENABLE_OSMOSDR=ON

fdgm0001, 2021-01-15:
Hi all. I'm trying to do a GPS receiver with bladeRF x40. I have installed GNSS-SDR and GNU-Radio using PyBOMBS, so I need to build the Osmosdr_Signal_Source implementation for the bladeRF x40. To do that I need to execute the next commands:

```
$ cmake -DENABLE_OSMOSDR=ON ../
$ make
$ sudo make install
```
However, when I execute the `$ make` command I obtain this result:

```
[  2%] Built target volk_gnsssdr_module
[  4%] Built target pugixml-1.11.4
[  5%] Built target gflags-2.2.2
[  6%] Built target glog-0.4.0
[ 54%] Built target core_libs_supl
[ 56%] Built target core_system_parameters
[ 60%] Built target algorithms_libs
[ 60%] Built target acquisition_libs
[ 62%] Built target algorithms_libs_rtklib
[ 64%] Built target pvt_libs
[ 65%] Built target core_libs
[ 66%] Built target channel_libs
[ 67%] Built target acquisition_gr_blocks
[ 67%] Built target gnss_sdr_flags
[ 70%] Built target acquisition_adapters
[ 70%] Built target channel_adapters
[ 70%] Built target conditioner_adapters
[ 71%] Built target data_type_gr_blocks
[ 72%] Built target data_type_adapters
[ 72%] Built target observables_libs
[ 72%] Built target obs_gr_blocks
[ 72%] Built target obs_adapters
[ 73%] Built target telemetry_decoder_libs
[ 73%] Built target telemetry_decoder_libswiftcnav
[ 74%] Built target telemetry_decoder_gr_blocks
[ 76%] Built target telemetry_decoder_adapters
[ 77%] Built target resampler_gr_blocks
[ 77%] Built target resampler_adapters
[ 77%] Built target signal_generator_gr_blocks
[ 78%] Built target signal_generator_adapters
[ 79%] Built target signal_source_libs
[ 80%] Built target signal_source_gr_blocks
[ 82%] Built target signal_source_adapters
[ 82%] Built target input_filter_gr_blocks
[ 83%] Built target input_filter_adapters
[ 85%] Built target tracking_libs
[ 86%] Built target tracking_gr_blocks
[ 89%] Built target tracking_adapters
[ 89%] Built target pvt_gr_blocks
[ 89%] Built target pvt_adapters
[ 90%] Built target core_monitor
[ 91%] Built target core_receiver
[ 91%] Linking CXX executable gnss-sdr
/usr/bin/ld: warning: libgnuradio-iqbalance.so.3.8.0, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libosmosdr.so.0, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: librtlsdr.so.0, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libhackrf.so.0, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libairspy.so.0, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libSoapySDR.so.0.8, needed by /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_stop_tx'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_get_sample_rate'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_start_rx'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `airspy_set_vga_gain'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_device_list_free'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_stop_rx'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_start_tx'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_get_sample_rates'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `airspy_set_rf_bias'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_usb_strings'
[... 69 lines cut ...]
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `gr::iqbalance::optimize_c::make(int)'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_version_string_read'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `airspy_version_string_read'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_read_async'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_direct_sampling'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_get_center_freq'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_sample_rate'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_bias_tee'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_xtal_freq'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_set_sample_rate'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `hackrf_close'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `airspy_set_conversion_filter_float32'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_index_by_serial'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_freq_correction'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `osmosdr_set_tuner_gain_mode'
/usr/bin/ld: /home/fdgm0001/pybombs/bladeRF/lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_xtal_freq'
collect2: error: ld returned 1 exit status
make[2]: *** [src/main/CMakeFiles/gnss-sdr.dir/build.make:169: src/main/gnss-sdr] Error 1
make[1]: *** [CMakeFiles/Makefile2:2734: src/main/CMakeFiles/gnss-sdr.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

carlesfernandez, 2021-01-29:
Hi @fdgm0001 ,

you can add the `-DENABLE_OSMOSDR=ON` flag in Pybomb's gnss-sdr recipe:

```
printf "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON \"\n" >> $HOME/.pybombs/recipes/gr-recipes/gnss-sdr.lwr
```

and then use Pybombs to build gnss-sdr:

```
pybombs install gr-osmosdr gnss-sdr
```

fdgm0001, 2021-04-22:
> Hi @fdgm0001 ,
> 
> you can add the `-DENABLE_OSMOSDR=ON` flag in Pybomb's gnss-sdr recipe:
> 
> ```
> printf "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON \"\n" >> $HOME/.pybombs/recipes/gr-recipes/gnss-sdr.lwr
> ```
> 
> and then use Pybombs to build gnss-sdr:
> 
> ```
> pybombs install gr-osmosdr gnss-sdr
> ```

I have tried to add the flag like you described in your post, but It didn't work. Probably It didn't work because I'm not sure how to add the flag correclty.

Besides, I have been looking for other people with the same problem, and I found [this post with the same issue](https://github.com/gnss-sdr/gnss-sdr/issues/445).  So, I have installed again GNSS-SDR but I still have the same problem.

In any case, thanks @carlesfernandez for your help.

fdgm0001, 2021-09-17:
printf "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON\"\n" >> $HOME/.pybombs/recipes/gr-recipes/gnss-sdr.lwr
printf "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON \"\n" >> $HOME/.pybombs/recipes/gr-recipes/gnss-sdr.lwr

closed 2021-09-17 by fdgm0001