gnss-sdr/gnss-sdr #465: /usr/bin/ld: cannot find -lgnuradio-uhd

AmirAllahvirdi, 2021-01-19:
**Hi,** 

**I'm trying to install gnss-sdr in Ubuntu 18.04 with and connect the LimeSDR to use it. I followed the following procedure:**

- sudo apt-get install git python3-pip
- sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
- pybombs auto-config
- pybombs recipes add-defaults
- pybombs prefix init $HOME/sdr -a myprefix -R gnuradio-default
- cd sdr
- . ./setup_env.sh
- cd ..
- pybombs install armadillo gflags glog gnutls
- sudo apt-get install autoconf automake libtool curl make g++ unzip
- wget https://github.com/protocolbuffers/protobuf/releases/download/v3.13.0/protobuf-cpp-3.13.0.tar.gz
- tar xvfz protobuf-cpp-3.13.0.tar.gz
- cd protobuf-3.13.0
- ./autogen.sh
- ./configure
- make
- sudo make install
- sudo ldconfig
- cd ..
- git clone git://git.osmocom.org/osmo-sdr.git
- cd osmo-sdr/software/libosmosdr
- mkdir build
- cd build/
- cmake ..
- make
- sudo make install
- sudo ldconfig
- cd ../..
- git clone git://git.osmocom.org/gr-osmosdr
- cd gr-osmosdr
- mkdir build
- cd build
- cmake .. -Wno-dev
- make
- sudo make install
- sudo ldconfig
- cd ../../../..
- git clone https://github.com/gnss-sdr/gnss-sdr
- cd gnss-sdr
- git checkout next
- cd build
- cmake -DENABLE_OSMOSDR=ON ..
- make
- sudo make install

**But there is an error in `sudo make install`, and the gnss-sdr is not installed:**

Scanning dependencies of target tracking_gr_blocks
[ 89%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/dll_pll_veml_tracking.cc.o
[ 89%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l1_ca_dll_pll_c_aid_tracking_cc.cc.o
[ 89%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l1_ca_dll_pll_c_aid_tracking_sc.cc.o
[ 89%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l1_ca_dll_pll_tracking_cc.cc.o
[ 90%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l2_ca_dll_pll_c_aid_tracking_cc.cc.o
[ 90%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l2_ca_dll_pll_c_aid_tracking_sc.cc.o
[ 90%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/glonass_l2_ca_dll_pll_tracking_cc.cc.o
[ 90%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/gps_l1_ca_kf_tracking_cc.cc.o
[ 90%] Linking CXX static library libtracking_gr_blocks.a
[ 90%] Built target tracking_gr_blocks
[ 93%] Built target tracking_adapters
[ 93%] Built target pvt_gr_blocks
[ 94%] Built target pvt_adapters
[ 94%] Built target core_monitor
[ 95%] Built target core_receiver
[ 95%] Linking CXX executable gnss-sdr
/usr/bin/ld: cannot find -lgnuradio-uhd
/usr/bin/ld: cannot find -luhd
/usr/bin/ld: cannot find -lgnuradio-filter
/usr/bin/ld: cannot find -lgnuradio-blocks
/usr/bin/ld: cannot find -lgnuradio-fft
/usr/bin/ld: cannot find -lgnuradio-runtime
/usr/bin/ld: cannot find -lgnuradio-pmt
/usr/bin/ld: cannot find -lvolk
collect2: error: ld returned 1 exit status
src/main/CMakeFiles/gnss-sdr.dir/build.make:188: recipe for target 'src/main/gnss-sdr' failed
make[2]: *** [src/main/gnss-sdr] Error 1
CMakeFiles/Makefile2:2971: recipe for target 'src/main/CMakeFiles/gnss-sdr.dir/all' failed
make[1]: *** [src/main/CMakeFiles/gnss-sdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I really appreciate it if you help me to solve the problem.

Thank you and best regards

carlesfernandez, 2021-01-20:
Try this:

* sudo apt-get install git python3-pip
* sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
* pybombs auto-config
* pybombs recipes add-defaults
* pybombs prefix init $HOME/sdr -a myprefix -R gnuradio-default
* cd sdr
* . ./setup_env.sh
* cd ..
* Remove iqbal from gr-osmosdr recipe, since it seems to fail:
  ```
  sed -i '/iqbal/d' $HOME/.pybombs/recipes/gr-recipes/gr-osmosdr.lwr  
  ```
* Add the osmosdr flag in gnss-sdr recipe:
  ```
  printf "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON \"\n" >> $HOME/.pybombs/recipes/gr-recipes/gnss-sdr.lwr 
  ```
* pybombs install gr-osmosdr gnss-sdr

This will install gnss-sdr (the `next` branch) in your Pybombs environment.

AmirAllahvirdi, 2021-01-22:
Dear @carlesfernandez 

I tried several times, but there is a problem. When I connect the LimeSDR, the `SoapySDRUtil --probe` cannot find the device. Please note that I'm using Ubuntu 18.04, and the GNSS-SDR works fine with `My first position fix`, but not with the LimeSDR. So I install:

- sudo add-apt-repository -y ppa:myriadrf/drivers
- sudo apt-get update
- sudo apt-get install limesuite liblimesuite-dev limesuite-udev limesuite-images
- sudo apt-get install soapysdr soapysdr-module-lms7

I can connect to the LimeSDR through LimeSuite, `SoapySDRUtil --probe` cannot find the device. I got confused during this week.  I really appreciate it if you provide a clean procedure on any version of Linux which works (I’m using VM) to install GNSS-SDR to use LimeSDR. (I have a simulator and want to simulate some data).

Thank you and best regards
Amir

AmirAllahvirdi, 2021-01-22:
Thank you @carlesfernandez, 

I made some efforts and finally both GNSS-SDR and LimeSDR are installed on ubuntu 18.04 and ready to work. The GNSS-SDR passed My first position fix and `SoapySDRUtil --prob` find the LimeSDR without any problem. Now I'm trying to feed the LimeSDR with a Spirent simulator and run the GNSS-SDR by using this conf. file https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/gnss-sdr_GPS_L1_LimeSDR.conf 

I encounter the following error:

Initializing GNSS-SDR v0.0.14.git-next-419eff942 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: driver=lime,soapy=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 

FATAL: SoapySDR::Device::make() no match

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

Set RX Antenna: 
Actual RX Rate: 0 [SPS]...
Actual RX Freq: 0 [Hz]...
PLL Frequency tune error: -1.57542e+09 [Hz]...
Actual RX Gain: 0 dB...
Actual Bandwidth: 0 [Hz]...
E0122 12:46:28.883955 29576 pass_through.cc:99] This implementation only supports one input stream
E0122 12:46:28.884174 29576 pass_through.cc:100] 8
E0122 12:46:28.950374 29576 gnss_flowgraph.cc:701] itemsize mismatch: hybrid_observables_gs0:0 using 152, copy2:0 using 8
E0122 12:46:28.950464 29576 control_thread.cc:303] Unable to connect flowgraph
Total GNSS-SDR run time: 0.00653797 [seconds]
GNSS-SDR program ended.

Any idea what is the problem and what should I do? I appreciate it...

Best regards
Amir

carlesfernandez, 2021-01-29:
Well, it seems that the Osmosdr_Signal_Source cannot set your device correctly, so the receiver cannot connect the flowgraph and exits. 

I see the line `OsmoSdr arguments: driver=lime,soapy=1`, but in the example file we have `SignalSource.osmosdr_args=driver=lime,soapy=0`. Could be that?

AmirAllahvirdi, 2021-01-29:
It solved, I first installed using docker, it did not work. I then installed Soapy bu using:

`sudo apt install soapysdr-tools`

and then installed the dependencies. There was and error 
`Failed to parse '/usr/share/mime/packages/gnuradio.xml`
 
So I removed the gnuradio.xml and Symlink gnuradio-grc.xml to gnuradio.xml and then update the mime database. Then make and installed GNSS-SDR and run it with sudo tp have the permissions to open the USB port. It worked!

closed 2021-01-29 by AmirAllahvirdi