gnss-sdr/gnss-sdr #470: On the content of the telemetryx.dat dump files

fgallardo, 2021-02-04:
Hi all,

I am trying to decode manually the dump files of the telemetry (telemetry[channel_number].dat) for Galileo I/NAV (E1B) messages.

So, I opened the files and search for the synchronization pattern (successfully).

Then I am applying the de-interleaver and then the Viterbi for the convolutional FEC.

I am pretty sure the de-interleaver and the viterbi work because it work with the example sequences form the Galileo ICD:

https://www.gsc-europa.eu/sites/default/files/sites/all/files/Galileo-OS-SIS-ICD.pdf

Section D, subsection D.2.

Am I missing something? I understand the content of the telemetry file is the INAV navigation message as defined by the ICD, no?
I may have some bug somewhere, but just to make sure I am processing the file properly: 

1. Search for the I/NAV synchronization pattern:  0101100000 [OK]
2. Take out those 10 symbols.
3. Take the following 240 symbols. 
4. De-interleave those 240 Symbols. [De-interleaver validated using the example from the ICD]
5. Apply de Viterbi on the de-interleaved symbols. [Viterbi validated using the example from the ICD]

The result should the bits of the INAV message.

Am I missing something there?

Thanks
Fran

carlesfernandez, 2021-02-05:
Dear @fgallardo 

your description is correct. I don't know if this is the case, but it *could* happen that your detected preamble is not a "real" one, but an unfortunate sequence of data bits in the payload. Checking the CRC and the distance between two consecutive preambles can help to avoid that.

Please also check that you are working with data from the same satellite (field `PRN`) since the channel could have dropped a satellite and picked up another one.

Hope this helps!

fgallardo, 2021-02-05:
Hi Carles,

Good to talk to you btw! :) Happy new year!
So, I think it has to do with the way I am reading the data.

I see that depending on the Prompt correlator sign output, you write in the binary file a 32int with 1 or -1 (or 0 should the value is 0).

But then I would say you are writting the PRN too, no?
See below the last lines or LoCs 956 and 957 of https://github.com/gnss-sdr/gnss-sdr/blob/master/src/algorithms/telemetry_decoder/gnuradio_blocks/galileo_telemetry_decoder_gs.cc

 uint64_t tmp_ulong_int;
 int32_t tmp_int;
 tmp_double = static_cast<double>(d_TOW_at_current_symbol_ms) / 1000.0;
 d_dump_file.write(reinterpret_cast<char *>(&tmp_double), sizeof(double));
 tmp_ulong_int = current_symbol.Tracking_sample_counter;
 d_dump_file.write(reinterpret_cast<char *>(&tmp_ulong_int), sizeof(uint64_t));
 tmp_double = static_cast<double>(d_TOW_at_Preamble_ms) / 1000.0;
 d_dump_file.write(reinterpret_cast<char *>(&tmp_double), sizeof(double));
 switch (d_frame_type)
 {
    case 1:
       tmp_int = (current_symbol.Prompt_I > 0.0 ? 1 : -1);
       break;
    case 2:
       tmp_int = (current_symbol.Prompt_Q > 0.0 ? 1 : -1);
       break;
    case 3:
       tmp_int = (current_symbol.Prompt_I > 0.0 ? 1 : -1);
       break;
    default:
       tmp_int = 0;
       break;
   }
d_dump_file.write(reinterpret_cast<char *>(&tmp_int), sizeof(int32_t));
tmp_int = static_cast<int32_t>(current_symbol.PRN);
d_dump_file.write(reinterpret_cast<char *>(&tmp_int), sizeof(int32_t));
  }

fgallardo, 2021-02-05:
Ok, I just found the following in the  V0.0.14 Release Note:

https://github.com/gnss-sdr/gnss-sdr/releases/tag/v0.0.14

"Improved dumps in Telemetry Decoding blocks. ->Now they include the raw
navigation message bits <-. If TelemetryDecoder_XX.dump=true, the resulting
.dat binary file is also delivered in .mat format, which is readable from
Matlab and Python."

It was there all the time. Ok, downloading v0.0.14!

BTW: I see you included E6 (thanks!)

fgallardo, 2021-02-05:
Yes!

It works perfectly!!

I installed V0.0.14. 

The format in the dat is:

double Tow (factor of 1000)
uint64_t sample_counter
double tow_preamble (factor 1000)
int32_t Promt_I (as it is INAV)
int32_t PRN

It works perfectly!! :)

Thanks Carles!

closed 2021-02-05 by fgallardo

carlesfernandez, 2021-02-08:
Hi @fgallardo! I'm happy to hear from you, and to know that the navigation bits retrieval worked well. Happy decoding :-)