gnss-sdr/gnss-sdr #471: Loss of lock constantly

Bilelhm98, 2021-02-07:
Hello everyone. I am getting some problems using GNSS-SDR. and I do not get to solve it. 
About the equipment used: 
- I have a hackrf one, in addition 
- I am using an external clock. 
- The antenna used is the ANT500. How it didn't work I got the b25-ja3 which is an antenna oriented to GPS.

About the config file, here it is:

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000
GNSS-SDR.telecommand_enabled=true
GNSS-SDR.telecommand_tcp_port=3333
GNSS-SDR.telecommand_udp_port=1112

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=45
SignalSource.rf_gain=45
SignalSource.if_gain=40
SignalSource.AGC_enabled=true
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.antenna=LNAW
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;## Lectura de datos asistida:
;##GNSS-SDR.AGNSS_XML_enabled=true
;##GNSS-SDR.AGNS_ref_location=37.176,-3.611
;##GNSS-SDR.AGNSS_gal_almanac_xml=2021-01-12

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
[... 158 lines cut ...]
Current receiver time: 1 min 1 s
Current receiver time: 1 min 2 s
Current receiver time: 1 min 3 s
Current receiver time: 1 min 4 s
Current receiver time: 1 min 5 s
Current receiver time: 1 min 6 s
Current receiver time: 1 min 7 s
Current receiver time: 1 min 8 s
Current receiver time: 1 min 9 s
Current receiver time: 1 min 10 s
Current receiver time: 1 min 11 s
Current receiver time: 1 min 12 s
**_### Loss of lock in channel 3!
Loss of lock in channel 6!
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 21 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 23 (Block III)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block Decommissioned)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 11 (Block IIR)
Loss of lock in channel 7!
Loss of lock in channel 4!_**
Current receiver time: 1 min 13 s
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 09 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 29 (Block IIR-M)

ZiglioUK, 2021-02-07:
It's tricky to understand what's going on because of the lack of debug info. For example you need to check what doppler and CNos are acquired and check that the doppler is correct. I've had problems with the frequency translating filter with satellites appearing at wrong frequencies. Maybe there was some aliasing going on.

After acquisition there's FLL pull-in that Jesus l lasts a default of 5s if U remember correctly.
That should bring the carrier frequency close to the true value (that also needs to be checked).
Then the preamble is searched for bit synchronization and I suspect that's what's timing out for you.

So yeah, you should make sure the CNos are strong enough first after acquisition. 
Can you record a snippet and maybe go over that file only for the strongest satellite? 
In open sky you should expect a CNo>40dB

Bilelhm98, 2021-02-07:
> It's tricky to understand what's going on because of the lack of debug info. For example you need to check what doppler and CNos are acquired and check that the doppler is correct. I've had problems with the frequency translating filter with satellites appearing at wrong frequencies. Maybe there was some aliasing going on.
> 
> After acquisition there's FLL pull-in that should last a default of 5s if I remember correctly.
> That should bring the carrier frequency close to the true value (that also needs to be checked).
> Then the preamble is searched for bit synchronization and I suspect that's what's timing out for you.
> 
> So yeah, you should make sure the CNos are strong enough first after acquisition.
> Can you record a snippet and maybe go over that file only for the strongest satellite?
> In open sky you should expect a CNo>40dB

Hello,

Thank you so much for your answer. It is really helpful.
However, I do not know how to analyze the CNo. Would I have to export the data to matlab or any other software and analyze them or could I do it directly using GNSS-SDR?

ZiglioUK, 2021-02-07:
Yes, unfortunately there's quite a bit to learn if you're not familiar with gnss receivers.
You could enable tracking dump, then you could run a matlab script to plot the result.
But basically, all you're interested in to start with is the CNo levels after acquisition.
What I do, I just print them to the screen. That requires modifying the source code.
I don't have a HackRF myself but if you're happy to capture a 5 second radio file and share it I'd be happy to help.

But first check basic things like if your active antenna or LNA is powered.
Acquisition is more sensitive than tracking and the acquired signal might just not be powerful enough to be tracked.

Bilelhm98, 2021-02-08:
> Yes, unfortunately there's quite a bit to learn if you're not familiar with gnss receivers.
> You could enable acquisition and tracking dump, then you could run a matlab script to plot the result.
> But basically, all you're interested in to start with is the CNo levels after acquisition.
> What I do, I just print them to the screen. That requires modifying the source code.
> I don't have a HackRF myself but if you're happy to capture a 5 second radio file and share it I'd be happy to help.
> 
> But first check basic things like if your active antenna or LNA is powered.
> Acquisition is more sensitive than tracking and the acquired signal might just not be powerful enough to be tracked.

I attach two pics for a better understanding: 

Pic1: The fourier transform of a fragment of a Signal source. This is the signal the equivalent to the datas introduced on my computer withouth being processed:
[image]

Pic 2:: In this case I am showing the signal equivalent to the 1C band (GPS), channel 1 and sat1:
[image]

- [ ]

ZiglioUK, 2021-02-08:
Yeah, 30dB is enough, how did you generate that second plot?
BTW, [ in this file ](https://github.com/gnss-sdr/gnss-sdr/blob/71026773589816217e37a0e363011169171fde24/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L1750) at line 1750 it says:

```
                LOG(INFO) << "Number of samples between Acquisition and Tracking = " << acq_trk_diff_samples << " ( " << acq_trk_diff_seconds << " s)";
                DLOG(INFO) << "PULL-IN Doppler [Hz] = " << d_carrier_doppler_hz
                           << ". PULL-IN Code Phase [samples] = " << d_acq_code_phase_samples;

```

It's quite the acquired signal frequency, it's after PULL-IN but if you change that to std::cout, or look at the log, that should help.

Bilelhm98, 2021-02-08:
> Yeah, 30dB is enough, how did you generate that second plot?
> BTW, [ in this file ](https://github.com/gnss-sdr/gnss-sdr/blob/71026773589816217e37a0e363011169171fde24/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L1750) at line 1750 it says:
> 
> ```
> LOG(INFO) << "Number of samples between Acquisition and Tracking = " << acq_trk_diff_samples << " ( " << acq_trk_diff_seconds << " s)";
>  DLOG(INFO) << "PULL-IN Doppler [Hz] = " << d_carrier_doppler_hz
>                            << ". PULL-IN Code Phase [samples] = " << d_acq_code_phase_samples;
> ```
> 
> It's not quite the acquired signal frequency, it's after PULL-IN but if you change that to std::cout, or look at the log, that should help.
> 
> Also, further on, [at line 1780](https://github.com/gnss-sdr/gnss-sdr/blob/71026773589816217e37a0e363011169171fde24/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L1780), where it says:
> 
> ```
>  LOG(INFO) << d_systemName << " " << d_signal_pretty_name << " tracking synchronization time limit reached in channel " << d_channel
>    << " for satellite " << Gnss_Satellite(d_systemName, d_acquisition_gnss_synchro->PRN) << '\n';
> ```
> 
> If you print that message, I'm pretty sure that's what's happening because otherwise you'd see this ([line 1830](https://github.com/gnss-sdr/gnss-sdr/blob/71026773589816217e37a0e363011169171fde24/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L1830)):
> 
> ```
>   std::cout << d_systemName << " " << d_signal_pretty_name << " tracking bit synchronization locked in channel " << d_channel
>                                                  << " for satellite " << Gnss_Satellite(d_systemName, d_acquisition_gnss_synchro->PRN) << '\n';
> ```

The second plot is the result of enable the dum in the adquisition (in the config file I wrote: Acquisition_1C.dump=true)
Then I opened the file in Matlab, and represented the fourier transform of the variable "acq_grid".

About your advices, maybe it is a really basic question but I do not know how to do what you say. Do I have to change line 1750 and write this:
`LOG(INFO) << "Number of samples between Acquisition and Tracking = " << acq_trk_diff_samples << " ( " << acq_trk_diff_seconds << " s)";
 DLOG(INFO) << "PULL-IN Doppler [Hz] = " << d_carrier_doppler_hz
                           << ". PULL-IN Code Phase [samples] = " << **_### std::cout_**;`

And write in line 1780:
printf(LOG(INFO) << d_systemName << " " << d_signal_pretty_name << " tracking synchronization time limit reached in channel " << d_channel
   << " for satellite " << Gnss_Satellite(d_systemName, d_acquisition_gnss_synchro->PRN) << '\n';)

Am I right?

ZiglioUK, 2021-02-08:
> The second plot is the result of enable the dum in the adquisition (in the config file I wrote: Acquisition_1C.dump=true)
> Then I opened the file in Matlab, and represented the fourier transform of the variable "acq_grid".

Great, that's gnss-sdr sees. It looks like a peak. There will be the frequency/doppler information in that script too.
 
> About your advices, maybe it is a really basic question but I do not know how to do what you say. Do I have to change line 1750 where it says d_acq_code_phase_samples; and write std::cout; is this right?
> 
> And write in line 1780:
> printf(LOG(INFO) << d_systemName << " " << d_signal_pretty_name << " tracking synchronization time limit reached in channel " << d_channel
> << " for satellite " << Gnss_Satellite(d_systemName, d_acquisition_gnss_synchro->PRN) << '\n';)
> 
> Am I right?

std::cout is the cpp equivalent of printf. printf should work too

Bilelhm98, 2021-02-08:
> > The second plot is the result of enable the dum in the adquisition (in the config file I wrote: Acquisition_1C.dump=true)
> > Then I opened the file in Matlab, and represented the fourier transform of the variable "acq_grid".
> 
> Great, that's what gnss-sdr sees too. It looks like a valid peak. There will be the frequency/doppler information in that script too.
> 
> > About your advices, maybe it is a really basic question but I do not know how to do what you say. Do I have to change line 1750 where it says d_acq_code_phase_samples; and write std::cout; is this right?
> > And write in line 1780:
> > printf(LOG(INFO) << d_systemName << " " << d_signal_pretty_name << " tracking synchronization time limit reached in channel " << d_channel
> > << " for satellite " << Gnss_Satellite(d_systemName, d_acquisition_gnss_synchro->PRN) << '\n';)
> > Am I right?
> 
> std::cout is the cpp equivalent of printf. printf should work too

Ok, I modified the code and in line 1750 I wrote:
`std:: cout << "Number of samples between Acquisition and Tracking = " << acq_trk_diff_samples << " ( " << acq_trk_diff_seconds << " s)";
std:: cout << "PULL-IN Doppler [Hz] = " << d_carrier_doppler_hz
                           << ". PULL-IN Code Phase [samples] = " << d_acq_code_phase_samples;`

And the same in line 1780.
However, when I iniciate gnss-sdr I do not see any change on the screen. It does not print any message.

ZiglioUK, 2021-02-08:
Silly question, did you recompile the code?
```
doppler_grid_narrow_min=-250
doppler_max=5000
dopler_step=250
doppler_stel_narrow=125
```

those are the acquisition parameters, not the output. Sorry, I don't have time to look at that script right now.

Bilelhm98, 2021-02-08:
> Silly question, did you recompile the code?
> 
> ```
> doppler_grid_narrow_min=-250
> doppler_max=5000
> dopler_step=250
> doppler_stel_narrow=125
> ```
> 
> those are the acquisition parameters, not the output. Sorry, I don't have time to look at that script right now.

True, I forgot to compile. Typical mistake I guess haha. However, when I try to compile I receive the next message:

In file included from dll_pll_veml_tracking.cc:23:
dll_pll_veml_tracking.h:21:10: fatal error: cpu_multicorrelator_real_codes.h: No existe el archivo o el directorio
   21 | #include "cpu_multicorrelator_real_codes.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

To compile what I did is to go to the directory and then:
gcc dll_pll_veml_tracking.cc -o dll_pll_veml_tracking

ZiglioUK, 2021-02-08:
No, unfortunately it's more complicated than that. You must run Cmake and then make, follow the instructions on how to build from the documentation pages

Bilelhm98, 2021-02-08:
> No, unfortunately it's more complicated than that. You must run Cmake and then make, follow the instructions on how to build from the documentation pages

I already got that information, thank you so much for all your help!
I attach the results using the antenna ANT500: 
Initializing GNSS-SDR v0.0.14 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file fcd rtl_tcp uhd hackrf rfspace redpitaya 
Using HackRF One with firmware 2017.02.1
Set RX Antenna: TX/RX
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
AGC enabled
Actual Bandwidth: 6e+06 [Hz]...
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
TcpCmdInterface: Telecommand TCP interface listening on port 3333
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Number of samples between Acquisition and Tracking = 2096 ( 0.001048 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 373
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Number of samples between Acquisition and Tracking = -1732 ( -0.000866 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 2598
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 21 (Block IIR)
Number of samples between Acquisition and Tracking = 2096 ( 0.001048 s)PULL-IN Doppler [Hz] = 4375. PULL-IN Code Phase [samples] = 854
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 23 (Block III)
Number of samples between Acquisition and Tracking = 6187 ( 0.0030935 s)PULL-IN Doppler [Hz] = -4375. PULL-IN Code Phase [samples] = 50
Current receiver time: 2 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 30 (Block IIF)
Number of samples between Acquisition and Tracking = -4727 ( -0.0023635 s)PULL-IN Doppler [Hz] = -750. PULL-IN Code Phase [samples] = 3763
Loss of lock in channel 6!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 29 (Block IIR-M)
Number of samples between Acquisition and Tracking = -5461 ( -0.0027305 s)PULL-IN Doppler [Hz] = 4750. PULL-IN Code Phase [samples] = 2430
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 30 (Block IIF)
Number of samples between Acquisition and Tracking = -2351 ( -0.0011755 s)PULL-IN Doppler [Hz] = -625. PULL-IN Code Phase [samples] = 3410
Loss of lock in channel 5!
Current receiver time: 3 s
OOOOCurrent receiver time: 4 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 29 (Block IIR-M)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 3052
OOTracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 04 (Block III)
Number of samples between Acquisition and Tracking = 2096 ( 0.001048 s)PULL-IN Doppler [Hz] = -4625. PULL-IN Code Phase [samples] = 1133
OTracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 15 (Block IIR-M)
Number of samples between Acquisition and Tracking = -4512 ( -0.002256 s)PULL-IN Doppler [Hz] = -2875. PULL-IN Code Phase [samples] = 3664
Loss of lock in channel 4!
Current receiver time: 5 s
Current receiver time: 6 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 14 (Block Decommissioned)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = -1625. PULL-IN Code Phase [samples] = 3082
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
[... 209 lines cut ...]
Loss of lock in channel 6!
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 21 (Block IIR)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = -4375. PULL-IN Code Phase [samples] = 2977
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 23 (Block III)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = 4250. PULL-IN Code Phase [samples] = 2252
Loss of lock in channel 5!
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 09 (Block IIF)
Number of samples between Acquisition and Tracking = -4512 ( -0.002256 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 3181
Current receiver time: 1 min 13 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 29 (Block IIR-M)
Number of samples between Acquisition and Tracking = -4512 ( -0.002256 s)PULL-IN Doppler [Hz] = -4625. PULL-IN Code Phase [samples] = 3558
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 29 (Block IIR-M)
[... cut ...]
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 09 (Block IIF)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 2499
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 29 (Block IIR-M)
Number of samples between Acquisition and Tracking = -2000 ( -0.001 s)PULL-IN Doppler [Hz] = 4625. PULL-IN Code Phase [samples] = 2874
Current receiver time: 1 min 14 s
Current receiver time: 1 min 15 s

ZiglioUK, 2021-02-08:
Ok, any doppler >+-4.2KHz is suspicious. See for example PRN 29 keeps returning different dopplers every time.
That's a sign of wrong acquisition or wrong pull-in. Either way, I've had a similar problem using the Translating Filter myself.
I would pick a single PRN that doesn't show a big change of doppler and only try and track that one, for example PRN 30.
Again, I don't have a HackRF so I don't know if I can help any further with the configuration.
Happy to have a look when I have time at a radio snippet if you like.

Bilelhm98, 2021-02-08:
> Ok, any doppler >+-4.2KHz is suspicious. See for example PRN 29 keeps returning different dopplers every time.
> That's a sign of wrong acquisition or wrong pull-in. Either way, I've had a similar [...]

I've been reading in GNSS-SDR how to track only one PRN (in the Tracking configuration) and I didn't find anything. Do you know how I could do it? 
In addition, I would like to ask this Doppler efect, maybe changing the parameters of the filters could it be solved?

Thank you so much for your help. I am really grateful.

ZiglioUK, 2021-02-08:
> I've been reading in GNSS-SDR how to track only one PRN (in the Tracking configuration) and I didn't find anything. Do you know how I could do it?

So, if you download an app like [GPS Test](https://play.google.com/store/apps/details?id=com.android.gpstest&hl=en_GB&gl=US) before you capture a sample look at the skyplot
[image]

Pick a satellite above your head, for example PRN 13 in the picture.
The doppler there will be near zero. Then you can configure gnss-sdr just to look for that satellite:
```
Channels_1C.count=1
Channels.in_acquisition=1
Channel0.satellite=13
```

Restrict the doppler search space to 1KHz and reduce the probability of false acquisition:

```
Acquisition_1C.doppler_max=1000
Acquisition_1C.pfa=0.0001
```

See what happens. Then you should look a the Freq_Xlating_Fir_Filter settings, I suspect something's going on there.

> In addition, I would like to ask this Doppler efect, maybe changing the parameters of the filters could it be solved?
> 
> Thank you so much for your help. I am really grateful.

ZiglioUK, 2021-02-08:
I've found [this thread](https://github.com/gnss-sdr/gnss-sdr/issues/295) that could be useful.

Bilelhm98, 2021-02-09:
Here are the results on the app:

[image]

Then I tried your configuration for the satellite 15 and it doesn't get any channel. Results here:
Initializing GNSS-SDR v0.0.14 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file fcd rtl_tcp uhd hackrf rfspace redpitaya 
Using HackRF One with firmware 2017.02.1
Set RX Antenna: TX/RX
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
AGC enabled
Actual Bandwidth: 6e+06 [Hz]...
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
TcpCmdInterface: Telecommand TCP interface listening on port 3333
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
OOOOOCurrent receiver time: 6 s
OOOOOOOCurrent receiver time: 7 s
Current receiver time: 8 s
OOCurrent receiver time: 9 s
OOOOOOOOOOCurrent receiver time: 10 s
OOOOOOOOCurrent receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
OOOOOOOOOCurrent receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
OOOOOOCurrent receiver time: 17 s
Current receiver time: 18 s
OOOOOOOCurrent receiver time: 19 s
OOOCurrent receiver time: 20 s
Current receiver time: 21 s
OOOOOOOOOOOOCurrent receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
OOOOOOOOOOOOOOOOOOOOCurrent receiver time: 25 s
OCurrent receiver time: 26 s
OOOOOCurrent receiver time: 27 s
OOOOCurrent receiver time: 28 s
Current receiver time: 29 s
OOOOOOOCurrent receiver time: 30 s
OOOOOCurrent receiver time: 31 s
Current receiver time: 32 s
OCurrent receiver time: 33 s
OOCurrent receiver time: 34 s
OOOCurrent receiver time: 35 s
OOCurrent receiver time: 36 s
OCurrent receiver time: 37 s
OCurrent receiver time: 38 s
OOOCurrent receiver time: 39 s
OOOOOOCurrent receiver time: 40 s
OOOOOOOOOOOOOOOOCurrent receiver time: 41 s
OOOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 42 s
OOOOOOOOOOOOOOOOOOOOCurrent receiver time: 43 s
OOOOOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 44 s
OOOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 45 s
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 46 s
OOOOOOOOOOOOOOCurrent receiver time: 47 s
OOOOOOOOOOOOOOOCurrent receiver time: 48 s
OOOOOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 49 s
OOOOOOOOOOOOOOOCurrent receiver time: 50 s
OOOOOOOOOOOO

About the Freq_Xlatin_Fir_Filter settings, I've been reading the code and I do not exactly what parameters I should pay attention. Maybe the cut frequency?
> So, if you download an app like [GPS Test](https://play.google.com/store/apps/details?id=com.android.gpstest&hl=en_GB&gl=US) before you capture a sample look at the skyplot
> [image]
>
> Pick a satellite [...]

ZiglioUK, 2021-02-09:
The fact PRN wasn't acquired shows that the HackRF setup you have there is not sensitive enough (or that the clock is so biased to be outside the +/- 1KHz range). What does your phone think the CN0 for PRN 15 should be?
You can keep trying by slightly increasing the pfa and perhaps going as far as 2Khz until gnss-sdr can acquire that satellite (you know it's there).
For further increasing the chances of acquisition, you can use `Acquisition_1C.max_dwells=10`

Anyway, lots of discussions here on GitHub of people having poor success with the HackRF. Why do you insist on it when a rtl-sdr stick with a good TCXO and bias-t could give you better results?

 As far as the filter goes, you could simply skip it:
```
;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through
```

More threads of discussion:
https://github.com/gnss-sdr/gnss-sdr/issues/443
https://github.com/gnss-sdr/gnss-sdr/issues/435
https://github.com/gnss-sdr/gnss-sdr/issues/460

Bilelhm98, 2021-02-09:
Sorry, it was my wrong. I had configurated the file to only find the PRN 15. Now it finds more channels but the result is the same than before.

I use the HackRF because it is the equipment I have. I've already read those threads. Thank you so much for your help! I'll keep reading more about it and I hope I get to solve it.

Best Regards

> The fact PRN 15 wasn't acquired shows that the HackRF setup you have there is not sensitive enough (or that the clock is so biased to be outside the +/- 1KHz range). What does your phone think the CN0 for PRN 15 should be?
> You can keep trying by slightly increasing the pfa and perhaps going as far as 2Khz until gnss-sdr can acquire that satellite (you know it's there).
> For further increasing the chances of acquisition, you can use `Acquisition_1C.max_dwells=10`
> 
> Anyway, lots of discussions here on GitHub of people having poor success with the HackRF. Why do you insist on it when a rtl-sdr stick with a good TCXO and bias-t could give you better results?
> 
> As far as the filter goes, you could simply skip it:
> 
> ```
> ;######### INPUT_FILTER CONFIG ############
> InputFilter.implementation=Pass_Through
> ```
> 
> More threads of discussion:
> #443
> #435
> #460