gnss-sdr/gnss-sdr #481: build using incorrect UHD include

cinaeds, 2021-02-27:
There appears to some confusion between using the paths 

  GNURADIO_UHD_INCLUDE_DIRS:PATH=/opt/gnuradio/include

and 

  UHD_INCLUDE_DIRS:PATH=/opt/uhd/include

in the build process.
  
In the first part of the build, it needs the UHD path

   GNURADIO_UHD_INCLUDE_DIRS:PATH=/opt/gnuradio/include

to find GNURADIO's api.h file. 

In the last part of the build - at the point where it crashes and burns - it needs the UHD path 

   UHD_INCLUDE_DIRS:PATH=/opt/uhd/include

to find UHD's config.hpp file.   

But if can't find it because it's searching in GNURADIO_UHD_INCLUDE_DIRS:PATH instead of the UHD_INCLUDE_DIRS:PATH. 

And PKG_CONFIG_PATH finds the correct include paths - it's how the paths are being feed  to the build process that's the problem. 

When the build crashes and burns, I can hack the CMakeCache.txt file, and change

  GNURADIO_UHD_INCLUDE_DIRS:PATH=/opt/uhd/include 

so it pointing to the UHD_INCLUDE_DIRS:PATH and the build will finish and I can install it without any further errors.

-- Cinaed

-------------------------------------------- error log ---------------------------------------------------------------------
 91%] Building CXX object src/core/receiver/CMakeFiles/core_receiver.dir/control_thread.cc.o
[ 91%] Building CXX object src/core/receiver/CMakeFiles/core_receiver.dir/file_configuration.cc.o
[ 91%] Building CXX object src/core/receiver/CMakeFiles/core_receiver.dir/gnss_block_factory.cc.o
In file included from /opt/gnuradio/include/gnuradio/uhd/usrp_block.h:27,
                 from /opt/gnuradio/include/gnuradio/uhd/usrp_source.h:26,
                 from /opt/gnss-sdr/src/gnss-sdr/src/algorithms/signal_source/adapters/uhd_signal_source.h:24,
                 from /opt/gnss-sdr/src/gnss-sdr/src/core/receiver/gnss_block_factory.cc:150:
/opt/gnuradio/include/gnuradio/uhd/api.h:25:10: fatal error: uhd/config.hpp: No such file or directory
 #include <uhd/config.hpp>
          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/core/receiver/CMakeFiles/core_receiver.dir/build.make:89: src/core/receiver/CMakeFiles/core_receiver.dir/gnss_block_factory.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2741: src/core/receiver/CMakeFiles/core_receiver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
root@alfven:/opt/gnss-sdr/src/gnss-sdr/build#

cinaeds, 2021-02-28:
I forgot the version - the version was GNSS-SDR v0.0.14.git-next-10fc0eb62. 

And versions  for gnuradio was 3.8.2, gr-osmosdr was 0.23, and for UHD software was v4.0.0.0-rc1. 

All the sofware was the latest pull at the fime from the corresponding github sources. 

The OS was Debian 10.8.

carlesfernandez, 2021-02-28:
Thanks @cinaeds for reporting this issue. Please let us know if commit a669ecb solves it for you.

cinaeds, 2021-03-02:
Yes, it works! Great - thank you!

commit a669ecbf (2021-02-28): Include UHD path in the public interface of signal_source_adapters (fixes: #481)
closed 2021-03-03 by carlesfernandez