gnss-sdr/gnss-sdr #490: Can Almanac Data be Streamed Over UDP?

jt3232, 2021-03-29:
The project I'm working on also uses the Almanac data. Is it currently possible to get access to this data? If so, could it be sent over UDP?

Thanks, you guys are doing a fantastic job.

Jaxon

carlesfernandez, 2021-04-05:
Nope, the transmission of Almanac data over UDP is not implemented :disappointed:

jwmelto, 2021-04-06:
Could it be sent? Anything is possible...

carlesfernandez, 2021-04-06:
Of course, with some development.

The almanac reading in GPS is not implemented. This should be done at https://github.com/gnss-sdr/gnss-sdr/blob/next/src/core/system_parameters/gps_navigation_message.cc ( see TODOs in lines 282 and 315). Then, we would need to define a .proto file for the almanac (as it is done for ephemeris, see https://github.com/gnss-sdr/gnss-sdr/tree/next/docs/protobuf ), define a serializer (equivalent to what it is done at https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/PVT/libs/serdes_gps_eph.h ), and the UDP sink (e.g, https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/PVT/libs/monitor_ephemeris_udp_sink.h)

jt3232, 2021-12-10:
> Of course, with some development.
>
> The almanac reading in GPS is not implemented. This should be done at [...]

We have attempted to implement the Alamanac streaming code, however, we have ran into a few software issues.

On, some of the files, it mentions that the protobuf files (gps_almanac.pb.cc and the .h) should be automatically generated when the code is compiled. However, this does not appear to be true and it ends up throwing a file not found error during compiling. So I manually ran the protoc commands to generate the files and placed them in the proper place in the "build" folder. However, it appears that maybe the make or cmake files do not list these when the software compiles as we get errors saying "set_IODA" is not defined when it is indeed defined in the gps_almanac.pb.cc and .h files.

We run the following to compile the software from the build directory:

`cmake -DENABLE_OSMOSDR=ON ../
make`

We created the gps_almanac.proto file, so it doesn't currently exist in the public version anywhere. The generated pb.cc and pb.h files look good too.

Have we made a mistake somewhere?

Thanks,

Jaxon