gnss-sdr/gnss-sdr #491: can't find gr-iio

black-pigeon, 2021-04-02:
When I make this GNSS-SDR project to support FMCOMMS2 or PLUTO SDR, it's shows the error blow. I have GNU  Radio 3.8.2.0 installed on my laptop.
> [ 81%] Building CXX object src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/fmcomms2_signal_source.cc.o
In file included from /home/wcc/GNSS/gnss-sdr/src/algorithms/signal_source/adapters/fmcomms2_signal_source.cc:19:
/home/wcc/GNSS/gnss-sdr/src/algorithms/signal_source/adapters/fmcomms2_signal_source.h:26:10: fatal error: gnuradio/iio/fmcomms2_source.h: No such file or directory
 #include <gnuradio/iio/fmcomms2_source.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/build.make:134: recipe for target 'src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/fmcomms2_signal_source.cc.o' failed
make[2]: *** [src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/fmcomms2_signal_source.cc.o] Error 1
CMakeFiles/Makefile2:1831: recipe for target 'src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/all' failed
make[1]: *** [src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I also have a virtue machine which have a GNU Radio 3.7.9 installed on it. I alse build the project, and it seems to no error when I enable FMCOMMS and PLUTO SDR
```bash
$ cd gnss-sdr/build
$ cmake -DENABLE_FMCOMMS2=ON ..
$ make
$ sudo make install

$ cmake -DENABLE_PLUTOSDR=ON ..
$ make
$ sudo make install
```
Is the GNU Radio or gr-iio version caused this problem?

carlesfernandez, 2021-04-04:
Hi @black-pigeon 

Thanks for spotting this issue. This was a bug in the CMake script that was showing up if CMake was re-run several times. Now this should be fixed in the next branch by commit 09b4c8f7a32c044d1eb1701acef6d1b5aba3a080

black-pigeon, 2021-04-04:
Thanks for your nice work.

carlesfernandez, 2021-04-05:
I'm closing the issue since the problem seems solved.

commit 09b4c8f7 (2021-04-04): Fix bug in searching for gr-iio when CMake is re-run several times (fixes #491)
closed 2021-04-05 by carlesfernandez