gnss-sdr/gnss-sdr #492: Failed to install the gr-osmosdr

dengxu3, 2021-04-07:
Hi, I successfully build and install the latest version of GNSS-SDR by installing dependencies using software packages.

My computer OS is Ubuntu 18.04.5 LTS 64-bit.

For using the LimeSDR as a RF signal input, as I known, to build OSMOSDR support is needed, but some fatal errors occured when I followed the instructions in README.md  "install the OsmoSDR library and GNU Radio's source block". 

Error details are below:
`$ cmake ../ -Wno-dev
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   chrono
--   thread
--   system
--   date_time
[... 193 lines cut ...]

CMake Error at lib/CMakeLists.txt:51 (add_library):
  Target "gnuradio-osmosdr" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/sys503/dx/gnss-sdr/src/osmo-sdr/software/gr-osmosdr/build
$ make 
Scanning dependencies of target gnuradio-osmosdr
[  3%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o
In file included from /home/sys503/dx/gnss-sdr/src/osmo-sdr/software/gr-osmosdr/lib/source_impl.cc:36:0:
/osmo-sdr/software/gr-osmosdr/lib/fcd/fcd_source_c.h:25:10: fatal error: fcdproplus/fcd.h: no such file or content
 #include <fcdproplus/fcd.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o] Error 1
CMakeFiles/Makefile2:152: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
`

Any advice will be appreciated.

dengxu3, 2021-04-11:
It is no big deal. I found it is no need to build OSMOSDR support since it already built well in the next branch.
Do not update the GNU radio to 3.9.

closed 2021-04-11 by dengxu3

SpaceJacket19, 2022-04-10:
@dengxu3 I'm having the same issue. What exactly did you do to get it to run? Did you just build the next branch of gnss-sdr and not worry about osmoSDR at all? Any help is much appreciated!