gnss-sdr/gnss-sdr #493: rtl sdr version 3 with gnss sdr fails to provide a solution..

nalinsaxenapesu, 2021-04-11:
I am using a gps active antenna, with the bias tee turned on. I have attached my config file as well please see it and help if possible.There is almost immediate loss of tracking and no navigation solution is obtained. I even tried to use assisted gnss,
Please help.
Thanks

[config.txt](https://github.com/gnss-sdr/gnss-sdr/files/6291295/config.txt)

[image]

matteocutugno, 2021-04-11:
Ho @nalinsaxenapesu 

I shared some data logged with GNSS-SDR.
You can find them at 

https://www.researchgate.net/project/Utilizzo-di-un-ricevitore-software-GNSS-per-il-posizionamento-di-codice

They include a working configuration file.

nalinsaxenapesu, 2021-04-11:
Hi matteo,

Thanks for replying.. I tried with this config file as well, but it still
doesn't work. Any other suggestions?

nalinsaxenapesu, 2021-04-11:
Alright so I started getting navigation messages, but still no position is obtained...
Also can i log these navigation messages and see it later?
[image]

matteocutugno, 2021-04-11:
To obtain positions you have to receive at least four complete navigation messages from four different satellites. According to the snapshot you provided you are tracking just three.

nalinsaxenapesu, 2021-04-11:
You mean 4 set of consecutive navigation messages together right?

matteocutugno, 2021-04-11:
Correct.

Scarica Outlook per Android<https://aka.ms/AAb9ysg>

nalinsaxenapesu, 2021-04-11:
is there any way i can log these navigation messages?

nalinsaxenapesu, 2021-04-11:
Also i am getting these ,dat files, how do i  open these files and check
the data?
[image: Screenshot from 2021-04-11 17-30-02.png]

FayeWong98, 2022-04-28:
@nalinsaxenapesu 
Hello,I encountered the same problem of losing locks,and I used bias tee,can you give some suggestions?