gnss-sdr/gnss-sdr #496: Dudas GNSS-SDR

Hassio108, 2021-04-18:
Buenas tardes,

me llamo Diego y soy un estudiante de Telecomunicaciones en la Universidad Politécnica de Madrid.

Estoy planteándome usar el software _GNSS-SDR_ para realizar ciertas medidas para mi Trabajo Fin de Máster  (TFM) y querría preguntarle algunas dudas que tengo.

En primer lugar, quisiera saber si necesito una antena GNSS si empleo el _ADALM PLUTO_ como dispositivo SDR (es decir, como 'signal source'). He estado probando con las que trae el _PLUTO_ pero no consigo que se enganche a ningún satélite (No para de salirme este mensaje: “_Loss of lock in channel X!_”). He leído en varios foros que puede deberse a los parámetros de ancho de banda del _PLL_ y del _DLL_. He probado a reducirlos y a aumentarlos pero no consigo nada. La señal que estoy intentando recibir es la de GPS banda L1. ¿Puede indicarme cuál puede ser mi error(es)/problema(s)? (Quiero aclararle que lo he usado en distintos lugares en exterior).

Y en segundo lugar, me gustaría preguntarle si el software está actualizado para la última versión de _MacOS_ (_Big Sur_). He probado a instalarlo tanto usando “_Macports_" o “_Homebrew_" como  "instalándolo desde la fuente” (con todas las librerías necesarias) pero siempre me acaba dando algún error. ¿Puede ser que las librerías no estén actualizadas para esta versión de _MacOS_?

Quiero concluir diciéndole que el _GNSS-SDR_ me parece un software  muy completo y con muchas posibilidades y que espero poder usarlo para la realización de mi TFM.

Gracias por su atención y quedo a la espera de su respuesta.

Un saludo,

Diego.

jwmelto, 2021-04-18:
Diego,

My Spanish is rusty and not up to the task, but Google Translate did a credible job on your message. Hopefully you can do the same, if your English is poor.

I suggest looking at #471 and also particularly #376 for hints on your loss of lock problem.

I haven't tried either of the MacOS packagers for GNSS-SDR; my general preference is homebrew. If you have errors building from source, it would be helpful to know what they are.

Good luck with your TFM.

acebrianjuan, 2021-04-19:
Hola Diego (@Hassio108),

> En primer lugar, quisiera saber si necesito una antena GNSS si empleo el _ADALM PLUTO_ como dispositivo SDR (es decir, como 'signal source'). He estado probando con las que trae el _PLUTO_ pero no [...]

Sí, en general las señales GNSS son tan débiles que para poder recibirlas necesitas usar una [antena GNSS](https://www.amazon.es/Adafruit-GPS-Antenna-External-Active/dp/B01M0QB38R) con [low-noise amplifier (LNA)](https://en.wikipedia.org/wiki/Low-noise_amplifier) que debes alimentar con corriente contínua (DC) mediante un circuito [bias-T](https://www.onesdr.com/2020/01/19/what-is-a-bias-tee/) (o _"bias tee"_). Algunos SDRs traen de serie el circuito bias-T integrado en la PCB y se activa facilmente por software (como puedes leer [aquí](https://www.onesdr.com/2020/02/07/software-defined-radios-and-bias-tees/)), pero no es el caso del ADALM PLUTO. Es decir, si quieres seguir adelante con el ADALM PLUTO, a parte de la antena GNSS vas a necesitar hacerte con un bias-T y una fuente de alimentación DC.

Si tienes la opción de cambiar de SDR, una opción muy popular y económica es el dongle [RTL-SDR Blog V3](https://www.astroradio.com/p/receptor_rtlsdr_blog_r820t2_rtl2832u/) que, por experiencia propia, siempre me ha funcionado muy bien.

> He leído en varios foros que puede deberse a los parámetros de ancho de banda del _PLL_ y del _DLL_. He probado a reducirlos y a aumentarlos pero no consigo nada. La señal que estoy intentando [...]

Realmente, hasta que no resuelvas el problema del hardware, el ajuste de los anchos de banda no va a tener efecto alguno. Además creo que es poco probable que vayas a obtener resultados satisfactorios dado que, según la [ficha técnica](https://www.analog.com/media/en/news-marketing-collateral/product-highlight/ADALM-PLUTO-Product-Highlight.pdf), el oscilador del ADALM PLUTO tiene una estabilidad de frecuencia de ±25 ppm. Para recepción GNSS se necesitan estabilidades de < 1 ppm como se indica [aquí](https://gnss-sdr.org/conf/#required-equipment-1). Entonces, si pese a todo lo anterior todavía quisieras seguir adelante con el ADALM PLUTO, deberías hacerte con una señal de reloj externa de mayor estabilidad e injectarsela al PLUTO tal como se detalla [aquí](https://ez.analog.com/adieducation/university-program/f/q-a/77922/will-it-be-possible-to-feed-in-a-reference-clock-to-the-adalm-pluto).

> Y en segundo lugar, me gustaría preguntarle si el software está actualizado para la última versión de _MacOS_ (_Big Sur_). He probado a instalarlo tanto usando “_Macports_" o “_Homebrew_" como [...]

Para poder ayudarte con esto por favor indícanos los errores concretos que te estás encontrando.

Un saludo y suerte.
Álvaro

Hassio108, 2021-05-18:
Buenos días,

al final solucioné los problemas de instalación en _macOS_ del GNSS-SDR. 

También compré el **RTL-SDR Blog V3** (Nº Ped. : 693510) y funciona muy bien.

Os escribo, en primer lugar, para daros las gracias por la ayuda y, en segundo lugar, porque tengo varias dudas.

Primero, quisiera saber si con el dongle RTL-SDR es posible recibir de dos constelaciones al mismo tiempo o necesito un RTL con dos receptores. La idea sería comparar, por ejemplo, el funcionamiento de GPS + Galileo con el de GPS o Galileo por separado, es decir, las mejoras de emplear multi-constelación.

Por otro lado, cuando configuro el RTL para recibir Galileo, a la salida se crean varios archivos, entiendo que _RINEX_, con las extensiones: ‘_._F’, ‘._P’ y ‘._B’, ‘._G’…_ y no sé qué son. No sé si tiene que ver con algún tipo de corrección.

También me gustaría saber en qué momento se realiza la **conversión a Banda Base** (BB). Entiendo que el RTL no trabaja a frecuencia de Downlink del satélite y quisiera saber si hay algún cambio a frecuencia intermedia o si directamente va a BB y dónde se realiza.

Por último, me gustaría saber cómo puedo hacer funcionar la **asistencia online con Galileo**. Imagino que es igual que la que proponéis para GPS pero aún así (para GPS) no me funciona. No sé si estoy poniendo mal los servers/ports… Soy de Madrid y mi compañía es Orange. He probado la web: [https://www.mcc-mnc.com] y la aplicación movil que proponéis en la documentación pero no consigo que se conecte. No sé si necesitareis más datos para esta incidencia.

De nuevo, muchas gracias por su atención y ayuda y quedo a la espera de su respuesta.

Un saludo,

Diego.

ZiglioUK, 2021-05-18:
> Buenos días,
> 
> al final solucioné los problemas de instalación en _macOS_ del GNSS-SDR.
> 
> También compré el **RTL-SDR Blog V3** (Nº Ped. : 693510) y funciona muy bien.

Conseguiste utilizar el rtl-sdr dongle en MacOS? nunca he conseguido yo mismo.

> Os escribo, en primer lugar, para daros las gracias por la ayuda y, en segundo lugar, porque tengo varias dudas.
> Primero, quisiera saber si con el dongle RTL-SDR es posible recibir de dos [...]

En teoria podrias recebir GPS y Galileo con un receptor porque se encuentran en la misma frecuencia.
Lo que pasa es que el ancho de banda del rtl-sdr es solamente 2.4MHz, entonces muy poco para Galileo que necesitaria como 4MHz por lo menos. A lo mejor podrias receibir algun satelite, [con una perdida  de señal de unos dB](https://dsp.stackexchange.com/a/43985/4524).
El mismo problema saldría con multiplos receptores. No habrían problemas con Glonass por ejemplo, con banda mas estrecha.

> Por otro lado, cuando configuro el RTL para recibir Galileo, a la salida se crean varios archivos, entiendo que _RINEX_, con las extensiones: ‘_._F’, ‘._P’ y ‘._B’, ‘._G’…_ y no sé qué son. No sé si [...]

La señal por un rtl-sdr es [de banda base, aka IF 0](http://www.pa3fwm.nl/technotes/tn20.html).

Emanuele

Hassio108, 2021-05-19:
Hola Emanuele,

entiendo que re refieres a que no puedo recibir GPS y Galileo al mismo tiempo, pero por separado no tengo problema. De hecho, he conseguido recibir sólo Galileo con el RTL-SDR pero no consigo que se guarde el fichero de Navegación RINEX y no sé porqué (se guardan únicamente el de Observación y un '._L'). No sé porqué.

Un saludo,
Diego.

ZiglioUK, 2021-06-11:
Hola Diego, acabo de conseguir fixes en real time con 3 GPS + 3 GAL satelites a una vez en tiempo real con el dongle (<1ppm TCXO) :-)
A ver el ejemplo que he copiado aquì abajo.

Para el Rinex, solamente hay que añadir la configuracion, y esperar que salgan unas posiciones.
Cuidado que termines el gnss-sdr con la tecla 'q', si no no guardarà los ficheros

[rtl-gps-gal.conf.txt](https://github.com/gnss-sdr/gnss-sdr/files/6637748/rtl-gps-gal.conf.txt)

Hassio108, 2021-06-13:
Hola Emanuele,
Muchas gracias!!!

lo acabo de probar y también he conseguido fix. Estoy intentando guardar los observables en un '.mat' pero me sale vacio completamente. La idea es analizar los resultados (Doppler, pseudorangos, SNR...) con MATLAB pero no consigo que se guarden. 
Para ello simplemente añado estas líneas:
```
;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true

```
No se si tú puedes guardar esta información de los observables.

Espero tu respuesta.

Muchas gracias y un saludo,
Diego.

ZiglioUK, 2021-06-16:
Diego

el Rinex lo consiques no? eso lleva todas esas informaciones
```
PVT.output_enabled=true
PVT.rinex_output_enabled=true

```

Hay parsers en Matlab for Rinex en [goGPS](https://github.com/goGPS-Project/goGPS_MATLAB) o creo en el [Google GPS Measurement Tools ](https://github.com/google/gps-measurement-tools)(o tambien en el gnss-sdr mismo)

Hassio108, 2021-06-21:
¡Solucionado!

¡Muchas gracias a todos por la ayuda!

closed 2021-06-21 by Hassio108