gnss-sdr/gnss-sdr #497: Unable to get Position fix Ad9361 FPGA signal source

noelpedro, 2021-04-22:
Hi guys I have 7 correlators  and a sample counter on my hardware and was wondering why when I have 4 of the 7 correlators getting Navigation messages I’m still able to get a position fix.

I know on cold start might take a while.

Is there any additional configuration that can help to get a fix?

i was looking at rx_clock_correction in the PVT.

carlesfernandez, 2021-04-22:
Hi @noelpedro ,

when the receiver starts tracking a satellite signal, it needs to achieve bit synchronization and then decode some navigation pages until it gets the ephemeris data. When this process is completed for at least four channels simultaneously, then you should start getting PVT fixes.

If navigation pages are decoded correctly, you should see the messages in the console terminal. For GPS L1, you need at least pages 1, 2, and 3 to get the full ephemeris dataset.

Hope this helps,
Carles

noelpedro, 2021-04-29:
Hi Carles,

Yes Im reaching bit synchronization on at least 4 channels and hence the Navigation messages from all five subframes from the telemetary decoder.
  
By pages you also mean subframes correct?

Its weird that I'm getting NAV messages from the telemetary decoder for over 5 minutes on at least for channels and still no PVT.

zoqueton99, 2021-05-18:
Hello! I've the same problem, I recieve NAV messages and eph data but no PVT solution, could you find a solution?

noelpedro, 2021-05-20:
None yet, I've even attempted to ban known non-operational vehicles yet still no position.

noelpedro, 2021-06-05:
Is PVT affected by over sampling?? I'm doing this real time processing at 40 MSPS. I wonder if that is my issue.

noelpedro, 2021-06-10:
Found the issue I'm producing empty observables for some reason, going to keeep digging.

noelpedro, 2021-06-17:
Made some improvements but now getting an error in rtklib_solver.cc line 904

because result is 0

`
result = rtkpos(&amp;rtk_, obs_data.data(), valid_obs + glo_valid_obs, &amp;nav_data);
            if (result == 0)                {                    LOG(INFO) &lt;&lt; "RTKLIB rtkpos error: " &lt;&lt; rtk_.errbuf;                    rtk_.neb = 0;                  // clear error buffer to avoid repeating the error message                    this-&gt;set_time_offset_s(0.0);  // reset rx time estimation                    this-&gt;set_num_valid_observations(0);                }`

noelpedro, 2021-09-09:
Got position.

Sample counters were out of sync.

closed 2021-09-09 by noelpedro