gnss-sdr/gnss-sdr #500: Config Parameters for Slower Processors (Raspberry Pi 4 B)

jt3232, 2021-04-30:
Hi, I have a Raspberry Pi 4 B 8GB running GNSS-SDR on Ubuntu 21.04 Desktop. After running my original config (with 32 acquisition channels), the software would never be able to lock onto satellites. Switching the number of channels to 8 seemed to fix this problem and everything worked well, until it lost lock of all satellites and never regained its position solution.. I would ideally like to use 12 channels or more. With 8 not working, I wonder if there is some room for efficiency in my config file. Are there any parameters I can change to make this more stable/faster?

I might consider using the Raspberry Pi OS instead of Ubuntu. I am posting my current config file below:

`[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000
GNSS-SDR.telecommand_enabled=true
GNSS-SDR.telecommand_tcp_port=3333

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=50
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf=0,bias=0
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1c

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=30000
Acquisition_1C.doppler_step=150
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static
;output rate was 100
PVT.output_rate_ms=500
;display rate was 500
PVT.display_rate_ms=1000
PVT.iono_model=Broadcast
PVT.output_enabled=false
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2
PVT.enable_monitor=true
PVT.enable_protobuf=true
PVT.monitor_client_addresses=10.10.10.1
PVT.monitor_udp_port=1112
PVT.enable_monitor_ephemeris=true
PVT.monitor_ephemeris_client_addresses=10.10.10.1
PVT.monitor_ephemeris_udp_port=1113

;######### MONITOR CONFIG ############
Monitor.enable_monitor=true
Monitor.enable_protobuf=true
Monitor.decimation_factor=1
Monitor.client_addresses=10.10.10.1
Monitor.udp_port=1111`

Thanks fellas,

Jaxon

jwmelto, 2021-04-30:
I'll be honest, I don't exactly know what `Channels.in_acquisition=1` does, but I had a similar issue with tracking and I commented it out and got much better response. YMMV

jt3232, 2021-04-30:
> I'll be honest, I don't exactly know what `Channels.in_acquisition=1` does, but I had a similar issue with tracking and I commented it out and got much better response. YMMV

Yeah, this definitely made it worse. This parameter controls how many satellites are doing acquisition at the same time. 1 is good,. If commented out, it will default to the number of channels, meaning the CPU works even harder.

jt3232, 2021-04-30:
Setting signalsource_enable_throttle=true has helped. Running 12 channels feels pretty stable. I do have an issue of if it loses position but then restarts tracking satellites, it cannot get its position. Perhaps this is because the channels changed? I am not sure, but I would have figured it would correct itself after a fee seconds. Anyone have ideas about that?