gnss-sdr/gnss-sdr #507: Urgent!!!!!Jack problem while running the config file for rtl-sdr2838

aishwarya418112, 2021-05-19:
When I run the rtl-sdr config file, the content of which is given in gnss-sdr.org, I get the following error. My laptop is able to recognise the sdr but I still get this error. Please help me with this as soon as possible.

 _gnss-sdr --config_file=./rtl-gps.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.14.git-next-6dedcb6c3 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
gr-osmosdr 0.1.4 (0.1.4) gnuradio v3.8.2.0-118-g8b2012ab
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[INFO] Using format CF32.
Segmentation fault (core dumped)_

jwmelto, 2021-06-09:
Hi @aishwarya418112 

You should not assume that anyone knows what your config file contains. If you could either provide a link to the file you are using, or attach it, it might be easier to help.

> RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

This is a pretty good indication of the problem. It looks to me like something else has the device open, so GNSS-SDR fails. You mentioned that your laptop can recognize it ... do you have some other application running, talking to it?

> jack server is not running or cannot be started

I'll admit, I know nothing about this device, but this is also a problem, I would think.

carlesfernandez, 2021-06-10:
Hi @aishwarya418112 

Maybe you need to set some udev rules. From https://www.instructables.com/rtl-sdr-on-Ubuntu/ :

> You need to add some udev rules to make the dongle available for the non-root users. First you want to find the vendor id and product id for your dongle.
>
>The way I did this was to run:
>```
>lsusb
>```
>The last line was the Realtek dongle:
>```
>Bus 001 Device 008: ID 0bda:2838 Realtek Semiconductor Corp.
>```
>The important parts are "`0bda`" (the vendor id) and "`2838`" (the product id).
>
>Create a new file as root named `/etc/udev/rules.d/20.rtlsdr.rules` that contains the following line:
>```
>SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838", GROUP="adm", MODE="0666", SYMLINK+="rtl_sdr"
>```
>With the vendor and product ids for your particular dongle. This should make the dongle accessible to any user in the adm >group. and add a /dev/rtl_sdr symlink when the dongle is attached.
>
>It's probably a good idea to unplug the dongle, restart udev (`sudo restart udev`) and re-plug in the dongle at this point.

Hope this helps!