gnss-sdr/gnss-sdr #509: PlutoSDR : Loss of lock in channel

Arunk0548, 2021-05-25:
I'm trying to get coordinates with GNSS-SDR (next branch) and PlutoSDR as source of signal. But there is always Loss of lock in channel after few Seconds.

I'm using below Configuration file for PlutoSDR. Also find the attached DEBUG logs. 

NOTE: With RtlSDR as different signal source & *.conf, I'm able to get LAT/LNG and its working very consistent.  

Thanks in advance for your input and suggestion.

`; You can define your own receiver and invoke it by doing
; gnss-sdr --config_file=my_GNSS_SDR_configuration.conf
;

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_hz: Internal signal sampling frequency after the signal conditioning stage [Hz].
;FOR USE GNSS-SDR WITH RTLSDR DONGLES USER MUST SET THE CALIBRATED SAMPLE RATE HERE
; i.e. using front-end-cal as reported here:http://www.cttc.es/publication/turning-a-television-into-a-gnss-receiver/
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Plutosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.device_address=192.168.2.1
SignalSource.sampling_frequency=3000000
SignalSource.freq=1575420000
SignalSource.bandwidth=2600000
SignalSource.decimation=0
SignalSource.gain_mode=manual
SignalSource.gain=30
SignalSource.samples=0
SignalSource.buffer_size=65000
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=./capture.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=4000000
Resampler.sample_freq_out=2000000
Resampler.item_type=gr_complex

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=4
Channels.in_acquisition=1
;Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=100
;Acquisition_1C.pfa=0.01
Acquisition_1C.threshold=2.38
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=acq_dump

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=2.0;
;Tracking_1C.fll_bw_hz=5.0
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=true
Tracking_1C.order=3
Tracking_1C.early_late_space_chips=0.5

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=GPS_L1_CA_Observables

;######### PVT CONFIG ############
PVT.implementation=GPS_L1_CA_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500
`

Console Output as below:
`Initializing GNSS-SDR v0.0.14.git-next-786feddd1 ... Please wait.
Logging will be written at .
device address: 192.168.2.1
frequency : 1575420000 Hz
sample rate: 3000000 Hz
gain mode: manual
item type: gr_complex
RF Channels: 1
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 14 (Block Decommissioned)
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 03 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 18 s
Current receiver time: 19 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
Loss of lock in channel 0!
Loss of lock in channel 1!
Current receiver time: 31 s
Current receiver time: 32 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 33 s
Loss of lock in channel 3!
Current receiver time: 34 s
Current receiver time: 35 s
Loss of lock in channel 2!
Current receiver time: 36 s
Current receiver time: 37 s
Current receiver time: 38 s
Current receiver time: 39 s
Current receiver time: 40 s
Current receiver time: 41 s
Current receiver time: 42 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 13 (Block IIR)
Current receiver time: 43 s
Current receiver time: 44 s
Current receiver time: 45 s
Current receiver time: 46 s
Current receiver time: 47 s
Current receiver time: 48 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 49 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 50 s
Current receiver time: 51 s
Quit keystroke order received, stopping GNSS-SDR !!
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 66.992 [seconds]
GNSS-SDR program ended.
`

[gnss-sdr.pluto.pluto.log.INFO.20210525-185851.txt](https://github.com/gnss-sdr/gnss-sdr/files/6539655/gnss-sdr.pluto.pluto.log.INFO.20210525-185851.txt)

AgileEngineeringLLC, 2021-06-02:
How are you powering your antenna? Are you using an external bias tee?

Arunk0548, 2021-06-03:
Yes, we are using a bias tee.  The external LNA gain is about 20db.

Also, When we feed it to RTL SDR Dongle(USB), Everything works fine and able to get LAT/LNG.