gnss-sdr/gnss-sdr #515: Two_Bit_Packed_File_Signal_Source doesn't output the correct data type

Rigel09, 2021-06-09:
When using ```Two_Bit_Packed_File_Signal_Source```  the output never seems to match the input that ```Signal Conditioner``` is expecting. I get an item mismatch between signal source and signal conditioner.

```
Can't connect SignalSource to SignalConditioner : itemsize mismatch: valve0:0 using 8, interleaved_char_to_complex0:0 using 1
Unable to connect flowgraph
```

Here are my settings for the conf file. 
```
;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Two_Bit_Packed_File_Signal_Source
SignalSource.filename=/Path/To/File
SignalSource.item_type=byte
SignalSource.sampling_frequency=60000000
SignalSource.sample_type=iq

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ibyte_To_Complex
InputFilter.implementation=Pass_Through
Resampler.implementation=Pass_Through
```

Any help is appreciated.

jwmelto, 2021-06-09:
It's entirely possible that I broke this. The two-bit file sources are interesting. 

- There is `.item_type` that drives the input. This defaults to "byte". This defines the size of a single input item. The only supported values are "byte" and "short" (although I have my doubts about "short" in little-endian architectures)
- There is `.sample_type` that defines whether the input is raw amplitude ("real"), or I/Q ("iq" or "qi")
- There is also `.big_endian_items` (default to `true`) that affects how "short" items are read. I think it tells if the data is stored in Big Endian form (network order) or not.
- There is also `.big_endian_bytes` (default to `false`). This flag and the prior one are passed to the `unpack_2bit_samples` block constructor

The 2-bit file source inserts an "unpack" block followed by a "char to float" block between the file reader and the signal conditioner. So the output of this file source is either float or complex. Since you specified "iq", it will be complex.

It looks like your signal conditioner is superfluous; the samples are already complex coming out of the signal source. What if you just make your SignalConditioner `Pass_Through`?

Rigel09, 2021-06-10:
Thanks for the reply.

I gave that a shot and it failed as well. Here is the error it gave.

'''
* The Signal Source implementation Two_Bit_Packed_File_Signal_Source has an output with a SignalSource.item_size of 1 bytes, but it is connected to the Signal Conditioner implementation Pass_Through with input item size of 8 bytes.
     Output ports must be connected to input ports with the same item size. 
'''

Here is the new config file that I used to get that error.

'''
;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Two_Bit_Packed_File_Signal_Source
SignalSource.filename=/Path/To/File
SignalSource.item_type=byte
SignalSource.sampling_frequency=60000000
SignalSource.sample_type=iq

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through
DataTypeAdapter.implementation=Pass_Through
InputFilter.implementation=Pass_Through
Resampler.implementation=Pass_Through
```

Rigel09, 2021-06-10:
Also, when reading the [documentation](https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-two_bit_packed_file_signal_source)  it seems that ```Two_Bit_Packed_File_Signal_Source``` will output either a float or gr_complex. When I run it with byte input it seems to be outputting it in bytes as well. Out of curiosity, I changed ```item_type``` to short and the Signal Source started outputting in 2 Bytes in the previous error message. From my understanding of the error, it seems that it is not outputting in the gr_complex form as it should be? I'm still new to GNSS-SDR so it's entirely possible that it is something on my part as well. I'm not for sure if this adds any helpful value to this problem explanation but figured I would mention it either way.

jwmelto, 2021-06-10:
That would be a bug.

Rigel09, 2021-06-10:
Do you have any suggestions on how to proceed from here? Would you be able to point to a stable commit or maybe to the area of GNSS-SDR's codebase that I could start looking in order to fix it and maybe create a pull request? I'm not very familiar with GNSS-SDR and get lost easily when looking through the codebase.

jwmelto, 2021-06-11:
@Rigel09 if you can pull my branch, from #517 and let me know if that solves your problem. You should use a pass-through signal conditioner, unless you want particular filtering.

Rigel09, 2021-06-14:
@jwmelto Sorry that it's been a couple of days since I could look at this. So here is a couple of things.

1. I pulled down your branch on #517 and it didn't change anything. Looks like you may already know that since you deleted it. 

2. I pulled down your other branch that has already been merged in #518. It did allow me to run a complex two-bit file but didn't allow real two-bit files. It always lost lock on all of the satellites it started tracking. I found out the cause of this is due to how it is reading in samples. The reason I say this is because GNSS-SDR stopped at what should have been 1/3 of the way through the file.
This can be referenced by this printout.

```Processing file , which contains 1811481984 samples (3622963968 bytes)```

The first thing that I noticed was that ```packetsPerSample()``` return 0.5 instead of 4. I changed the following code in ```two_bit_packed_file_signal_source.cpp``` so that this was temporarily hardcoded. 
```c++
double TwoBitPackedFileSignalSource::packetsPerSample() const { return 8 / 2.0; }
```
This increased my run time by twice as long as it should be.  I got the following printout. 
```Processing file  which contains 14491855872 samples (3622963968 bytes)```
From that printout, it almost seems as though it is pulling in the IQ data as regular data maybe?

Thanks for your help thus far.

Rigel09, 2021-06-14:
The ```real ``` implementation is also returning the incorrect size. I wasn't able to get very far down that path.

jwmelto, 2021-06-15:
> `* The Signal Source implementation Two_Bit_Packed_File_Signal_Source has an output with a SignalSource.item_size of 4 bytes, but it is connected to the Signal Conditioner implementation Pass_Through with input item size of 8 bytes.
     Output ports must be connected to input ports with the same item size. `

That one is on you. Ultimately, GNSS-SDR needs I/Q to run. If you select a "real" output, you are saying that the data file is raw amplitude data. In that case, you would need to do whatever signal processing step turns "real" into complex. In that case, you'll need something other than a pass-through signal conditioner.

The 2-bit file code is a bit odd. If you've been in the code, you've seen my comments. Obviously, what you get out will very much depend upon what you started with as input. In particular, the "short" `.item_type` looks very suspicious to me. There is a head-nod as to the vagaries of reading in 16-bit quantities on a little-endian system, but I just can't accept that it's OK to read 16 bits in one architecture and only read 8 in the other.

To your point:
```c++
double TwoBitPackedFileSignalSource::packetsPerSample() const { return item_size() / 2.0; }
```
hmmm. You are on to something. `item_size()` is the size in bytes, but this method assumes it is the size in bits. Pedantically, the proper computation would be `item_size() * CHAR_BIT / 2` but since POSIX [requires](https://stackoverflow.com/questions/6631491/why-did-posix-mandate-char-bit-8) that `CHAR_BIT==8`, I think we can reasonably simplify to `item_size() * 4`.

The file length is used to drive the **valve**, which limits the number of samples read. Based on code comments, this is defaulted to on, because there is (was?) a bug in GNU Radio that doesn't give indication when end of file is consumed, so we have no other way to shut down the application. 

The base class `FileSourceBase` tries to accommodate this in `computeSamplesInFile()` where it multiplies the number of packets (output samples) per (input) sample by the file length divided by the item size. This **should** produce the number of samples in the file, which is what the valve operates on. Since there does appear to be a bug in `packetsPerSample`, this would definitely reduce the number of samples you process.

On the other hand, for interleaved I/Q ("iq" or "qi") the packets per sample should be reduced by half, since it takes two packets to produce an output sample. That is not present in the implementation.

So it looks like the correct implementation should be
```c++
double TwoBitPackedFileSignalSource::packetsPerSample() const
{
    auto packets = item_size() * 4.0;
    if (is_complex()) packets /= 2;
    return packets;
}
```

If you want to check that out, it should give a more proper number of samples.

Do you have 2-bit files of both I/Q and real? If so, would you be willing to share them? I'm happy to run for sanity, so my code changes aren't untested.
 return item_size() / 2.0; }

Rigel09, 2021-11-11:
@jwmelto Sorry it's been so long since writing back. Life got busy and it slipped my mind. Unfortunately, the files don't belong to me and the person they do belong to doesn't want me posting them on the internet. That being the case I am still happy to test any recent builds. I just pulled the next branch and am attempting to build it now. Once I get it to build I will let you know how it goes.  Currently, the latest is giving me some build issues with Google Test.