gnss-sdr/gnss-sdr #520: PLL not locked with RTL-SDR Blog V3

fdgm0001, 2021-06-22:
Hi all. I'm using the dongle RTL-SDR Blog V3 to obtain a GPS position with GNSS-SDR. However, when I run the configuration file, I obtain the following result:
 
```
Initializing GNSS-SDR v0.0.14.git-next-d999a5e6a3 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: rtl,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.0.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 40.2 dB...
Actual Bandwidth: 0 [Hz]...
RF Channels: 1
```

You can see that the PLL isn't locked, so, what I can do to solve this? Besides, I would like to know if there are others thing that I should look at run GNSS-SDR with RTL-SDR Blog V3.

Thank you so much for your help.

ZiglioUK, 2021-06-22:
Some of those error messages don't actually mean much. I still managed to track with the same dongle. 
I've posted a [conf file here](https://github.com/gnss-sdr/gnss-sdr/issues/496#issuecomment-859488811), if you wanna try.

fdgm0001, 2021-06-22:
> Some of those error messages don't actually mean much. I still managed to track with the same dongle.
> I've posted a [conf file here](https://github.com/gnss-sdr/gnss-sdr/issues/496#issuecomment-859488811), if you wanna try.

I have been using your configuration file and it works well. However, I can obtain a position. I think that the reason why this occurs is because I only obtain GPS messages from 3 satellites:

```
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 23 (Block III)
New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 31 (Block IIR-M)
New GPS NAV message received in channel 2: subframe 3 from satellite GPS PRN 10 (Block IIF)
```

ZiglioUK, 2021-06-22:
Good ut helped. Definitely you need at least 4 satellites for a Static PVT fix. Just increase the number of GPS channels and make sure your antenna has a clear view of the sky

`Channels_1C.count=6
`

ValZapod, 2021-07-10:
> what I can do to solve this?

That is just regression bug that is fixed here in 3 stages.
https://github.com/librtlsdr/librtlsdr/commit/220d41c55ffe3f8b8bfb69f1611f5cd45d2b8676
https://github.com/librtlsdr/librtlsdr/commit/883229dfc65b9f2e19a02c349cc7598bd29ef797
https://github.com/librtlsdr/librtlsdr/commit/e315a41ec5814523caaa3d4ac8f4fb775f1e96e6
> Some of those error messages don't actually mean much.

This is false. It was indeed not locked, you can even check for what frequency. https://github.com/librtlsdr/librtlsdr/commit/c28886f89b24e4cc73541710b2d6e6c013ed90c5

The bug was very problematic. https://github.com/librtlsdr/librtlsdr/issues/57#issuecomment-678707860

closed 2021-09-17 by fdgm0001