gnss-sdr/gnss-sdr #524: Implementation using GnU-radio

nalinsaxenapesu, 2021-06-30:
Hey all I needed some help and guidance .
 I wanted to implement these steps and obtain the user position with an rtl-sdr and gnu radio itself?

can someone please help me regarding it?

Thanks
Please go through this document to get a better idea 
https://ieeexplore.ieee.org/document/8455475

lenhart, 2021-08-20:
Hi @nalinsaxenapesu 
some unqualified answer from a non-expert and not-involved-in-the-project person:

I think in general it would be a good idea if you added more information with your motivation to do so and how much time you want to spend on this.

GNSS-SDR uses gnu radio under the hood (only not in the UI sense, but programatically) and I think it supports rtl-sdr already, so you could use.

If you want to do something specific there are a couple options:
- write code/adapt gnss-sdr classes directly to fit your need
- tap a gnss-sdr signal processing block to redicrect the signal stream (eg.g to an externally running gnu-radio instance via a FIFO) 
- investigate what is missing and write (guessing here) appropriate yaml files and other missing components so that gnss-sdr modules can be used as gnu-radio blocks
- take the desired code chunks and stuff them into custom gnu-radio blocks to build a custom receiver

Don't know if there is anything among those to help you with your project. Your linked paper is very short and doesn't describe much and especially not in detail. Don't let it fool you that re-implementing this is a trivial project :-)

Cheers

nalinsaxenapesu, 2021-08-20:
> Hi @nalinsaxenapesu
> some unqualified answer from a non-expert and not-involved-in-the-project person:
>
> I think in general it would be a good idea if you added more information with your motivation [...]

jwmelto, 2021-08-30:
We still don't know what you want. Did you look here? https://gnss-sdr.org/my-first-fix/

nalinsaxenapesu, 2021-09-10:
> We still don't know what you want. Did you look here? https://gnss-sdr.org/my-first-fix/

Hey @jwmelto  I want to implement the process which gnss-sdr is doing via gnu-blocks,
If not the entire thing then at least acquisition stage via gnu radio and data captured via rtl sdr for a college project

jwmelto, 2021-09-10:
GNSS-SDR is implemented using gnuradio blocks. Are you saying you just want to run it?  Have you read the documentation? Have you tried the tutorial? How is this an issue for this project?