gnss-sdr/gnss-sdr #527: SIGSEGV, Segmentation fault when running  My first position fix example

a3one, 2021-07-15:
I have built the GNSS-SDR on a cyclone V board and when I run the example in the My first position fix I get the following error when I run the program in GDB

Thread 1 "gnss-sdr" received signal SIGSEGV, Segmentation fault.
0xb599655a in ?? () from target:/usr/lib/libfftw3f.so.3

a3one, 2021-07-15:
I have built the same source code on the x86 machine and it worked fine.

acebrianjuan, 2021-07-15:
Hi @a3one,

- What embedded operating system are you running on the Cyclone V board?
- What version of GNSS-SDR did you build?
- How did you install the dependencies?
- Can you paste the GDB backtrace when the segfault occurs?

Cheers,
Álvaro

a3one, 2021-07-15:
Hi Álvaro, 

the following are answers to provided questions.
- What embedded operating system are you running on the Cyclone V board?

I have used open embedded (oe-gnss-sdr gatesgarth). I have added the intel-fpga meta file (gatesgarth) and chosen the following options
```
PREFERRED_PROVIDER_virtual/kernel = "linux-socfpga"
PREFERRED_VERSION_linux-socfpga = "5.8%"

```

- What version of GNSS-SDR did you build?
I have built GNSS-SDR v0.0.14 

- How did you install the dependencies?
 I have built the SDK and the Linux image using (oe-gnss-sdr provided by gnss-sdr.org)

- Can you paste the GDB backtrace when the segfault occurs?
```
(gdb) bt
#0  0xb599655a in ?? () from /usr/lib/libfftw3f.so.3
#1  0x00d0a540 in ?? ()

```

regards

A3one

a3one, 2021-07-25:
I have manged to solve the issue by cross compiling the FFTW3 library again independent of the openembedded project. 

 But I have noticed the receiver is very  slow. 58s in receiver time took 30m in real time.

closed 2021-07-25 by a3one