gnss-sdr/gnss-sdr #528: Getting 4 Valid Observables, but still no Position please help

noelpedro, 2021-07-15:
Getting this error on my LOG file: point pos err();

Called from https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc 

`if (d_internal_pvt_solver->get_PVT(d_gnss_observables_map, false))`

https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/PVT/libs/rtklib_solver.cc

`result = rtkpos(&rtk_, obs_data.data(), valid_obs + glo_valid_obs, &nav_data);`

https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/libs/rtklib/rtklib_rtkpos.cc

`if (!pntpos(obs, nu, nav, &rtk->opt, &rtk->sol, nullptr, rtk->ssat, msg))
        {
            errmsg(rtk, "point pos error (%s)\n", msg);
            if (!rtk->opt.dynamics)
                {
                    outsolstat(rtk);
                    return 0;
                }
        }`

`;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.iono_model=OFF
PVT.trop_model=OFF
PVT.raim_fde=1
PVT.output_rate_ms=20;
PVT.display_rate_ms=500;
PVT.elevation_mask=0;
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false
PVT.dump_filename=./PVT`

carlesfernandez, 2021-07-28:
Try deactivating the RAIM algorithm, i.e., `PVT.raim_fde=0`.

noelpedro, 2021-07-29:
@carlesfernandez i believe I’ve tried that.

but I’ll double check. I was using the PVT configs from this .conf file https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/gnss-sdr_GPS_L1_FPGA.conf

noelpedro, 2021-08-04:
@carlesfernandez Is there a reason the PVT is attempted to be computed after only 3 Pages and not after the 5? The error above is happening after only the 3rd page.

noelpedro, 2021-08-10:
estpos() is returning zero

noelpedro, 2021-08-11:
Line 665 in https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/libs/rtklib/rtklib_pntpos.cc

 `if (norm_rtk(dx, NX) < 1e-4)` is Not true

noelpedro, 2021-08-12:
SHould the above error cause the tracking correlators to lose lock? For some reason shortly after seeing this error I lose lock of all 4 satellites.

noelpedro, 2021-08-18:
In `rescode()` in `rtklib_pntpos.cc`

`Elaux < opt->elmin` 

causing loop to continue and thus not updating `nv` 

I have elevation mask set to 0 can i make it negative?

noelpedro, 2021-09-03:
@carlesfernandez  I'm getting different TOW [ms] for all channels can this be the reason for failure in RTK? is there a config to allow for greater seperation in TOW when computing position?

noelpedro, 2021-09-03:
ALso my pseudoranges are very different. Some positive other negative.

noelpedro, 2021-09-09:
My tracking correlators respective sample counts were not synchronized with each other nor with that of the observables sample counter. I have fixed the issue and able to get PVT.

closed 2021-09-09 by noelpedro

bqui, 2021-09-13:
How were you able to the synchronization issue?  I am seeing the same problem.

noelpedro, 2021-09-22:
All sample counters must the same.

observables sample counter with those of the multi correlators.

Rafaykhan12, 2022-12-22:
@noelpedro can yo please explain how you solve this issue? I have also mailed you . I am getting the same problem as yours although I am running v17