gnss-sdr/gnss-sdr #529: What Causes the Increase in Sample Counter from Tracking Start?

jmtobin-uh, 2021-07-16:
When using GNSS-SDR there always seems to be an increase in the sample counter at some point during tracking. Here is the output of PRN_start_sample_counter from the tracking dump using the "my first position fix" data.
[image]

What causes the sharp change in sample count at the elbow of the plot? Thanks!

noelpedro, 2021-07-19:
There may exists a downsampler before acquisition and thus number of samples are shorter ( or the sample count is a lot less). The data code length might be greater for Tracking.

For example if sample rate is 4msps and Acquisition is done at 1msps. The sample count for ACQ is a a fourth less than the tracker's sample count.

jmtobin-uh, 2021-07-19:
It seems to be that the system is sampling at 1kHz and that elbow is when it drops down to 50Hz. I suppose my question becomes what causes the change in sampling frequency?

noelpedro, 2021-07-20:
Save some computational resources in Acquisition (i.e. reduce size of data vector of FFTs).

jmtobin-uh, 2021-07-20:
Sure, but why not 50 Hz for the entirety then? Why begin with 1kHz?

noelpedro, 2021-07-22:
Sample counter get uploaded by 50Hz or 20ms after bit synchronization.

jmtobin-uh, 2021-07-22:
Do you know where that cahnge is in the source code? I can't seem to find it.

noelpedro, 2021-07-23:
Should be at the bottom of the respective trackings’s general_work function.

noelpedro, 2021-07-23:
Actually look at where the log_data() occurs in case 2 and case 4 of tracking loop.

case 2 logs every ms while case4 logs every 20ms.

closed 2021-09-07 by jmtobin-uh