gnss-sdr/gnss-sdr #531: What functionality causes the most CPU usage?

jt3232, 2021-07-22:
Short: I am wondering what functions of GNSS SDR causes the most cpu usage. Is it tracking? Finding a position solution? Outputting data of some sort?

Long: I am wanting to run GNSS SDR just to output information about satellites like the 
GPS ephemera, GPS almanac, and gnss_synchro messages. Can I disable the other functionality like position solution while keeping my message functionality? This will help me run the software more efficiently on a slower/cheaper processor. Currently, running the software maxes out my cpu and it begins to lose its lock on the satellites and will have to start over on the acquisition and tracking.

noelpedro, 2021-07-22:
Tracking for sure.

ZiglioUK, 2021-07-22:
I suspect it's acquisition. I find tracking to be pretty efficient with GPS L1 up to 6 channels on an old 2 core Intel i5. Galileo is much more CPU intensive than GPS.  I would restrict the search range of acquisition, and increase the step size, use only 1 channel and don't use the standard algorithm, not the fine doppler one

noelpedro, 2021-07-23:
You can profile the code with gperftools and gprof.

AgileEngineeringLLC, 2021-08-07:
For what you're trying do, you may want to consider trying the gnss-sdrlib code on github instead. I'm working on something sort of similar myself using gnss-sdrlib, and may port some or all of it over to a Pi 4 compute module.

kurtulmehtap, 2021-08-20:
I wish we had the FPGA code for acceleration.