gnss-sdr/gnss-sdr #536: Bandwidth with RTL-SDR Blog V3

fdgm0001, 2021-08-16:
Hello guys. I'm using RTL-SDR Blog V3 to receive the GPS L1 C/A signal and it works perfectly. However, at the beginning of the GNSS-SDR execution appears that the actual bandwidth is 0 Hz. How is signal reception possible with zero bandwidth?

Here is the log ouptut when I execute GNSS-SDR:

`Initializing GNSS-SDR v0.0.14.git-next-d999a5e6a3 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: rtl,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.0.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 40.2 dB...
Actual Bandwidth: 0 [Hz]...
RF Channels: 1`

acebrianjuan, 2021-08-16:
Hi @fdgm0001,

This was already asked in the past.
You will find an explanation here: https://github.com/gnss-sdr/gnss-sdr/issues/427#issuecomment-696273264

Regards,
Álvaro

fdgm0001, 2021-08-22:
Thank you for the explanation, @acebrianjuan. However, I have another doubt ,¿how do I know which intermediate frequency GNSS-SDR uses?

closed 2021-09-17 by fdgm0001

ValZapod, 2021-11-21:
That was 3 bugs in the driver on top of each other that were fixed here. https://github.com/librtlsdr/librtlsdr