gnss-sdr/gnss-sdr #537: Extrange pseudranges values

fdgm0001, 2021-08-23:
Hi guys. I am using RTL-SDR Blog V3 to receive GPS L1 C/A signals. The receptor works well, however when I see the pseudornages values in gnss-sdr-monitor they are very extrange.  A example of this values is shown in the next image:

https://imgur.com/czLIh75

Are these values normal?

noelpedro, 2021-09-04:
Hi @fdgm0001 were u able to get a position fix with those values?

I’d say as long as ur able to get PVT I wouldn’t worry about the pseudo range values.

I’m currently having errors in PVT because pseudoranges are all mixed up some are even negative I see yours are all positive.

noelpedro, 2021-09-04:
Are ur TOW[ms] values mostly the same for each channel? None of my channels have the same TOW[ms] in real time.

noelpedro, 2021-09-04:
Is this real time results or file?

fdgm0001, 2021-09-06:
> Hi @fdgm0001 were u able to get a position fix with those values?
> 
> I’d say as long as ur able to get PVT I wouldn’t worry about the pseudo range values.
> 
> I’m currently having errors in PVT because pseudoranges are all mixed up some are even negative I see yours are all positive.

I could get a position fix @noelpedro. This results are in real time, but the prost-processing values are very similar. The TOW values are showed in the image and they are almost equal.