gnss-sdr/gnss-sdr #544: Pseudoranges from Multiple HackRF Ones on Same Antenna Far Off

jt3232, 2021-09-09:
I am running multiple HRF1s on the same antenna and they all output different pseudoranges, by a large amount. I do have a fix when this is happening. This is extremely unsettling to me. They each have the NooElec TCXO (10MHz at 0.5ppm). However, I get differences of 189km between two and 380km on another. I am not sure, but this seems like out of the realm of what should be acceptable. They each have the same configuration and are running on 3 identical computers. 

It seems that these weird errors follow the HRF1 if I switch them between computers. How can they possibly be so far off? I am needing to retrieve the pseudorange, so having them as accurate as possible is a necessity.

As a side note, we have some old receivers we are wanting to replace and they are all within 30m of each other. I was hoping for something around that precision.

Thanks,

Jaxon

jt3232, 2021-09-09:
From the GNSS-SDR website, it appears that the way the pseudorange is calculated may be the cause of this. It appears to only use one time correction across all the satellites. I don't fully understand this yet, but will keep looking into it.
[https://gnss-sdr.org/docs/sp-blocks/observables/]

JohnNatHaysys, 2023-09-19:
The observables document clearly states the pseudo range is set to the earliest TOW, and is 68.802ms. This does not appear to work, and I cannot find anything in the code which would do this. We observe raw pseudoranges which are random, even negative.
Is there an explanation? I can see that PVT can correct for this but only after a fix has been achieved.