gnss-sdr/gnss-sdr #551: Loss of Lock on Satellites that Seem Good

bqui, 2021-09-22:
I am having an intermittent issue where occasionally I am not able to track any satellite for a sustained amount of time.  The console output says that each channel is able to initially acquire a PRN within the first second of operation.  Then 16 seconds later, all channels lose lock and the process starts anew.  The output looks like this and is repeated over and over again if it is allowed to run indefinitely...

> I0922 21:13:45.842315  5372 control_thread.cc:362] Flowgraph started
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
> I0922 21:13:45.851287  5394 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 2610 ( 0.001305 s)
> Current receiver time: 1 s
> Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
> I0922 21:13:45.860208  5397 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 2890 ( 0.001445 s)
> Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 31 (Block IIR-M)
> Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 32 (Block IIF)
> I0922 21:13:45.871477  5391 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 4145 ( 0.0020725 s)
> I0922 21:13:45.871490  5388 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 3403 ( 0.0017015 s)
> Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 22 (Block IIR)
> I0922 21:13:46.348507  5400 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 4656 ( 0.002328 s)
> Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 01 (Block IIF)
> I0922 21:13:46.621855  5385 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 4656 ( 0.002328 s)
> Current receiver time: 2 s
> Current receiver time: 3 s
> Current receiver time: 4 s
> Current receiver time: 5 s
> Current receiver time: 6 s
> Current receiver time: 7 s
> Current receiver time: 8 s
> Current receiver time: 9 s
> Current receiver time: 10 s
> Current receiver time: 11 s
> Current receiver time: 12 s
> Current receiver time: 13 s
> Current receiver time: 14 s
> Current receiver time: 15 s
> Current receiver time: 16 s
> Loss of lock in channel 2!
> I0922 21:14:01.850934  5394 dll_pll_veml_tracking.cc:986] Loss of lock in channel 2 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Current receiver time: 17 s
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
> I0922 21:14:01.861338  5394 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 5923 ( 0.0029615 s)
> Loss of lock in channel 1!
> I0922 21:14:01.861460  5397 dll_pll_veml_tracking.cc:986] Loss of lock in channel 1 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
> I0922 21:14:01.869690  5397 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 7575 ( 0.0037875 s)
> Loss of lock in channel 4!
> I0922 21:14:01.869752  5388 dll_pll_veml_tracking.cc:986] Loss of lock in channel 4 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Loss of lock in channel 3!
> I0922 21:14:01.870155  5391 dll_pll_veml_tracking.cc:986] Loss of lock in channel 3 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 32 (Block IIF)
> I0922 21:14:01.883471  5388 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 7575 ( 0.0037875 s)
> Loss of lock in channel 0!
> I0922 21:14:02.349059  5400 dll_pll_veml_tracking.cc:986] Loss of lock in channel 0 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 31 (Block IIR-M)
> I0922 21:14:02.579843  5400 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 3389 ( 0.0016945 s)
> Loss of lock in channel 5!
> I0922 21:14:02.623445  5385 dll_pll_veml_tracking.cc:986] Loss of lock in channel 5 (carrier_lock_fail_counter:5001 code_lock_fail_counter : 0)
> Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 01 (Block IIF)
> I0922 21:14:02.631170  5385 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 6499 ( 0.0032495 s)
> Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 22 (Block IIR)
> I0922 21:14:02.830731  5391 dll_pll_veml_tracking.cc:1738] Number of samples between Acquisition and Tracking = 5155 ( 0.0025775 s)

The weird thing is that the if I plot the tracking data using the Matlab tool, it appears that it is being decoded correctly initially and then all of the sudden, the tracking loops completely lose track of all the data.  The system stays in a lost state for 15 or so seconds until resetting when the process begins again.  The short term and long term effects are shown in the Matlab plots below.  I have tried experimenting with the various command line tracking parameters but as of yet, I cannot find a way to get the system to not lose track on each PRN.  It is worth noting that I am using a system with a high end clock so I don't think it is an issue with a low fidelity oscillator.

My configuration for acquisition through PVT is 

> ;######### ACQUISITION GLOBAL CONFIG ############
> Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
> Acquisition_1C.item_type=gr_complex
> Acquisition_1C.pfa=0.02
> Acquisition_1C.doppler_max=6000
> Acquisition_1C.doppler_step=100
> 
> ;######### TRACKING GLOBAL CONFIG ############
> Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
> Tracking_1C.item_type=gr_complex
> Tracking_1C.pll_bw_hz=50;
> Tracking_1C.dll_bw_hz=2;
> Tracking_1C.dump=true
> Tracking.1C.dump_filename=no_nav
> ;Tracking.1C.max_lock_fail=200
> ;Tracking.1C.pull_in_time_s=10
> Tracking.1C.carrier_lock_th=0.75
> Tracking.1C.max_carrier_lock_fail=7500
> 
> ;######### TELEMETRY DECODER GPS CONFIG ############
> TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
> 
> ;######### OBSERVABLES CONFIG ############
> Observables.implementation=Hybrid_Observables
> Observables.dump=true;
> Observables.dump_mat=true;
> 
> 
> ;######### PVT CONFIG ############
> PVT.implementation=RTKLIB_PVT
> PVT.averaging_depth=100
> PVT.flag_averaging=true
> PVT.output_rate_ms=10
> PVT.display_rate_ms=500
> PVT.dump_mat=true
> PVT.trace=3

The tracking outputs are shown below

> [image]
> [image]
> [image]
> [image]

ZiglioUK, 2021-09-23:
I know, it's very frustrating. I've been trying to find whether there's a bug or not for a long time. Sometimes it seems like the discriminator for the tracking loops (either PLL or FLL) returns a result in the opposite "direction" and the loops starts diverging until tracking is lost. And that's despite the CN0 being strong and all other indicators. I've played with the bandwidth and filter order. Frustratingly, the behaviour of GNSS-SDR is non-deterministic so you can't just run a test over and over again and expect the same result.

That's coupled with another problem: acquistion. The standard algorithm (not the Tong one or fine doppler) also returns a value for doppler that's often quite wrong, no matter what settings I use. And that's bad because there's no way the subsequent state (FLL pull-in) can work. I mean, it can work if its bandwidth is quite large and the value from acquisition is not too far off.
That's another frustrating part.

I don't know what the solution is, maybe more unit tests? I'd love to ditch tracking loops altogether in a modern GPS receiver. They're basically digital version of analog control loops, with all their limitations.

shuoyuanxu, 2021-11-23:
GLONASS with 10MHz fps have no such issues, just a piece of information if it helps.

bqui, 2021-11-23:
It turned out to be an issue with the data missing from the incoming data stream.  Thanks for your help.

closed 2021-11-23 by bqui

shuoyuanxu, 2021-11-23:
Hello Bqui

I am still having the issue of losing perfectly good satellites on GPS L1. Do you mind sharing some of your experiences?

I am using a USRP to connect to a simulator. I have pushed the fps to 25MHz and still no luck. I have seen no 'DDD' or 'OOO' from my terminal. And I haven't noticed any missing packet from your description. Can you elaborate on that?

My config
https://www.dropbox.com/s/ftqhp5vpbi1crj5/base.conf?dl=0

Many Thanks
Shu

bqui, 2021-11-23:
Shu,

When I examined the data, all the satellites went out at the same instant midway through a nav message bit.  If I plotted the tracking data dump using dll_pll_veml_plot_sample.m, this was apparent.  Following this, the software guy I am working with dug into the interface into the interface between our sampling hardware and gnu-radio and found where we were dropping data.  I am not a software guy and don't understand the particulars, but that is where we started.  One possible approach you could pursue is to see if you see the same effect in E1 Galileo as it is the same frequency so discontinuities should affect both at the same instant in time.  I hope that helps.

Britton