gnss-sdr/gnss-sdr #564: GNSS SDR Flowgraph causes performance issues for real-time operations

jwmelto, 2021-10-19:
This issue is for tracking performance issues noted in real-time processing.

I believe that there are design/implementation details in GNSS-SDR that work at odds to the GNU Radio scheduler. The scheduler is responsible for calling the `work()` (or `general_work()`) methods of each block. Basically, the scheduler is driven by available space in the output buffer. If there is insufficient space for a block to produce output, it won't be called. This ultimately backs up the flowgraph to the front end. If a real-time source is not called often enough, data will be dropped.

Obviously, logic based on maintaining phase lock and demodulating data from a signal is going to be sensitive to dropped data, because it affects the coherency of the decoded signal. 

My initial investigation has shown that the default buffer sizes are too small for a moderate (8msps) throughput signal. Manually increasing the buffer size by calling `set_min_output_buffer()` has proven effective, until the start of PVT. Then the signal source backs up. I'm still investigating the behavior of downstream blocks.

noelpedro, 2021-11-16:
Hi @jwmelto Im doing ad9361 signal source and noticing my hardware is overflowing when I reach PVT. I'm able to compute PVT and display it to std. out then shortly after I lose lock and all channels.

jwmelto, 2021-11-16:
the ultimate fix is to add `set_tag_propagation_policy(TPP_DONT);` in the tracking block ctor. I'm working on a refactoring of **dll_pll_veml_tracking.cc** that may have wider application, but I am not in a position to test all algorithms

noelpedro, 2021-11-17:
Didn’t solve the issue for my front end.