gnss-sdr/gnss-sdr #571: HackRF, "Loss of lock" frequently!

unalfaruk, 2021-11-21:
I prepared a setup consists of HackRF, TCXO, GNSS Antenna and PC. I placed the antenna on the balcony. 

The antenna is from JC Antenna brand. Its specs from local market:
[image].

When I run GNSS-SDR (bias-tee always enabled), it always repeates the message of "Loss of lock in channel X";

> Initializing GNSS-SDR v0.0.11 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2017.02.1
Actual RX Rate: 2000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
Actual RX Gain: 14.000000 dB...
Actual Bandwidth: 6000000.000000 [Hz]...
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 22 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 24 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 31 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 32 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 02 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 03 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 2 s
Current receiver time: 3 s
OOOOOOOCurrent receiver time: 4 s
OOOCurrent receiver time: 5 s
OOOOCurrent receiver time: 6 s
OOOOCurrent receiver time: 7 s
OOOOOCurrent receiver time: 8 s
OOOOCurrent receiver time: 9 s
OOOOOCurrent receiver time: 10 s
OOOOCurrent receiver time: 11 s
OOOOCurrent receiver time: 12 s
OOOOCurrent receiver time: 13 s
OOOCurrent receiver time: 14 s
OOOOCurrent receiver time: 15 s
OLoss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 12 (Block IIR-M)
OOLoss of lock in channel 1!
Current receiver time: 16 s
OOTracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 30 (Block IIF)
OLoss of lock in channel 7!
OOCurrent receiver time: 17 s
OLoss of lock in channel 3!
OTracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 10 (Block IIF)
OLoss of lock in channel 4!
Loss of lock in channel 6!
Loss of lock in channel 5!
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 08 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 24 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 13 (Block IIR)
OTracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 18 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 31 (Block IIR-M)

GNSS-SDR version: 0.0.11
My PC Specs: Intel Atom N570 1.66 GHz, 2 GB RAM, Ubuntu 20 LSE, 64 bit (It seems ok while capturing, however is it enough for signal capture?)
My setup:
[image]

I could not find the problem, I also look at the 1.5 G spectrum using CubicSDR, everything seems ok, the spectrum flows well and the pattern is like expected. Could someone have an idea to debug it?

jwmelto, 2021-11-21:
I believe that nearly all of the “loss of lock” issues people have are related to data loss.  That’s clearly the case in your situation.  Those “O”s in the output are your device driver telling you that it overflowed. You can’t track a signal when you are randomly dropping parts of it.

ZiglioUK, 2021-11-21:
Good point about overflowing. That shouldn't happen with a 2MHz sampling rate. Try with fewer satellites. It looks like a portion of the sky would be blocked anyway. Also a 14dB gain is too low in my experience. Can you share your conf file?

ZiglioUK, 2021-11-21:
One more thing. The Intel Atom is a very low end CPU. Definitely try with fewer satellites, like 4 and see what happens. Even one satellite to make sure tracking is stable. Make sure to run volk-profile and volk_gnss-sdr_profile

unalfaruk, 2021-11-22:
> I believe that nearly all of the “loss of lock” issues people have are related to data loss. That’s clearly the case in your situation. Those “O”s in the output are your device driver telling you that it overflowed. You can’t track a signal when you are randomly dropping parts of it.
I changed the test PC with another PC which is a workstation. "O"s are not appeared anymore. But still, it losing the lock state frequently.

unalfaruk, 2021-11-22:
> Good point about overflowing. That shouldn't happen with a 2MHz sampling rate. Try with fewer satellites. It looks like a portion of the sky would be blocked anyway. Also a 14dB gain is too low in my experience. Can you share your conf file?

Sure, I just copied the HackRF configuration from the website:

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

> One more thing. The Intel Atom is a very low end CPU. Definitely try with fewer satellites, like 4 and see what happens. Even one satellite to make sure tracking is stable. Make sure to run volk-profile and volk_gnss-sdr_profile. (The antenna should be flat on a horizontal plane)

After I replaced the PC with a good one, the frequency of "loss of lock" state decreased, however still it loses. I just run "volk-profile" and "volk_gnss-sdr_profile", tomorrow night I will try it again.

BTW, when I change the gain values, the period of "loss of lock" also changes. I am not sure which gain values I should use for RF, IF and BB while I am using an active GNSS antenna (it also has LNA). Btw, I do not know why it says "14 dB", there is no parameters which set as "14 dB". Maybe it is related to HackRF's [on/off amplifier](https://pub.nethence.com/radio/hackrf).

ZiglioUK, 2021-11-22:
Yeah, I don't know about why it says 14dB either.
I think that's too much. The HackRF doesn't have an IF gain, I set gain and rf_gain to 20 with an outdoor antenna.
Important: have you installed an external TCXO? the standard oscillator is really poor.
With an external TCXO you wouldn't need a translating filter either.
Finally, with a faster CPU, you could increase the sampling rate and gain some signal that way, not a dramatic difference.

Here are my settings:
````
;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=400000
                         
;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.freq=1575420000
SignalSource.sampling_frequency=4000000
                                
SignalSource.gain=20
SignalSource.rf_gain=20
SignalSource.AGC_enabled = false
SignalSource.osmosdr_args=hackrf,bias=1

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### GPS ACQUISITION CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.max_dwells=1
Acquisition_1C.threshold=2.5
Acquisition_1C.doppler_max=6500
Acquisition_1C.doppler_step=250

;######### TRACKING GPS CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.order=2;
Tracking_1C.carrier_aiding=true
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.pull_in_time_s=3;
Tracking_1C.fll_bw_hz=100
Tracking_1C.pll_filter_order=2
Tracking_1C.pll_bw_hz=100.0;
Tracking_1C.dll_filter_order=1
Tracking_1C.dll_bw_hz=8.0;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.extend_correlation_symbols=5;
Tracking_1C.cn0_min=15
Tracking_1C.max_lock_fail=200

unalfaruk, 2021-11-24:
> Important: have you installed an external TCXO? the standard oscillator is really poor.

Sure, I installed it, you can see that in the photo.

Your config has a few wrong settings. I think your sampling frequency is not normal, it is 400 kHz. Also, when I tried to run your conf, I got errors. Then I just changed my conf by looking at your values, still did not work.

I tried the setup with a GPS simulator, passive antenna with an external amplifier, and active antenna also. It is not working. Interesting. How can I be sure that my TXCO working properly?

ZiglioUK, 2021-11-24:
Ok, I've fixed my settings above, it should work now. Thanks for spotting the error with the frequency value.

The command to check if the HackRF can see the external clock is
`hackrf_debug --si5351c -n 0 -r`
It should return
`[  0] -> 0x01`

Also you could verify that the bias tee is enabled with a multimeter.

unalfaruk, 2021-11-25:
Thank you for your assist. It is still not working.

This command gives [0] -> 0x01, it is ok. It is probably showing the detection status of TCXO, not its accuracy, but still ok.

Bias tee is working, I verified it. Interesting, everything seems ok!

By the way, I am sure about HackRF, it is working, at least for Radio FM frequency. I could decode FM radio signals. I have tried different antennas, different amplification values/devices, different configurations... but it is not working for L1 frequency. So what should I check now, any idea? :)

ZiglioUK, 2021-11-26:
Hi, I would suggest to try with an inexpensive RTL-SDR dongle, the ones with a TCXO.
They work more reliably than some HackRFs. I've been lucky with mine. Although, not an "official" one it came with a metal case and external TCXO, it works as intended. I've upgraded the firmware too, not sure whether that might help. Also I use a shielded cable, and a large patch antenna from the rtl-sdr blog web site. I don't have a problem with other antennas though.

Today I tried the [front-end-cal](https://github.com/gnss-sdr/gnss-sdr/tree/main/src/utils/front-end-cal) project. I was curious to see whether it would work with the HackRF.
It took me ages, I think there are a few bugs but eventually I got something out, using an rtl-sdr dongle.
The doppler offset result was ok, but the sampling frequency offset was completely off so didn't go any further.

The fact that the FM radio works unfortunately doesn't help with GPS. Perhaps if you have the chance to generate a tone on some frequency you could look at it with the HackRF, and see if the frequency is very different. I don't have any more ideas at this stage maybe someone else can chip in.

fener1efes2, 2022-07-24:
I have tried

> Hi, I would suggest to try with an inexpensive RTL-SDR dongle, the ones with a TCXO. They work more reliably than some HackRFs. I've been lucky with mine. Although, not an "official" one it came [...]

I have tried with RTL-SDR, and it does not work in real space.
There are two test setups that I use:

First one:
I created GPS L1 C/A signal with GPS-SDR-SIM and and transmitted it by using Hack-RF. 
Receiver is RTL-SDR by using GNSS-SDR.
I connected Hack-RF to RTL-SDR with coax cable.
GNSS-SDR successfully tracked and calculated positions.

Second one:
I used an active antenna having 15dB gain, after antenna I placed an LNA having 20dB gain.
Antenna placed on surface where it could see the clear sky.
Experiment has done when there is no clouds in the sky.
Receiver is RTL-SDR by using GNSS-SDR.
I connected the LNA to RTL-SDR with coax cable.
GNSS-SDR could not calculate positions and lost its track to satellites constantly.

I do not get why it does not work?

unalfaruk, 2022-07-24:
Guys,

SOLVED! I also shared my results on the HackRF Discord community. My problem completely was related to the version inside HackRF. I just upgraded it.

I generated satellite signals by a GNSS simulator, then captured the signal with HackRF One. Finally, I processed the signal to see whether I can acquire any or not, and it worked!

[image]
[image]

Moreover, I did try an outdoor test, and it also worked!

@fener1efes2 I think, you have a problem with your setup in the second case. Don't forget to enable bias tee on RTL-SDR if possible for the active antenna. Try different antennas. For debugging, try signal records on a ".bin" file, and analyze it with GNSS-SDR (MATLAB version). Sorry for closing the issue now, please open a new issue here.

closed 2022-07-24 by unalfaruk

fener1efes2, 2022-07-27:
congrats @unalfaruk !
my antenna works well, since I tested it with U-BLOX. I use an external bias tee for the active antenna.