gnss-sdr/gnss-sdr #573: Pseudorange Comparison of HackRF One and Certified GPS Receiver

jt3232, 2021-12-10:
I am comparing the raw pseudorange values of 4 HackRF Ones and 3 Thales GG12 certified GPS receivers all connected to the same antenna. I took the average GG12 pseudorange and then subtracted all receivers' pseudoranges relative to that average (which I can reasonably expect to be very close to the true pseudorange). Please view the results of a single satellite.

[image]

Does anyone have any idea why these receivers act this way? Someone mentioned to me it may be the internal oscillator which has many "jumps" to correct itself. I am using smoothed pseudoranges through GNSS SDR and I am manually subtracting the user clock offset from the pseudorange since the software doesn't appear to do that (when streaming over UDP).

Jaxon

jt3232, 2021-12-15:
> 

Thanks for the reply. I can try to capture the DLL output. Would that parameter be located in the config file somewhere? 

I am using the Nooelec TCXO on 3 of them and the Leo Bodnar GPS mini external reference on the other. From my tests, they all act the same no matter their external clock. If I didn't have any external clock, no satellite could be locked with the HackRF One. Every PRN looks different but has the same general idea. The pseudoranges jump at a consistent frequency.

I believe the sampling frequency is 2MHz, whichever is the default. I am running these HRF1s with Intel NUCs so processing speed is not an issue in this case.

ZiglioUK, 2021-12-15:
It's not su much jumping, it looks like phase unwrapping that's going on. 
The DLL tracks the code phase, which is the 1ms fractional part of a pseudorange.
A code phase period is 300km while the pseudorange "jump" you're seeing is more like 23m, so clearly the two things are not directly related. I was asking about the sampling frequency because a sawtooth behaviour in the DLL is observed when the sampling frequency is a multiple of the chipping rate (1023chips/ms). That's a property of all PLLs, it was a remote possibility though.
The DLL output can be captured by setting `Tracking_1C.dump=true` in the [configuration file](https://gnss-sdr.org/docs/sp-blocks/tracking/#implementation-gps_l1_ca_dll_pll_tracking).
Perhaps if you don't mind you could share your conf file and snippets of the analysis code, I'll try and reproduce with my HackRF.

ZiglioUK, 2021-12-15:
One more thing, about

> I am manually subtracting the user clock offset from the pseudorange since the software doesn't appear to do that 

Have you tried setting `PVT.enable_rx_clock_correction=true`? [It's set to `false` by default](https://gnss-sdr.org/docs/sp-blocks/pvt/#implementation-rtklib_pvt).

jt3232, 2021-12-17:
> [GNSS-SDR]
> 
> ;######### GLOBAL OPTIONS ##################
> GNSS-SDR.internal_fs_sps=2000000
> GNSS-SDR.telecommand_enabled=true
> GNSS-SDR.telecommand_tcp_port=3333
> 
> ;######### SIGNAL_SOURCE CONFIG ############
> SignalSource.implementation=Osmosdr_Signal_Source
> SignalSource.item_type=gr_complex
> SignalSource.sampling_frequency=2000000
> SignalSource.freq=1575420000
> SignalSource.gain=40
> SignalSource.rf_gain=40
> SignalSource.if_gain=50
> SignalSource.AGC_enabled=false
> SignalSource.samples=0
> SignalSource.repeat=false
> ;# Next line enables the internal HackRF One bias (3.3 VDC)
> SignalSource.osmosdr_args=hackrf=0,bias=0
> SignalSource.enable_throttle_control=false
> SignalSource.dump=false
> SignalSource.dump_filename=./signal_source.dat
> 
> ;######### SIGNAL_CONDITIONER CONFIG ############
> SignalConditioner.implementation=Signal_Conditioner
> 
> ;######### DATA_TYPE_ADAPTER CONFIG ############
> DataTypeAdapter.implementation=Pass_Through
> 
> ;######### INPUT_FILTER CONFIG ############
> InputFilter.implementation=Freq_Xlating_Fir_Filter
> InputFilter.decimation_factor=1
> InputFilter.input_item_type=gr_complex
> InputFilter.output_item_type=gr_complex
> InputFilter.taps_item_type=float
> InputFilter.number_of_taps=5
> InputFilter.number_of_bands=2
> InputFilter.band1_begin=0.0
> InputFilter.band1_end=0.85
> InputFilter.band2_begin=0.9
> InputFilter.band2_end=1.0
> InputFilter.ampl1_begin=1.0
> InputFilter.ampl1_end=1.0
> InputFilter.ampl2_begin=0.0
> InputFilter.ampl2_end=0.0
> InputFilter.band1_error=1.0
> InputFilter.band2_error=1.0
> InputFilter.filter_type=bandpass
> InputFilter.grid_density=16
> InputFilter.dump=false
> InputFilter.dump_filename=../data/input_filter.dat
> 
> ;######### RESAMPLER CONFIG ############
> Resampler.implementation=Pass_Through
> 
> ;######### CHANNELS GLOBAL CONFIG ############
> Channels_1C.count=32
> Channels.in_acquisition=1
> Channel.signal=1c
> 
> ;######### ACQUISITION GLOBAL CONFIG ############
> Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
> Acquisition_1C.item_type=gr_complex
> Acquisition_1C.coherent_integration_time_ms=1
> Acquisition_1C.pfa=0.01
> Acquisition_1C.doppler_max=30000
> Acquisition_1C.doppler_step=150
> Acquisition_1C.max_dwells=1
> Acquisition_1C.dump=false
> Acquisition_1C.dump_filename=./acq_dump.dat
> 
> ;######### TRACKING GLOBAL CONFIG ############
> Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
> Tracking_1C.item_type=gr_complex
> Tracking_1C.extend_correlation_symbols=10
> Tracking_1C.early_late_space_chips=0.5
> Tracking_1C.early_late_space_narrow_chips=0.15
> Tracking_1C.pll_bw_hz=40
> Tracking_1C.dll_bw_hz=2.0
> Tracking_1C.pll_bw_narrow_hz=5.0
> Tracking_1C.dll_bw_narrow_hz=1.50
> Tracking_1C.fll_bw_hz=10
> Tracking_1C.enable_fll_pull_in=true
> Tracking_1C.enable_fll_steady_state=false
> Tracking_1C.dump=true
> Tracking_1C.dump_filename=tracking_ch_
> 
> ;######### TELEMETRY DECODER GPS CONFIG ############
> TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
> TelemetryDecoder_1C.dump=false
> 
> ;######### OBSERVABLES CONFIG ############
> Observables.implementation=Hybrid_Observables
> Observables.dump=false
> Observables.dump_filename=./observables.dat
> Observables.enable_carrier_smoothing=true
> 
> ;######### PVT CONFIG ############
> PVT.implementation=RTKLIB_PVT
> PVT.positioning_mode=PPP_Static
> ;PVT.enable_rx_clock_correction=true
> PVT.output_rate_ms=100
> PVT.display_rate_ms=500
> PVT.iono_model=Broadcast
> PVT.trop_model=Saastamoinen
> PVT.flag_rtcm_server=true
> PVT.flag_rtcm_tty_port=false
> PVT.rtcm_dump_devname=/dev/pts/1
> PVT.rtcm_tcp_port=2101
> PVT.rtcm_MT1019_rate_ms=5000
> PVT.rtcm_MT1077_rate_ms=1000
> PVT.output_enabled=true
> PVT.rinex_version=2
> PVT.enable_monitor=true
> PVT.enable_protobuf=true
> PVT.monitor_client_addresses=127.0.0.1_10.10.10.1
> PVT.monitor_udp_port=1132
> PVT.enable_monitor_ephemeris=true
> PVT.monitor_ephemeris_client_addresses=127.0.0.1_10.10.10.1
> PVT.monitor_ephemeris_udp_port=1133
> PVT.elevation_mask=0
> 
> 
> 
> ;######### MONITOR CONFIG ############
> Monitor.enable_monitor=true
> Monitor.enable_protobuf=true
> Monitor.decimation_factor=25
> Monitor.client_addresses=127.0.0.1_10.10.10.1
> Monitor.udp_port=1131

The dump files are attached below as well. I do not know what to do with them.

> One more thing, about

> >  I am manually subtracting the user clock offset from the pseudorange since the software doesn't appear to do that

> Have you tried setting PVT.enable_rx_clock_correction=true? It's set to false by default.

I tried this a long time ago and from what I recall, it made very little difference to the pseudorange values that were spit out over UDP. 
The most efficient (albeit very tedious) way I have used is to subtract the corresponding RX time's user_clock_offset from the pseudorange. I was able to align those data points well by editing the source code to output the monitor_udp data every 0.50 seconds exactly. I also edited the code to output the gnss_synchro data exactly every .50 and .00 seconds. Using just the decimation factor for 0.5 second output could allow the RX times to be 0.25 or 0.75 or any random numbers after the decimal. I had to do this because the certified GPS receivers output exactly every .00 and .50. 

[PRN30_TrackingFiles.zip](https://github.com/gnss-sdr/gnss-sdr/files/7731347/PRN30_TrackingFiles.zip)

ZiglioUK, 2021-12-19:
hi @jt3232 

wow I can't believe `Acquisition_1C.doppler_max=30000`.
A low ppm TCXO search space wouldn't be greater than 6.5KHz. 
That makes me think the HackRFs you were using are still using their stock oscillator.

Regardless, I've processed the file you sent with `dll_pll_veml_plot_sample.m`. Tracking looks perfect (skipping the first 15,000 samples). The output of the prompt correlator taps (the yellow line in Correlation Results) is steady above the others as it should be. There's a tiny oscillation but I don't think that's the cause of the problem you're seeing

[image]

ZiglioUK, 2021-12-19:
I have a question myself if you don't mind. I'm trying to plot something similar to yours.
I've generated a simulated file with gps-sdr-sim and played it first into a uBlox receiver then I've processed it with gnss-sdr.
Theres a range gap between the two, is that the clock offset you've manually removed?

[image]

jt3232, 2021-12-19:
> I have a question myself if you don't mind. I'm trying to plot something similar to yours. I've generated a simulated file with gps-sdr-sim and played it first into a uBlox receiver then I've [...]

Yes, I subtracted `user_clk_offset` multiplied by the speed of light from the "raw" pseudorange value. 

Also, I really appreciate how neat the `dll_pll_veml_plot_sample.m` output looks. I will look into trying that for myself!

Jaxon

jt3232, 2021-12-19:
Also, I inherited the HRF1 code. I can guarantee the HRF1s were typical stock with Nooelec external TCXOs as I had bought them myself from Amazon.

ZiglioUK, 2021-12-21:
> Yes, I subtracted `user_clk_offset` multiplied by the speed of light from the "raw" pseudorange value.

I'm still working on this. I didn't log the message with that solution from uBlox or RTK-LIB, might have to solve the Least Square System myself, it won't be hard

> Also, I really appreciate how neat the `dll_pll_veml_plot_sample.m` output looks. I will look into trying that for myself!

You're welcome, hopefully we'll get to the bottom of this

ZiglioUK, 2021-12-21:
> Also, I inherited the HRF1 code. I can guarantee the HRF1s were typical stock with Nooelec external TCXOs as I had bought them myself from Amazon.

look at the observed dopplers, they really should all be within +/-6.5Khz. If you restrict the range, gnss-sdr won't use as much CPU

ZiglioUK, 2021-12-22:
Hi again, here's an update. Still after processing a simulated file. This time I've corrected for the receiver clock bias.
The range values I get from the uBlox M8N and gnss-sdr are not as close as yours. In fact they seem to diverge. There's no periodicity in their difference though. Incidentally the CN0 values from gnss-sdr are vey noisy.

[image]

Interesting to see what the local clock bias looks like

[image]

I'll now try and do the same over real data capture.

jt3232, 2022-09-08:
@ZiglioUK 
I thought I'd give an update. I took quite a long break from this, but I tried using a USRP E310 receiver and here are my results:
[image]

By the way, how did you get the above plots? Did you use another GNSS-SDR script or generate your own? Thanks!

jt3232, 2022-09-08:
This image shows the oscillation is due to the clock error's oscillations. This graph corresponds to the HRF1 4 above.
[image]

ZiglioUK, 2022-09-20:
hi @jt3232 

> By the way, how did you get the above plots? Did you use another GNSS-SDR script or generate your own? Thanks!

sorry, it's been a while! they're my own scripts, nothing special. I need to dig them out again